GPSBabel/gpsbabel #5: Export Humminbird HT

Beaulieu80, 2015-08-10:
hi, when I export i do not have any depth values in CSV or GPX: can this because I have a none English PC?

<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" creator="GPSBabel - http://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/0">
  <time>2015-08-10T20:33:07.062Z</time>
  <bounds minlat="48.756284016" minlon="-72.088132040" maxlat="48.765394338" maxlon="-72.055048391"/>
  <trk>
    <name>07/26/15 11:45 AM</name>
    <trkseg>
      <trkpt lat="48.765394338" lon="-72.055048391">
        <ele>0.000000</ele>
      </trkpt>
      <trkpt lat="48.765376523" lon="-72.055111270">
        <ele>0.000000</ele>
      </trkpt>
      <trkpt lat="48.765352769" lon="-72.055183133">
        <ele>0.000000</ele>
      </trkpt>
      <trkpt lat="48.765340893" lon="-72.055254995">
        <ele>0.000000</ele>
      </trkpt>
      <trkpt lat="48.765329016" lon="-72.055326857">
        <ele>0.000000</ele>
      </trkpt>
      <trkpt lat="48.765317139" lon="-72.055407703">
        <ele>0.000000</ele>
      </trkpt>
      <trkpt lat="48.765293385" lon="-72.055479565">
        <ele>0.000000</ele>
      </trkpt>

48.69745, -72.10734, 
48.69748, -72.10738, 
48.69751, -72.10744, 
48.69754, -72.10749, 
48.69757, -72.10753, 
48.69759, -72.10759, 
48.69762, -72.10764,

gpsbabel, 2015-08-10:
Humminbird depth is supported only in formats that support Humminbird depth.  csv is not one of those formats.  GPX is, but you have to turn it on.

/gpsbabel -i humminbird -f ~/Downloads/015.HT -o gpx,humminbirdextensions -F - | head -30
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.1" creator="GPSBabel - http://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/1" xmlns:h="http://humminbird.com">
  <metadata>
    <time>2015-08-10T20:23:52.372Z</time>
    <bounds minlat="48.697448603" minlon="-72.122419385" maxlat="48.759639706" maxlon="-72.071316240"/>
  </metadata>
  <trk>
    <name>08/09/15 11:25 AM</name>
    <number>15</number>
    <trkseg>
      <trkpt lat="48.697448603" lon="-72.107337265">
        <ele>0.000000</ele>
      </trkpt>
      <trkpt lat="48.697478336" lon="-72.107382179">
        <ele>0.000000</ele>
        <extensions>
          <h:depth>709.000000000</h:depth>
        </extensions>
      </trkpt>
      <trkpt lat="48.697508068" lon="-72.107436076">
        <ele>0.000000</ele>
        <extensions>
          <h:depth>696.000000000</h:depth>
        </extensions>
      </trkpt>
      <trkpt lat="48.697537800" lon="-72.107489973">
        <ele>0.000000</ele>
        <extensions>
          <h:depth>706.000000000</h:depth>
        </extensions>

closed 2015-08-10 by gpsbabel

Beaulieu80, 2015-08-11:
In the help document from the humminbird section I do not see how to export with that option on? thank you

Beaulieu80, 2015-08-11:
I would like to merge all my HT files into fewer one to maximize each HT maximum trace points capacity (21k). So i'm trying to find a way to export all my HT files into fewer one. thank you

tsteven4, 2015-08-11:
look for

      |humminbirdextensions|

on
http://www.gpsbabel.org/htmldoc-development/fmt_gpx.html/#fmt_gpx_o_humminbirdextensions