GPSBabel/gpsbabel #13: Garmin VIRB Edit

sendbits, 2015-11-17:
Receive the following error when using the most recent release of GPSbabel on Windows 7 with Garmin's VIRB Edit files:

fit: Bad field size in data message

When attempting to merge fit files such as the two attached files. The attached files have been renamed from *.fit to *.txt.

[2015-09-01-12-07-54.txt](https://github.com/gpsbabel/gpsbabel/files/36879/2015-09-01-12-07-54.txt)
[2015-09-01-12-15-11.txt](https://github.com/gpsbabel/gpsbabel/files/36880/2015-09-01-12-15-11.txt)

tsteven4, 2015-11-17:
Can you let us know what command line you used to generate the error?

It seems to work for me:
> % ./gpsbabel -i garmin_fit -f 2015-09-01-12-15-11.fit -f 
> 2015-09-01-12-07-54.fit -x validate,debug -o gpx -F x.gpx
>
> Processing waypts
> point ct: 0, waypt_count: 0
>
> Processing routes
> route head ct: 0, route_count: 0
> total route point ct: 0, route_waypt_count: 0
>
> Processing tracks
> track 0 ct: 21, waypt_count: 21
> track 0 ct: 46, waypt_count: 46
> track head ct: 2, track_count: 2
> total track point ct: 67, track_waypt_count: 67

tsteven4, 2015-11-17:
this is broken in 1.5.2 and fixed in the master branch.  If you can 
build from source you can enjoy the fix now.

sendbits, 2015-11-18:
What timeframe to get deployed to the pre-built windows version?

sendbits, 2015-11-18:
gpsbabel -t -i garmin_fit -f 2015-10-04-22-10-31.fit -x track,merge -o kml -F gpsbabel-merge.gpx -o gpx -F GPSBabel.Zx1248

 

when run from Windows CLI

sendbits, 2015-11-18:
Quasi-related – does GPSbuild extract the accelerometer / gyro info from the Virb FIT file &/or would that be in a separate file stored on the Garmin VIRB?

robertlipe, 2015-11-18:
We don't hear much about Virb; my archives show you're only the second
person to ever ask about them. There's certainly no Virb-specific code in
GPSBabel.

sendbits, 2015-11-18:
Generalizing, if it’s possible for a FIT file to contain accelerometer / gyroscopic data in some standard way, will that data be translated to any of the text data formats, and if so, which data formats support exporting those fields, assuming they exist?

robertlipe, 2015-11-18:
There are three pieces to that in addition to the more general questions of
"does it make sense for GPSBabel"" and "who is doing the work?"

1) It would be up to the reader in garmin_fit.cc to find documentation on
the new fields or reverse engineer their content and put them into our
common internal data structure.
2) If some sensible place for that data exists in Waypoint (for example,
maybe orientation fits into 'heading' and maybe it doesn't) great, but if
not, members would have to be added.  It's not set in stone, but in general
if we're having to add something there, that means it's probably a
domain-specific concept and probably a poor fit for us.  We stretch beyond
location a little bit for common concepts like temperature or heartbeat,
but I don't know that we want to do pitch, yaw, roll for 3D camera
orientation.
3) The writers (gpx.cc, kml.cc, whatever) would have to be modified to do
something sensible with that data.

sendbits, 2015-11-18:
That’s helpful.

 

Does the license permit us to do a separate branch / fork so we can add that capability internally even if it doesn’t make its way back to the main branch?

robertlipe, 2015-11-18:
GPSBabel is licensed under GPLv2.  You received a copy of the license when
you received the code.
https://github.com/gpsbabel/gpsbabel/blob/master/COPYING

Who is "we"?

sendbits, 2015-11-18:
I’m happy to better describe we, however, is there a better email that won’t make its way to the public archives?

 

https://github.com/gpsbabel/gpsbabel/blob/master/COPYING

robertlipe, 2015-11-18:
robertlipe on the domain that hosts the project.  It's in many of the
source files, such as kml.cc.  Humans should leave out the +source tag
which is there exactly to identify web scrapers.

robertlipe, 2016-01-03:
Fixed and released in 1.5.3.

closed 2016-01-03 by robertlipe