GPSBabel/gpsbabel #16: Push a 1.5.3?

cemeyer, 2015-12-16:
It's been 11 months since 1.5.2 was released (January).  Maybe we should push a 1.5.3.

1.5.2 is tripping over .fit files from my Garmin ForeRunner 230 with "fit: Bad field size in data message".  I suspect https://github.com/gpsbabel/gpsbabel/commit/d6ec2fedf65c54151baa6b4bf26528be5fa715ca or https://github.com/gpsbabel/gpsbabel/commit/ff0dea7afa0f6a02461a7e4dabdc4cf43ca91a7b may resolve the problem.  Neither is present in 1.5.2.

cemeyer, 2015-12-16:
(Yep, the file is sucessfully parsed with git master.)

tsteven4, 2015-12-16:
> (Yep, the file is successfully parsed with git master.)

that is probably the most common complaint about 1.5.2.

robertlipe, 2015-12-17:
I hope/plan to push a 1.5.3 over the holidays, though I don't think I said
that out loud.  As I'm currently on the "wrong" continent, but returning
about twelve hours from now, but have pressing personal matters upon my
return, it'll probably be after Christmas and perhaps even closer to New
Years.

If any of you have pressing changes to be merged (I know the Garmin USB
encoding thing is still in flight - it's not contentious, I just haven't
done it...) please bang a gong soon(ish).

Given that this should be just a minor maintenance release, I'll bump the
minor number, push a new doc set, and won't bother with a beta because
respinning a beta to make it not a beta isn't actually any easier than
spinning another version.

Obrigada. (LANG=pt_BR)
RJL

cemeyer, 2015-12-17:
Cool, works for me.

alanbur, 2016-01-02:
After the move to GitHub where is the correct place to log bugs? According to the website it's the still developers mailing list. Either way I sent a mail about a bug with EXIF writing to the mailing list, it would be good if that could be fixed in 1.5.3 as it looks like it's just a 1-line typo in exif.cc - see http://sourceforge.net/p/gpsbabel/mailman/gpsbabel-code/thread/5687EE30.4040108%40gmail.com/#msg34734204

cemeyer, 2016-01-02:
A new Github Issue or the mailing list, but certainly not on an unrelated issue (this one).

alanbur, 2016-01-02:
I asked here because of the comment above asking if there were any pressing issues, and it not being clear where bugs now need to go. I've logged https://github.com/gpsbabel/gpsbabel/issues/19

closed 2016-01-19 by robertlipe