GPSBabel/gpsbabel #25: libusb-1.0 support

bzed, 2016-01-09:
Hi,

libusb upstream switched to a new API (libusb-1.0 - http://libusb.org/static/api-1.0/) - please support or migrate to it as it will become the default libusb implementation in most Linux distrubutions soon(ish).

Thanks a lot,
Bernd

robertlipe, 2016-01-10:
The referenced site won't load for me, but we abandoned libusb head when
they abandoned cross platform and went Linux only. We would love to
jettison our local patches if they care about more than one os again. If
they've reconsidered, we can revisit.

Patches ... That don't break non Linux ... Welcome.

bzed, 2016-01-10:
Maybe http://libusb.info/ works for you.libusb-1.0 supports Linux, OS X, Windows, Windows CE, Android, OpenBSD/NetBSD, Haiku....

robertlipe, 2016-01-10:
Ah, they saw the light and added the other OSes back.

The USB layer is very expensive to change as it requires testing a large
variety of hardware. A Garmin 60cs and a 60csx use a different path of bulk
vs interrupt pipe paths, for example.  Garmin and Delorme are, I think, our
only libusb users. Since Delorme added mass storage and GPX support in
later firmware levels (and the yellow model is basically forgotten) I could
be convinced to drop delbin, but if you have changes for Garmin, be
prepared for pain.

bzed, 2016-01-10:
I don't even have usb hardware to test gpsbabel with. I'm afraid all I can do is to drop libusb from the build-dependencies in Debian at some point. That will take some months for sure - but the day will come. If you or somebody else is interested to have working usb support in the next Debian release, please implement support for the new libusb. I would prefer to send patches, but without being able to test changes that doesn't make much sense.

robertlipe, 2016-01-19:
Not to dwell on another project's history or to turn a feature request into a discussion forum, but for those following GPSBabel's history with libusb, I stumbled across an interesting talk by Peter Stuge, the maintainer of one of the six libusb strains.

I'm actually taking a couple of points from his talk to heart, but the history of libusb is well captured here and gives some background why I've not really been chasing their development head.

https://www.youtube.com/watch?v=9gJTsipu37g   #  Watch at 1.5 or 2x. Auto-captions choke on some of the tech terms.
Discussions and tl;dw at
https://www.reddit.com/r/linux/comments/3z9rlk/talk_libusb_maintainer_fail/
https://www.reddit.com/r/programming/comments/41hsrt/libusb_maintainer_fail_how_i_failed_to_run_an/

ra1fh, 2016-01-26:
Is Debian going to drop the legacy libusb-0.1 library without replacement or is the plan to provide libusb-compat-0.1 instead?

The libusb-compat-0.1 library implements the 0.1 API on top of libusb-1.0. With that, there would be no immediate need to convince all libusb-0.1 users to switch to the new API.

bzed, 2016-01-26:
According to the libusb maintainer there are various issues with libusb-compat-0.1, so it won't be packaged and the old one will be dropped.

PR #396 (2019-11-14): Replace libusb 0.1 usage with libusb 1.0
Replace our modified libusb 0.1.12 with libusb 1.0.22 on macos.  On macos we statically link to our inclcuded libusb.
Use libusb 1.0 instead of libusb 0.1 on linux.  On linux we dynamically link to a distribution provided libusb, or optionally build without libusb support.

This fixes #381 
This resolves #25 as requested (albeit 3.5 years later)
closed 2019-11-14 by tsteven4