GPSBabel/gpsbabel #32: Drag and Drop in v1.5.3 on OS/X fails  '/.file/id=6571367.13131907' for read. Error was 'Not a directory'.

robertlipe, 2016-01-19:
Stan Glaser reported:

I tried dragging a file named "2015-12-20 11.01.03 Day.gpx” to the Input section for File Name(s) and it came up with “/.file/id=6571367.13131907” in the box. Of course I tried clicking on “OK” but it just came up with an error of

> gpsbabel -w -t -i gpx -f /.file/id=6571367.13131907 -x position,distance=3F -o gpx,gpxver=1.0 -F /.file/id=6571367.13131907
> Cannot open '/.file/id=6571367.13131907' for read. Error was 'Not a directory'.

robertlipe, 2016-01-19:
I confrmed and chased the bug into Qt, described at https://bugreports.qt.io/browse/QTBUG-40449

"Oh, just upgrade the version of Qt on the build system!"
Well, no.  Now we run into the symptoms described at https://github.com/Cockatrice/Cockatrice/issues/1547 and in more detail in the thread at http://lists.qt-project.org/pipermail/development/2015-September/023286.html

Now I have to find a solution that doesn't end up with our distribution referencing frameworks that exist only on my development machine. Ugh.

robertlipe, 2016-01-19:
Gross. Macdeployqt doesn't seem to pick up the frameworks so the dmg generated by macdeployqt are totally busted. This is all fragile enough that I'm pushing the release back a few days.

On the bright side, the button centering is less hideously deformed on my 2012 iMac, but I need to spot-check it on a retina MBP.

gpsbabel, 2017-01-26:
Fixed in 1.5.4. Thanx for the report.

closed 2017-01-26 by gpsbabel