GPSBabel/gpsbabel #36: Error when exporting GPS xml to humminbird *.ht

Beaulieu80, 2016-02-25:
I can export all my HT files into a single GPS xml file. Not working from GPS to HT.

robertlipe, 2016-02-25:
There is nowhere near enough information here for this to be an actionable bugreport. Without the input files, the commands you used, the versions involved, there's no way this is going to advance. Please reopen if you have a reproducible case that doesn't work as expected.

closed 2016-02-25 by robertlipe

Beaulieu80, 2016-02-25:
I try to attach a GPS XML but the form cannot accept ZIP file, where can I send you the files to simulate the bug?

tsteven4, 2016-02-26:
I think zip files < 10MB will work if you hit the select link and use 
the resulting gui to select the file.

If you can get that to work you can try to email 
gpsbabel-misc@lists.sourceforge.net

In any event please make sure you answer all of Robert's questions
> Without the input files, the commands you used, the versions involved

Beaulieu80, 2016-02-26:
Here the GPS file that I try to export into *.HT files

when I use humminbird PC to view the HT file it does not work since the HT is not well build

thank you

Beaulieu80, 2016-02-26:
I have sent email with the ZIP file.   "Sonar Humminbird tracks"

1) I have exported all my 100+ *.HT files into a single GPS file. 
2) Split that GPS xml file so it can be grouped by 20K tracks each
3) Try to export each GPS into a HT file but it does not work, no error been displayed
4) When trying to read the HT with Humminbird PC it does not work (not in good format)

my goal is the allow me to merge all HT in a way to maximize each file to is full capacity allowed.

I see under options that we can 'MERGE' but with too much creteria.

thank you

tsteven4, 2016-02-26:
We still don't have your zip file.
We still don't know what version of gpsbabel you were running.
We still don't know the gpsbabel command line you were using.

robertlipe, 2016-02-27:
We also don't know if exporting ONE file works. It's unsurprising that any
GPS won't read and display 200000 tracks at once.

Beaulieu80, 2016-02-27:
The Humminbird sonar can read a maximum of 20k on each HT file.  

GPSBabel 1.5.3 
files here:  http://1drv.ms/1TJEWes

[image]

thank you