GPSBabel/gpsbabel #38: Accented characters not accepted in file names

shred, 2016-03-25:
gpsbabel does not seem to accept accented characters in file names. For example, I am trying to convert the "Kaktushöhle.kml" to gpx:

```
$ gpsbabel -i kml -f Kaktushöhle.kml -o gpx -F Kaktushöhle.gpx
Cannot open 'Kaktushöhle.kml' for read.  Error was 'No such file or directory'.
```

I can use STDIN to read the file. However, this call generates a file called "Kaktushhle.gpx" (without the "ö" umlaut):

```
$ gpsbabel -i kml -f - -o gpx -F Kaktushöhle.gpx < Kaktushöhle.kml
```

The only way is to use STDIN and STDOUT:

```
$ gpsbabel -i kml -f - -o gpx -F - > Kaktushöhle.gpx < Kaktushöhle.kml
```

It seems that gpsbabel strips all non-ascii characters from the file name before attempting to open it.

Tested on gpsbabel-1.5.2-3.fc23.x86_64 on Fedora 23. LANG is de_DE.utf8.

tsteven4, 2016-03-25:
Can you try this with the current development tree?  I think it has been 
fixed (specifically, the fix was in 
https://github.com/gpsbabel/gpsbabel/commit/a09285176af0079c964baf78a69b0cd5ab648ba5)

shred, 2016-03-25:
Yes, it works with the master branch. Sorry, I should have tried that first before opening a bug.

closed 2016-03-25 by shred