GPSBabel/gpsbabel #43: No cyrillic characters in garmin gpi output

squromiv, 2016-04-08:
The problem, which is described [there](https://sourceforge.net/p/gpsbabel/mailman/message/31508200/), is still not solved.

globalbus, 2016-12-22:
It's because support for codepages was very small right now. There is only support for ASCII, UTF8 and CP1252 codepages. In older versions there is much more codepages, but they are not exists in this repository.

You need to copy codepages from here https://github.com/twpayne/gpsbabel-flytec/tree/master/cet
and patch cet_util.cc to add other codepages (you're probably looking for cp1251 support). Everything else still working well with "writecodec" parameter. Only custom encoding for category string need patching in garmin_gpi.cc

Anyway, this issue is still alive? I can do fork and patch this things. I already add suport for cp1250 for my own.

robertlipe, 2016-12-30:
> gpsbabel-flytec/tree/master/cet

The 'writecodec' option was added to the Garmin GPI writer in 2014 to
address this for that format.

The direction of GPSBabel is away from doing our own encoding
transformations. We got some mileage out of the original implementation,
but our direction now is that everything uses a consistent character set
(UTF) internally and that we convert on output if we must.

I think the Russian "cp1251" encoding is probably spelled "Windows-1251"
according to http://doc.qt.io/qt-5/qtextcodec.html

WIthout specifics (a GPX input file, the precise command issued, the
details of what's reading that GPI file, a GPI file written by something
else for that same GPX file that behaves "right",  the version of
everything involved, etc.) this won't get very far. Debugging character set
encoding issues - especially on reverse-engineered formats is a very
meticulous exercise. That's likely why a one liner "it's still broken" that
quoted a two year old conversation that went cold - but had the suggestion
from the final post implemented anyway - didn't get much traction.

RJL

globalbus, 2017-01-17:
GPI format is still widely used in all Garmin devices. It's mouch more faster than GPX, small devices with slow processors have a slow XML parsing.
I can make a patch, I already made it for own usage. How to participate? Fork and pull request to master?

robertlipe, 2017-01-18:
Well engineered patches that are consistent with our direction (we're OUT
of the encoding business) are usually welcome. This conversation is hard to
follow since it's been fragmented over four years, but we're not putting
CET infrastructure back in.

iget-master, 2017-04-27:
Hi, I'm having problems to set category name using accents:

```
gpsbabel -i gpx -f Vinícolas.gpx -o garmin_gpi,category="Vinícolas" -F "Roteiro.gpi"
```

This create a category with wrong encoding on the gps: "Vinï½colas"

How can I fix it?

closed 2020-05-19 by gpsbabel