GPSBabel/gpsbabel #48: linux-gui will not compile on qt-5.x > 5.5

juanitotc, 2016-07-20:
Due to changes in webkit/webengine after qt-5.5, the linux-gui will not compile:

$ make linux-gui
Project ERROR: Unknown module(s) in QT: webkit webkitwidgets

robertlipe, 2016-07-20:
Seems that the Linux OSVs should have reported and addressed this when they upgraded. Please see if this works for you.

[fixqt56.txt](https://github.com/gpsbabel/gpsbabel/files/374383/fixqt56.txt)

tsteven4, 2016-07-20:
linux-gui from master builds fine for me without any changes under 
Centos7 with Qt 5.6.1 from the centos repo.

robertlipe, 2016-07-20:
I think it's actually 5.7 that's the breaking point.  5.4 added
QWebEngineView (blink) and 5.7 removed the WebKit renderer. I'm not totally
clear where the exact breaking point is.

http://doc.qt.io/qt-5/qtwebenginewidgets-qtwebkitportingguide.html

Maybe we raise our floor to 5.4 (?) and avoid the #ifdef this clutter. We
use such a small part of WebKit that it looks like it's just nuisance
renaming for us.

RJL

tsteven4, 2016-07-20:
travis is currently using 5.2.1

tsteven4, 2016-07-21:
https://wiki.qt.io/New_Features_in_Qt_5.6 :

>
>     Removed Modules
>
> With Qt 5.6 the following modules are no longer part of the release 
> packages, but users can still build them from source:
>
>   * Qt WebKit
>   * Qt Declarative (Qt Quick 1)
>

juanitotc, 2016-07-21:
The patch applies cleanly, but using qt-5.7:

$ make linux-gui
cd gui ; qmake app.pro && make
Info: creating stash file /usr/src/gpsbabel/gui/.qmake.stash
Project ERROR: Unknown module(s) in QT: webenginenewwidgets
Makefile:390: recipe for target 'gui' failed
make: *** [gui] Error 3

$ ls /usr/local/lib/libQt5WebEngineWidgets*
/usr/local/lib/libQt5WebEngineWidgets.la        /usr/local/lib/libQt5WebEngineWidgets.so.5
/usr/local/lib/libQt5WebEngineWidgets.prl       /usr/local/lib/libQt5WebEngineWidgets.so.5.7
/usr/local/lib/libQt5WebEngineWidgets.so        /usr/local/lib/libQt5WebEngineWidgets.so.5.7.0

changing:
+    QT += webenginenewwidgets -> +    QT += webenginewidgets

..improves things, but:

$ make linux-gui
...
In file included from gmapdlg.h:31:0,
                 from gmapdlg.cc:27:
map.h:60:1: error: expected class-name before '{' token
 {
 ^
gmapdlg.cc: In constructor 'GMapDialog::GMapDialog(QWidget*, const QString&, QPlainTextEdit*)':
gmapdlg.cc:145:28: error: no matching function for call to 'QHBoxLayout::addWidget(Map*&)'
   lay->addWidget(mapWidget_);

..with several more errors after that.

juanitotc, 2016-07-21:
BTW, why does gpsbabel, a cli app, need a dep on libQt5Core?

tsteven4, 2016-07-21:
We use a bunch of classes from QtCore, e.g. QString, QXmlStreamReader, 
QXmlStreamWriter, ... even though we haven't really wanted to be a 
QCoreApplication.

It looks like qtwebengine hasn't made it to RHEL7/Centos 7/... which is 
using Qt 5.6.1 with Qtwebkit.  There is work in Fedora but it hasn't 
trickled down yet.

So a decision on using qtwebkit vs. qtwebengine can't necessarily be 
based on the qt version.

robertlipe, 2016-07-24:
Wait, let me see if I have this right:

WebEngine was added - and always intended to replace WebKit - in Qt 5.4.
WebKit was removed (though still optionally compilable) in 5.6 because it
no longer met the portability and security requirements for Qt.
RHEL is shipping a 5.6 version of Qt without WebEngine, but WITH WebKit,
making their Qt pretty much incompatible with everyone else's.

RedHat's other OS, does ship with webengine, but only as an option.
https://fedoraproject.org/wiki/Changes/QtWebEngine

Do they provide any flag in qglobal.h that announces this so that an app
can test whether they're using such a bizarre combination?  I'd rather not
rely on configure tests for this. In fact, I'm about out of love with
configure itself.

I'm admittedly caught a bit off heel that such a major feature was removed
in 5.6 and am annoyed that we may have to support two paths for a long
time. Our use of the embedded web view is quite pedestrian so it's not like
it's even a major annoyance.

Do any of our Linux experts have advice on how other portable apps are
dealing with this?  In the absence of a sane way to detect which path is
really available, I may just disable this feature (it's really not used
that much anywhere...) on Linux and let the distro people work it out when
they make their builds. Then we can rely on a version-based test to handle
the case for Mac and Windows builders using the current official Qt
configuration to allow spanning the 5.6 boundary where QtWebKit is gone.

juanitotc, 2016-07-25:
Why not rely on the version based test for linux as well - just because rhel/fedora do things one way doesn't mean it should become difficult for everybody.

robertlipe, 2016-07-25:
Right. Not special casing RHEL is exactly the problem. 5.6 for everything
except RH (maybe other distros, I don't know...) uses WebEngine, not
WebKit.

If Version < 5.4 use WebKit
(in the middle here, you can choose either on a sane build of Qt)
If Version >= 5.6 must use WebEngine. Unless you're RHEL. Or maybe others.

I do have a better patch in progress to support WebEngine. I have
everything working except the WebChannel call to a C++ method.  Since
that's a debug method that I've never used, I'm considering just whacking
that call and moving on. The signals and slots are working and that's the
important part.

RJL

gpsbabel, 2016-07-26:
I went through the gyrations to get an even vaguely modern compilation
environment on on Centos 7. Frankly, I'm horrified to end up with a C++
from 2010.

RHEL/Centos jumped through so many hoops to keep the software tied to the
previous decade that, well, they'll simply be happier with software from
the last decade. Simply getting to Qt > 4.8 is an ordeal. I think that
GPSBabel's best stance will be to let RHEL/Centos to ship and support
whatever version is time appropriate for them and for us to forge on in
modern times.  The silence of Linux representatives on this topic mean it's
time to just leave them to ship whatever they're going to ship since
they're clearly not interested in supporting modern work. Debian/Redhat
devs can port GPSBabel fixes/enhancements back for the next six years; I
just can't.

I'm soon going to move our floor to C++11 (it's been five years since an
international standard organization has signed off...and they don't move
exactly fast) and supporting modern Qt.

Soon, GPSBabel will allow C++11 and depend on tools of that vintage. If
your environment traps you in a pre-2010 world, please use GPSBabel from
2010 and don't send resulting problems our way.

This sill impact our Windows and Mac users (over 98% of our users) exactly
zero.

Signed,
Sometimes it sucks to be the bad guy.

RJL

tsteven4, 2017-01-02:
https://github.com/gpsbabel/gpsbabel/commit/25e9f486c3ec1b143b1676ef10123c1ebad18473 broke the gui on CentOS 7 due to the lack of webenginewidgets with 5.6.1 for that platform.  As a suggestion can we change app.pro to use qtHaveModule instead of testing QT_MINOR_VERSION?  This does work on CentOS 7, but I haven't tested it elsewhere.

diff --git a/gui/app.pro b/gui/app.pro
index c512507..7cea1f3 100755
--- a/gui/app.pro
+++ b/gui/app.pro
@@ -16,7 +16,8 @@ QT += core \
       network \
       xml \
 
-greaterThan(QT_MINOR_VERSION, 5) {
+#greaterThan(QT_MINOR_VERSION, 5) {
+qtHaveModule(webenginewidgets) {
   QT += webenginewidgets
   DEFINES += HAVE_WEBENGINE
 } else {

gpsbabel, 2017-01-02:
> 25e9f48

Yes, please. I was pretty sure that was going to blow up somewhere. I've
not cut the master tarball yet, so if you don't get to it, I will today.

I can't believe that OSVs are clinging to such a weird configuration of Qt.
Putting back in a deprecated model and NOT offering the replacement that's
been there for years is really a pain.

tsteven4, 2017-01-02:
The change is in, but I wasn't able to test it on any distribution that 
used the new webenginewidgets.  Ubuntu 16.04 is still using webkit.

robertlipe, 2017-01-03:
Thank you, Steven.

tsteven4, 2018-11-07:
I was able to successfully test this on ubuntu bionic with the ubuntu supplied webenginewidgits (package qtwebengine5-dev).  It also builds with webkit under bionic if package qtwebengine5-dev isn't installed and package libqt5webkit5-dev is installed. (see e.g. https://travis-ci.org/tsteven4/gpsbabel/jobs/450541573).

closed 2018-11-07 by tsteven4