GPSBabel/gpsbabel #54: FIT to GPX conversion ignores base time and provides times in 1989

ajmas, 2016-09-30:
When I try converting a Garmin FIT file to GPX I find all the times in the file are in 1989, rather than the date recorded.

Looking at the specification, it would seem that we should probably be reading the time_created value in the file_id message and adding that to the timestamp value, if timestamp is less than 0x10000000.

Looking at the 'D00001309 FIT File Types Description Rev 2.1' in the FitSDKRelease_20.10.01.

gpsbabel, 2016-12-30:
The license for the FIT SDK isn't compatible with open source projects like ours. (At least it wasn't when I last looked.) We have a fair number of .fit users, so it seems odd that this is totally broken.  If you can provide a detailed textual description of how to make this better, why your fit files are different than the ones we have, perhaps referencing https://github.com/gpsbabel/gpsbabel/blob/master/garmin_fit.cc (hint: search for 'timestamp') perhaps someone can take a swing at improving it.

dark, 2017-06-28:
This was fixed by PR #75 too.

closed 2017-06-28 by gpsbabel