GPSBabel/gpsbabel #55: kml: There were more gx:coord elements than the number of when elements.

infernix, 2016-10-02:
Recently, exporting kml from google location history returns the error in the subject. Older kml files convert fine with the latest master, but any new export fails.

If needed I can provide an example file, but reproducing is as simple as fetching any recent Google location history kml and running it through gpsbabel.

robertlipe, 2016-10-04:
I just looked at Location History's KML and I think I agree with GPSBabel's
error as there are no <when> tags.  Using gx:coord without corresponding
<when> array is against the spirit of gx:Track - the whole point of
gx:Track was to associate location with time.  It may not be technically
forbidden by the schema, but that's against the spirit of gx:Track as that
effectively reduces it to a LineString.    Please submit this at
https://productforums.google.com/forum/#!forum/maps as it means Location
History isn't giving you back the time with your location.

infernix, 2016-10-05:
For reference, posted here:  https://productforums.google.com/forum/#!topic/maps/iI26ovonPIY;context-place=forum/maps - let's see what the response will be. 

Thanks.

gpsbabel, 2016-12-30:
Changed to "enhancement request" to support a spelling of KML that was against the wishes of the authors of KML. This should be fixed by Google's KML writer, not our reader.

cbrace, 2018-10-29:
Hi @infernix 
I was wondering whether you ever found a solution for this. Just now I using gspbabel to convert a large collection of KML files created using an Android app called **Geo Tracker**. For some 300 of these tracks from the last couple of years, gspbabel returned this error message and the output file is 0 bytes long. I took a quick look at one of the offending files:

> grep -c "gx:coord" ../2018-10-21_Berg_-_Schin_op_Geul_10_km.kml 
> 936
> $ grep -c "when" ../2018-10-21_Berg_-_Schin_op_Geul_10_km.kml 
> 937 

Looking at the file, I see there is an extra `<when>` tag in the header:
> `<TimeStamp>`
>      `<when>`2018-10-21T11:26:16.512Z`</when>`
>    `</TimeStamp>`

So this would account for the **one extra** `<when>` tag. I have really no idea what is going further wrong here. 

I looked at the Google Forum topic, but I don't see any kind of solution. Did you come up with anything in terms of a fix or workaround?

infernix, 2018-10-29:
I don't remember but I don't think so. Rarely use geolocation tagging anymore so not much help.

gpsbabel, 2018-10-29:
You really can't just count XML tags like that without context.  A when
within a TimeStamp is legal in certain contextx of KML, I'm pretty sure.

The bug originally being discussed is Google Location History creating
bogus KML. It seems you're using a completely unrelated writer, an Android
app. Unsurprisingly, the KML we'll see from two unrelated generators will
have different shapes to us.

Even a strict count isn't really enough to go on. You could have the same
number, but be in different gx:Track enclosures, or with nearby comment
start and end or in ditterent xgx:Strand and still be wrong, for example.
We've not changed anything material in our KML reader in a long time, but
please provide a full reproduction environment in another bugreport. Since
this only manivests in some of the files, and the app you're using is also
the one used by that person that originally tagged onto that bugreport in
Location Hisoty, be prepared to involved the GeoTracker group, too. There
are lots of cases, especially in mobile, where an app can get prematurely
closed, or lose a record and then end up writing mismatched contents in the
parallel arrays of the gx:track components. Any "solution" you can find
that applies to Google's location writer is highly unlikely to apply here.

Offhand, at least in the file that other person reported, the
non-interleaving of <gx:coord> and <when> and having gx:MultiTrack not be a
child of <Placemark> (or pershaps some other container feature, is mighty
suspicious. Pull that file into earth (with "complain about KML" turned on,
save it back out, and  diff that to see if this meets Earth's best
practices for ordering and grouping before we chase it into KML spec snf
yty to validate it. It may be an ordering artifact of the gx:coords" and
<here> tags. and their relainshop to each other..

But all this needs to be in a different bugreport, with a test case, with
versions, with the full command run, and without flogging Google Location
History, please. It needs to be a standalong reproducible test
case/bugreport . That file is definitely not how I remember them looking,
but I'm admittedly in bed and looking at on a fone. (Yes, which is bad for
me.)

RJL

robertlipe, 2018-10-29:
Once hacking away all the atom stuff, the file from that bugreport still
fails to validate - in the very area that's causing us grief:

/Users/robertlipe/Downloads/23_Mar_2017_11_50_16_am.kml:30: element
TimeSpan: Schemas validity error : Element '{
http://www.opengis.net/kml/2.2}TimeSpan': This element is not expected.
Expected is one of ( {
http://www.opengis.net/kml/2.2}AbstractFeatureSimpleExtensionGroup, {
http://www.google.com/kml/ext/2.2}balloonVisibility, {
http://www.google.com/kml/ext/2.2}rank, {
http://www.opengis.net/kml/2.2}AbstractGeometryGroup, {
http://www.opengis.net/kml/2.2}MultiGeometry, {
http://www.opengis.net/kml/2.2}Point, {
http://www.opengis.net/kml/2.2}LineString, {
http://www.opengis.net/kml/2.2}LinearRing, {
http://www.opengis.net/kml/2.2}Polygon, {
http://www.opengis.net/kml/2.2}Model ).

If I sort that file so that gx:Track starts with the <when> tags (which we
read and count to know how many vertices there are to insert geometries)
followed by the gx:coord block, it still doesn't validate, but GPSBabel can
read it.

Its really looks like that program is creating something that looks like
KML, but actually formally isn't.

RJL

tsteven4, 2018-10-31:
There are multiple schema violations in the file 
23_Mar_2017_11_50_16_am.kml.  The program that created 
23_Mar_2017_11_50_16_am.kml, which was reported to be GEO Track on a 
Samsung Galaxy Note 3 phone, is producing output that violates the kml 
schema.

If the elements are reordered according to the schema all these 
violations can be fixed, as in the attached file.  gpsbabel works fine 
when the schema violations are fixed.

> $  gpsbabel -i kml -o kml -f 23_Mar_2017_11_50_16_am.kml -x 
> transform,rte=trk -x nuketypes,waypoints,tracks -x simplify,error=10m 
> -F Cranbourne-to-Mordialloc_2017-03-23_rte-draft.kml
> kml: There were more gx:coord elements than the number of when elements.
> $  gpsbabel -i kml -o kml -f 23_Mar_2017_11_50_16_am_valid.kml -x 
> transform,rte=trk -x nuketypes,waypoints,tracks -x simplify,error=10m 
> -F Cranbourne-to-Mordialloc_2017-03-23_rte-draft.kml

tsteven4, 2018-11-02:
The Android GEO tracker sample provided had multiple schema violations that resulted in gpsbabel complaining "There were more coord elements than the number of when elements."  While the file produced by Android GEO Tracker violated the schema for Google extensions to OGC KML 2.2, I have modified the gsbabel kml reader to work with tracks were the when and coord children are not ordered as required (an equal number of when elements must preceded the same number of coord elements).  It is still required that a track contains the same number of when and coord elements, we will just be more tolerant of the order of their appearance in the track.  Pre-release code is available at https://github.com/gpsbabel/gpsbabel/releases.  These changes were incorporated in #264.

Originally this issue was opened for kml files created from google location history.  However, no samples were provided so it is unclear what that issue was and if it is still a probem.

cbrace, 2018-11-04:
@tsteven4 Thank for looking into this. The FreeBSD port is currently at v1.5.4. When v1.5.5 or higher shows up I will test this again.

gpsbabel, 2020-05-19:
We've since addressed the issue of reading points with mismatched time/point counts.
The issue of Google's KML export writing broken KML is, unfortunately, not up to us to solve.
Closing as we've taken all the actions here we can.

closed 2020-05-19 by gpsbabel