GPSBabel/gpsbabel #58: Synthesize speed GPX

drnextgis, 2016-11-23:
Trying to make gpx file with speed values:

```
gpsbabel -t -i gpx -f 1037084.gpx -x track,speed -o gpx -F output.gpx
```

But resulting file doesn't contain speed info. Is it limitation of GPX format and I have to use another format?

tsteven4, 2016-11-23:
This is a limitation of gpx version 1.1.   It is not a limitation of gpx 
version 1.0.  The speed element was removed from gpx version 1.1.

Try

||gpsbabel -|i gpx -f 1037084.gpx -x track,speed -o gpx,gpxver=1.0 -F 
output.gpx|

> ~/work/gpsbabel% gpsbabel -i gpx -f x.gpx -x track,speed -o 
> gpx,gpxver=1.0 -F -
> <?xml version="1.0" encoding="UTF-8"?>
> <gpx version="1.0" creator="GPSBabel - http://www.gpsbabel.org" 
> xmlns="http://www.topografix.com/GPX/1/0">
>   <time>2016-11-23T23:08:44.863Z</time>
>   <bounds minlat="30.045516667" minlon="-91.610566667" 
> maxlat="30.062783333" maxlon="-91.589750000"/>
>   <trk>
>     <name>meridian</name>
>     <trkseg>
>       <trkpt lat="30.062183333" lon="-91.610350000">
>         <ele>1.000000</ele>
>         <time>2002-05-25T17:06:21.250Z</time>
>         <speed>0.000000</speed>
>       </trkpt>
>       <trkpt lat="30.062783333" lon="-91.610566667">
>         <ele>0.000000</ele>
>         <time>2002-05-25T17:09:55.190Z</time>
>         <speed>0.326999</speed>
>       </trkpt>

drnextgis, 2016-11-24:
Thanks, it works! But it looks like [docs](https://www.gpsbabel.org/htmldoc-development/fmt_gpx.html) is outdated:

```
This option specifies the version of the GPX specification to use for output.
The default version is 1.0.
```

Also it would be useful to have information about speed element in this document.

gpsbabel, 2016-12-05:
Thanx. Updated to say The default version is 1.0 unless one or more of the
input files
is GPX 1.1, then it's 1.1

THis really is a very mess subject, I'm afraid.

closed 2016-12-30 by gpsbabel