GPSBabel/gpsbabel #62: Build failure with Clang4

OliverHe, 2017-01-23:
Build fails on FreeBSD with Clang 4.0 with following error:
```
bushnell.cc:138:36: error: ordered comparison between pointer and zero ('const char *' and 'int')
  for (t = bushnell_icons; t->icon > 0; t++) {
                           ~~~~~~~ ^ ~
bushnell.cc:150:36: error: ordered comparison between pointer and zero ('const char *' and 'int')
  for (t = bushnell_icons; t->icon > 0; t++) {
                           ~~~~~~~ ^ ~
```

Full build log [here](http://sprunge.us/MYYK)

I'll look to patch the FreeBSD port to resolve this. I created this issue to ensure upstream knows about it.

robertlipe, 2017-01-23:
If t->icon != nullptr works for you, I'd pre-approve that.

If you care about GPSBabel on FreeBSD, please set up a daily build that we
can integrate into github submit so we find about about this kind of thing
at commit time instead of release time.

OliverHe, 2017-01-23:
Clang 4 is [being tested](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=216008) at the moment, so it didn't break at release time.

Thanks for the tip, C++ is quite new to me, so I'll approach patching carefully.

robertlipe, 2017-01-23:
Please try  t->icon != nullptr. What we (I?) did here is a bit weird and
that's probably more right.  Looks like it's replacing two occurrences of
"> 0" with "!= nullptr" so I'd hope it's fairly painless, but I know how
automated build systems can be.

Technically, "nullptr" is C++11, but it's been widely accepted for ten
years or so, so it should Just Work. If it doesn't (sigh) use "NULL"
instead of "nullptr".

OliverHe, 2017-01-24:
nullptr appears to work.
Before patch:
```
$ gpsbabel -i bushnell -f bushnell.wpt -o gpx -F -
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" creator="GPSBabel - http://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/0">
  <time>2017-01-24T13:31:24.671Z</time>
  <bounds minlat="55.296300000" minlon="-133.243100000" maxlat="55.296300000" maxlon="-133.243100000"/>
  <wpt lat="55.296300000" lon="-133.243100000">
    <name>Waterfall Resort</name>
    <cmt>Waterfall Resort</cmt>
    <desc>Waterfall Resort</desc>
    <sym>Fishing</sym>
  </wpt>
</gpx>
```
After patch and build on FreeBSD with Clang 4:
```
$ gpsbabel -i bushnell -f bushnell.wpt -o gpx -F -
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" creator="GPSBabel - http://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/0">
  <time>2017-01-24T13:32:20.407Z</time>
  <bounds minlat="55.296300000" minlon="-133.243100000" maxlat="55.296300000" maxlon="-133.243100000"/>
  <wpt lat="55.296300000" lon="-133.243100000">
    <name>Waterfall Resort</name>
    <cmt>Waterfall Resort</cmt>
    <desc>Waterfall Resort</desc>
    <sym>Fishing</sym>
  </wpt>
</gpx>
```
[Full build log for those interested](https://secure.ohos.nl/downloads/gpsbabel-1.5.3.log)
I'm now doing runs for all current versions of FreeBSD, both i386 and amd64, if all builds well, I'll create a pull request.

PR #64 (2017-01-25): Use nullptr instead of comparing to 0
Fixes #62
closed 2017-01-25 by gpsbabel