GPSBabel/gpsbabel #66: V2.0 no success Unknown subtype (17)

guenter1952, 2017-02-14:
All my  V2 files are not running:

GPSBabel Version: 1.5.4 
ovl: header = DOMGVCRD Ovlfile V2.0: 
ovl: map name len      144 (0x0090)
ovl: name = Top. Karte 1:50.000 Bayern-S?d 
------------------------------------ 0xa9
ovl: entry type          5 (0x0005)
ovl: entry group         1 (0x0001)
ovl: entry zoom          1 (0x0001)
ovl: entry subtype      17 (0x0011)
ggv_bin: Unknown subtype (17)

ra1fh, 2017-02-14:
Could you attach a sample file?

This importer is reverse engineered, so it's certainly not perfect, although it did process all V2.0 files I could find online at the time I wrote it.

ra1fh, 2017-02-14:
Ok, the subtype decoding is wrong. Here is a preliminary patch:
```
diff --git a/ggv_bin.cc b/ggv_bin.cc
index 9a616cfe..6c87e2dc 100644
--- a/ggv_bin.cc
+++ b/ggv_bin.cc
@@ -143,17 +143,9 @@ ggv_bin_read_v2(QDataStream& stream)
     ggv_bin_read16(stream, "entry zoom");
     entry_subtype = ggv_bin_read16(stream, "entry subtype");
 
-    switch (entry_subtype) {
-    case 0x01:
-      // no data following
-      break;
-    case 0x10:
-      // text with 32 bit length field follows
+    if (entry_subtype != 1) {
       ggv_bin_read_text32(stream, buf, "text len");
       track_name = QString::fromLatin1(buf.constData()).simplified();
-      break;
-    default:
-      fatal(MYNAME ": Unknown subtype (%hu)\n", entry_subtype);
     }
```

I will submit a pull request after testing this a bit.

It would still be nice to get a sample file to make sure the rest of the file decodes correctly.

guenter1952, 2017-02-15:
Hi, thanks for your fast reply and your work. Meanwhile I have tested some more. These are files from one friend, from another guy the files V2 are ok. I found a way around the problem by converting the bad V2-files to V4 within Top10 and then it works with gpsbabel. 
My main wish was to get the text-parts which are not converted by Top10 to the ASCII-OVL files. But via V4 converting it works now. And via KML I come to my needed shape-files with QGIS.
Nevertheless I include one of these bad V2-files. It is short, has no text, but shows the problem like all the others from that ONE guy.
[Test.zip](https://github.com/gpsbabel/gpsbabel/files/776595/Test.zip)

closed 2017-02-15 by guenter1952

robertlipe, 2017-02-15:
I'd probably still integrate Ralf's proposed patch, especially if
associated with a test case containing that file and referencing this
report on github in the description. (BTW, in the future, please list the
file format used and a reproducible test case in any report.)

We actually do have a shapefile reader/writer, but I've heard that QGIS's
own GPSBabel integration works better. (Ditto for Grass.) I've not used
either personally, so I can't speak to that.