GPSBabel/gpsbabel #69: columbus v990 duplicates and duplicate filter

wrobell, 2017-03-09:
I own Columbus V990 and I use `v900` driver to convert data from the device.

I have noticed, that files generated by my device contain duplicates (time, 4th column), i.e.
```
202***,T,170226,075106,46.474680N,011.803118E,2417*,4***,0**,3D,SPS ,1.2**,0.9**,0.8**,*********
203***,T,170226,075107,46.474680N,011.803119E,2417*,4***,0**,3D,SPS ,1.2**,0.9**,0.8**,*********
204***,T,170226,075108,46.474681N,011.803124E,2417*,3***,0**,3D,SPS ,1.2**,0.9**,0.8**,*********
205***,T,170226,075109,46.474683N,011.803130E,2417*,0***,0**,3D,SPS ,1.2**,0.9**,0.8**,*********
206***,T,170226,075107,46.474684N,011.803136E,2417*,0***,0**,3D,SPS ,1.2**,0.9**,0.8**,*********
207***,T,170226,075108,46.474688N,011.803145E,2417*,0***,0**,3D,SPS ,1.2**,0.9**,0.8**,*********
208***,T,170226,075109,46.474690N,011.803154E,2417*,0***,0**,3D,SPS ,1.2**,0.9**,0.8**,*********
209***,T,170226,075110,46.474691N,011.803164E,2418*,0***,0**,3D,SPS ,1.2**,0.9**,0.8**,*********
```

I wonder if the `duplicate` filter could be extended to allow filtering timestamp duplicates?

robertlipe, 2017-03-09:
I'd actually be more worried that your device is showing time moving in
reverse. Reporting 107 after 109 - and with different locations - is surely
a device bug. How would you decide which one is correct?

This seems like custom work to fix a device firmware bug, but if you're
interested in doing it, you could probably modify duplicate_process to
store the point's creation_time into q QHash<Waypoint, QDateTime>.  Before
inserting into the hash, test if the hash already contains an entry for
that key and if so, figure out what to do with your time travel exercise
above: fudge the time to be a millisecond after the one before it to keep
time moving forward or delete the duplicates, perhaps creating backward
motion, or whatever.

I don't see this being generally useful enough for me to pay to develop it,
but if you do (and include doc, test cases, etc.) I'd probably accept it.
It'd be a good tiny starter project for someone wanting to get their feet
wet.  duplicate.cc could use a serious modification pass in light of modern
class libraries (like QHash) as it's pretty painful to read owing to its
C-era history.  Since half that file is implementing its own hash system,
you could probably recreate that file from scratch around QHash or QMap or
std::map or such and just keeping the command line compatibility options
easier than untangling it.  But adding the "obvious" hash to the existing
code would be OK, too.

gpsbabel, 2017-03-09:
Actually, try the track filter with the merge option... it just discards
points with duplicate time. You might still jump backward on the race
course or other action. It'll "damage" track segments, but I don't think
v900 really has that concept anyway.

wrobell, 2017-03-09:
Track filter it is... Thanks.

closed 2017-03-09 by wrobell