GPSBabel/gpsbabel #77: igc format parsing crash

languitar, 2017-06-30:
I have fuzzed the igc format parser with afl and the attached file results in a crash (current master) for the following command:

```
gpsbabel -i igc -o gpx -f crash.igc -F -
```

[crash.igc.txt](https://github.com/gpsbabel/gpsbabel/files/1115679/crash.igc.txt)

languitar, 2017-06-30:
I have found a few more files that crash gpsbabel. They are attached
[crashes.zip](https://github.com/gpsbabel/gpsbabel/files/1115844/crashes.zip)

robertlipe, 2017-06-30:
Thanx. The reality is that the IGC reader is full of fragiile constructs
that will fall apart when fed nonsense. I fixed the ones I could repro
in d75fb25..9a0713c  That IGC reader is a bit of house of cards and really
infrequently used so before I spent a weekend making it bulletproof against
malformed input, honestly I'd just remove it or let someone else take it on.

languitar, 2017-07-01:
Actually I'd be sad if the parser was dropped since [GpsDump](https://play.google.com/store/apps/details?id=com.stein.sorensen) for Android uses IGC when downloading tracks from my old Garmin eTrex H via a USB-Serial adapter.

languitar, 2017-07-01:
Btw, 24195b5 has fixed all crashes I could find :)

gpsbabel, 2017-07-26:
Thanx for the test case. You bought a stay of execution to IGC, which we hear almost nothing about on the list.  I think we can close this.

As an aside, does GpsDump for Android actually obey the GPL like displaying the license, offering source, etc?

closed 2017-07-26 by gpsbabel

languitar, 2017-07-26:
Hm why should it have to do that. It just produces such files, doesn't use gpsbabel internally.

gpsbabel, 2017-07-26:
Sorry. I misread your statement. I thought it used GPSBabel to read IGC or
our serial code for the eTrex.