GPSBabel/gpsbabel #78: KML reader may swallow all memory

languitar, 2017-07-01:
Another result of fuzzing: the KML reader may eventually swallow all available memory until the program crashes if fed with the attached files. Seems the parser gets stuck in an endless loop.

Its faster and safer to reproduce this when specifying artificial memory limits:

```bash
LIMIT_MB=200
( ulimit -Sv $[LIMIT_MB << 10]; gpsbabel -i kml -o gpx -f one-of-the-inputs.kml -F - )
```

[kml-crashes.zip](https://github.com/gpsbabel/gpsbabel/files/1117164/kml-crashes.zip)

robertlipe, 2017-07-01:
Looking at the first one, I see the basic problem: KML's polyline
(stupidly, IMO) is an infinitely long CSV string, not bound by XML tags.
It's a bit expensive for us to parse CSV inside XML, especially when one of
the fields is optional and the third, when zero, can either mean an actual
altitude of zero or "we have no idea what the altitude it, but there's
another lon/alt after this one and we just need filler".

inside trk_coord is a loop that is basically

  while ((n = sscanf(CSTRc(iargs), "%lf,%lf,%lf%n", &lon, &lat, &alt,
&consumed)
    if n == 2, handle lon/lat
    if n == 3, handle lon/lat/alt
    otherwise, use 'consumed' to indicate a new index into iargs and parse
again.

The low-rent fix would be to "else, throw an error" because the file I'm
looking at has  something like "1q2,3,4" (the first float is bad).

The betterer fix would be to build a QStringList from an iargs().split(",")
and loop to create trackpoints and populate from that list. I'll have to
think if it's wise to make another copy of the array in core as it could be
large - it's possible that a const QStringList is built from an in-place
index, but that seems improbable. Then again, it would always be correct
and the ratio of available VM to file size is pretty large compared to when
GPSBabel was created.

If anyone wants to beat me to that fix, great. It's a pretty self-contained
project for a budding babel-head.

Notice that both of the issues that Johannes has found have been scanf into
unbounded or variably sized fields. This is one reason I've been chipping
away at removing and replacing these dangerous libc parsers that have 70's
era APIs; they're really hard to make bulletproof for inputs even slightly
out of spec.

In terms of value to the project, I'd rather spend 5 hours working on KML
(it won't take 5 hours) than 1 working on IGC.

RJL

gpsbabel, 2020-05-19:
We've replaced most of that (old) code and improved our error handling. I don't claim we're bulletproof against all malformed input (and don't strive to be) but we now survive that entire folder of input with the tree that's about to be GPSBabel 1.7:

 % for i in ~/Downloads/kml-crashes/crashes.2017-06-30-21:15:34/*; do ./gpsbabel -i kml -f $i -o gpx -F /dev/null ; done
kml : malformed coordinates  "-1"
XML Reader:Read error: Unexpected ''. (/Users/robertlipe/Downloads/kml-crashes/crashes.2017-06-30-21:15:34/id:000002,sig:06,src:000002,op:flip1,pos:867, line 25, col 25)
kml : malformed coordinates  "-112.25507853"
XML Reader:Read error: Unexpected '[0-9]'. (/Users/robertlipe/Downloads/kml-crashes/crashes.2017-06-30-21:15:34/id:000003,sig:06,src:000002,op:flip4,pos:878, line 25, col 37)
kml : malformed coordinates  "0"
kml : malformed coordinates  "0"
kml : malformed coordinates  "0"
kml : malformed coordinates  "-77.0578"
XML Reader:Read error: Opening and ending tag mismatch. (/Users/robertlipe/Downloads/kml-crashes/crashes.2017-06-30-21:15:34/id:000007,sig:06,src:000003,op:havoc,rep:4, line 30, col 34)
kml : malformed coordinates  "-112.25lyStyle>"
XML Reader:Read error: Opening and ending tag mismatch. (/Users/robertlipe/Downloads/kml-crashes/crashes.2017-06-30-21:15:34/id:000008,sig:06,src:000084,op:havoc,rep:8, line 25, col 12)

Look, ma. No resource exhaustion!

Thanx for the report.

closed 2020-05-19 by gpsbabel