GPSBabel/gpsbabel #81: GPX to Naviguide returns an empty file

HarelM, 2017-08-20:
Running the following command line returns an empty file (0 bytes) - exit code 0:
```
gpsbabel.exe -N -i gpx,gpxver=1.1 -f "C:\Users\xxx\AppData\Local\Temp\tmp3C12.tmp" -o naviguide -F "C:\Users\xxx\AppData\Local\Temp\tmp3C22.tmp"
```
Here's the relevant input file.
[מסלול 1.txt](https://github.com/gpsbabel/gpsbabel/files/1237311/1.txt)

Running on Windows 10.
GPSBabel doesn't have a version in the file so it's dated from 02-jan-2016 (worth adding a version to the file...)

gpsbabel, 2017-08-20:
Your inputs are tracks. Our naviguide format supports only waypoints
according to https://www.gpsbabel.org/capabilities.html

You could convert the tracks to waypoint, but I don't know how well that'll
work on the device or progra

1.5.4 was from New Year this year, but I really don't think we've had any
non-mechanical changes in that format in years. I don't know that we have a
lot of Naviguide users on the dev team as it seemed to be of regional
interest. Perhaps post to gpsbabel-misc may turn up more.

HarelM, 2017-08-20:
Thanks for the super fast response!
A user posted on our facebook group that he was unable to export in naviguide format and I now remember that we only imported this format up until now.
I'll try with Waypoints never-the-less.

closed 2017-08-20 by HarelM

tsteven4, 2017-08-20:
This will produce a non-empty file with 1078 points from your sample 
input.  the gpxver option you used is used on write, not read, so it is 
unnecessary.

./gpsbabel -N -i gpx -f  1.gpx -x transform,wpt=trk -o naviguide -F 2.twl

HarelM, 2017-08-20:
The syntax is confusing but I'm sure you know what you are talking about :-)
Thanks!! Not sure I'll use it as I prefer to manipulate the data before passing it to GPSBabel, but it sure can be useful.

robertlipe, 2017-08-20:
From the department of Stupid GPSBabel Tricks, one of the things I do a lot
is feed our own output back to us. (Please don't think too much about
naming that...)  It won't prove that our output can be read by whatever,
but it can prove that at least our reader and writers are consistent with
each other.

For example:

gpsbabel -t -i gpx -f ~/Downloads/1.txt -xtransform,wpt=trk -o naviguide -F
- | gpsbabel -t -i naviguide -f - -o gpx -F -

produces a GPX output that ends in:

    <rtept lat="33.114657888" lon="35.556019305">
      <ele>0.000000</ele>
      <name>WPT1078</name>
      <cmt>WPT1078</cmt>
      <desc>WPT1078</desc>
    </rtept>
  </rte>
</gpx>

..and 1078 happens to be exactly the number that tsteven4 offered.  Taa Daa!

And, yes, we do have some confusing and inconsistent tricks in our toolbox.
For example, "m" means "meters" if the author wasn't American, then it
usually means "miles".

Steven does indeed know what he's talking about. He's been a HUGE help
inside the guts of GPSBabel in recent years.