GPSBabel/gpsbabel #85: MTK_logger.cc creates non-unique temp file 'data.bin' in OS environ temp path

philshem, 2017-09-12:
As discussed on the [_misc_ mailing list ](https://sourceforge.net/p/gpsbabel/mailman/message/36034377/), function GetTempName() at line 300 of [mtk_logger.cc](https://github.com/gpsbabel/gpsbabel/blob/master/mtk_logger.cc) creates the temp path from (in Windows) the TMP environmental path and a file name that is not unique, 'data.bin'. 

This is not optimal because

1. the tmp file is not deleted after export format (e.g. GPX) is created - a security issue as Robert Lipe brings up on the message board
2. the tmp file is not uniquely named, and can create data quality issues, especially if GPSBabel is called for parallel processes and TMP path is not uniquely set for each process.
3. (I've also noticed that if the data.bin already exists and is from the same tracker, the time to generate the GPX or CSV is 5 seconds, instead of 60 seconds with no data.bin file. I don't know what the code knows or does to make this the case, but deleting the temp file would probably be safer.)

A potential fix would be to use [QTemporaryFile](http://doc.qt.io/qt-5/qtemporaryfile.html), which create a unique and multi-OS temporary file, which can then be deleted after the process runs.

Here's my attempt at a fix, although I don't have a setup that is yet able to compile C++ or test the code:

**current:** 

```
static const char* GetTempName(bool backup) { 
  const char kData[]= "data.bin";
  const char kDataBackup[]= "data_old.bin";

  QString t = QDir::tempPath();
  t += QDir::separator();
  t += backup ? kDataBackup : kData;
  return t.toLatin1();
}

```
**suggested:**

```
static const char* GetTempName(bool backup) { 
  QTemporaryFile kFile; // create a unique temp file

  QString t = QDir::tempPath();
  t += QDir::separator();
  t += kFile;
  return t.toLatin1();
}
```

--------------

Background: My application uses GPSBabel for multiple identical devices. I use Python to scan the COM Ports and then generate a dos command to extract the GPX files from the MTK loggers. As a workaround, I can set a temporary and isolated environmental variable TMP which GPSBabel uses for each individual process. Sample Python script is [here](https://stackoverflow.com/a/46163760/2327328).

gpsbabel, 2017-09-12:
Flyby while AFK. Probably closer to
QTF kFile;
kFile.setAutoRemove(false) / Keep the file open even once QTF object goes
out of scope
kFile.setFileTemplate(backup ? data_oldXXX"  : "dataXXX")
return kFile.fileName().toLatin1();

Mixing C and Qt style is weird, but that's lower risk/cost than changing
everything to QIODevice.

robertlipe, 2017-09-13:
I spent some time looking at this code. It's really pretty icky.

This format, pretty uniquely, offers the option of destroying the data on
the device during a read on purpose via the erase and erase_only option.
This code jumps through hoops to read the device twice-ish in this weird
combination thing to create a file that it can then read. It saves this
file to a known name and renames the known name if it finds an existing
one.  This way if you (or the code) does something bad, like writing to an
output directory that you don't have write permissions to, you have a
second chance at dumpster-diving the data_old.bin. This doesn't work if we
randomly name these files and it doesn't work if we remove these on output.

This code is such an odd fit into so many GPSBabel norms that I'm not
totally sure what to do here. Writing files during read is just weird, but
it DOES give us a nice debuggability in the form of "send me the data.bin"
vs. "Here's a shipping address to send me your device to repro".

I'm going to keep thinking about this, but resolving the conflicting "use a
known place and name for repeatability" and "randomize the names and set
fire to them when you're done for security and instance isolation" doesn't
have an obvious solution right now. Fresh perspective welcome.

Sounds like setting the system temp dir before calling us for that (rare)
multi-instance case may be enough pain relief here.