GPSBabel/gpsbabel #94: Fix compiler warnings

fbonzon, 2017-10-26:
You probably want to fix these:

```
zlib/inflate.c:1507:61: warning: shifting a negative signed value is undefined [-Wshift-negative-value]
    if (strm == Z_NULL || strm->state == Z_NULL) return -1L << 16;
                                                        ~~~ ^
```

```
garmin_txt.cc:826:10: warning: implicit conversion from 'const int' to 'char' changes value from 176 to -80 [-Wconstant-conversion]
    *c = kDegreeSymbol;  /* degree sign */
       ~ ^~~~~~~~~~~~~
```

robertlipe, 2017-11-02:
Thanx. Fixed in 09cd1d70..9eb27c2a  
Sidebar: I'm not going to go crazy fixing warnings in third party packages like zlib.

closed 2017-11-02 by robertlipe

HarelM, 2018-04-28:
There are still some warnings when compiling, Any chance to fix them as well?

robertlipe, 2018-04-28:
PRs (that pass presubmit) welcome since there are different warnings for
different build environments.