GPSBabel/gpsbabel #101: GUI Options box for input file is confusing

moveson, 2017-11-05:
Using GPSBabel 1.5.4 on Mac OS X 10.13

Actual behavior: The Options box for the input file has an option with the title, "Target GPX version for output," but the option in fact specifies the gpxver parameter for the input file.

Expected behavior: The title of the option should be "Target GPX version for input"

tsteven4, 2017-11-05:
Actually that option is used to force the version of a gpx output file, 
as the title implies.  To do this you need to set it in the options for 
the output file.  That option doesn't apply for input files, but we use 
the same list of options for input and output, hence the confusion.

moveson, 2017-11-05:
Understood, but when this option is used in the input file options box, it does add the option to the command-line equivalent shown in the report box:

`gpsbabel -w -r -t -i gpx,gpxver=1.0 -f /Users/mmo/Downloads/original.gpx -o gpx -F Users/mmo/Downloads/modified.gpx`

To me, that is confusing.

tsteven4, 2017-11-05:
yes, "it does add the option to the command-line equivalent". Note that 
the GUI actually runs the gpsbabel executable with the command-line that 
is shown in the report box to do the conversion.   The basic issue is 
that the option list for each format doesn't distinguish between input 
and output.

In this case the gpx reader ignores that option.  Only the gpx writer 
uses that option.  We made an attempt to communicate this with the 
title, but having options shown that down apply in the case they are 
shown from is fundamentally confusing.

Without tackling rewriting the option support I can only think of the 
following potential improvements:

1) changing the title of the option

2) changing the documentation in the manual 
https://www.gpsbabel.org/readme.html

moveson, 2017-11-05:
Maybe it would be more clear if the title were something like: "Target output GPX version (ignored for input files)"