GPSBabel/gpsbabel #147: Symbols for fishing/boating application

tzerb, 2017-12-09:
GPS Babel is a great tool and I use it to convert information between my Humminbird and Lowrance fish finders.  One pain point I have is that the symbols used by these tools don't match the names that the respective companies use.  For example, when I convert a humminbird specific hwr file to a gpx file using GPS Babel, I can't import it into the manufacturers application called HumminbirdPC without losing symbol information.  The same is true for Lowrance which now supports gpx.  Furthermore, if I take a gpx file created by Humminbird, I can't import it into Lowrance without losing symbol information.

I propose to add a switch to GPS Babel called 'Use Fishing Symbols'.  When this switch is present, the behavior of the Humminbird format and two Lowrance formats work in a way that use the manufacturer symbol names and a common set of symbol mappings in their conversion.  I've done a bit of research on the devices so I have a good idea of what this symbol set and mapping has to look like.  I've also done research using the GPS Babel code base and I know how the binary files need to be handled to use them.  (I've also noticed that there are errors on how some of the Lowrance file formats are handled.)

I've never contributed to an open source project so I'm looking for direction on what do from here.  I'm an experienced software developer and got the tool chain working to a point where I can change and run the application on a Mac.  This is my first brush with C++ in years so the quality of the code will need some help.  Assuming the community likes the change, If I submit a pull request can/will someone help me create production code?

robertlipe, 2017-12-11:
Hi, and welcome.

As you observe, icon translation is tricky. We don't know the details of
every model out there so we'd welcome a hand. If Lowrance's icon tables are
different for their marine and handheld markets, I suppose we'll have to
rely on a switch for help. Please name the flag itself somewhat generically
like "marine" instead of too specifically "FloatenModel BR234-C", though
it's OK to call out the specific case in the doc itself.

Even if your C++ is weak, the icon tables are about the simplest thing to
modify. We tried (with varying successes across devices and formats...) to
be liberal in what we'd read. The original plan for formats that used
numbers internally and words on screen (computer or device) was to use big
ole tables (which you've now presumably found int he code to handle the
official match as the first one and then add safety aliases as fallbacks.
The idea was that whether you called it "campsite", "camp", "campground",
or "camping" in the text side or formats - like GPX - they'd all get
converted to 10023 (or whatever the device wanted) on the way to the
device, while the way out of the device, we'd prefer to use the "official"
one via the sequential table lookups as they'd match first.

Early in the GPX standardization effort, we realized that even within a
single language the number of variations would have been insane to try to
standardize, so we encouraged *all* GPX readers to do something like I
described above: accept multiple variations of the text input on read. For
example, we try to ignore case and we have duplications, such as using the
same icon ("g") for "bank" and "atm" if we're reading them from, say, a GPX
file. But it's not like we'll catch "Credit Union" or the German version of
any of these. So we try some heroic efforts, but can't get them all. If you
have two GPSes that support GPX and won't read (the majority of) the GPX,
those vendors failed in the spirit of the GPX spec.

As for the fear of production code, I think we have a pretty helpful team
of reviewers and examples that can help get it in...though I'll have to say
that if this is soon, I'll be of little help because I'm headed to the
hospital tomorrow for the first of two major surgeries, so I'll be offline
a lot.  Most of our code isn't particularly "tricky" - at some level, we're
taking sets of coordinates from one or more things, fluffing them a little
bit into a common internal format, then writing them in one or more
(probably different) things. The icon stuff is mostly "smart data, dumb
code"  - we have some table entries that we machine generate but mostly
it's just words in lists.

It's a bit funny that you bring this up. A few weeks ago, I was reminded of
early in the project when this seemed like a this was a super hot topic and
we were endlessly tuning each format to add aliases for each other.
Somewhere about the middle of last decade we either got good enough at it
or people quit caring because this topic just quit turning up a couple of
times a week. /shrug

Good luck!

gpsbabel, 2020-05-19:
User went quiet.

closed 2020-12-27 by robertlipe