GPSBabel/gpsbabel #150: New release?

fxcoudert, 2017-12-15:
Could a new version please be released with latest support for modern Qt (including, I hope, Qt 5.10)?

tsteven4, 2017-12-17:
We are working on a new release but I cannot make promises on when it will be available.

As far as Qt is concerned, I don't understand why you care what version of Qt we build the release with.  If you are using our release then the necessary bits of Qt from the version we release with will be included.  If you are building yourself then you can attempt to use the version of Qt you like as long as it is above our minimum which is currently Qt 5.2.  It is possible you  may run into an incompatibility with a newer version of Qt than we release with, in which case we would like to hear about it.  Personally I have built with versions as recent as Qt 5.9.3. 

We want to make it easy for developers to use Qt as distributed by their distros.  For example Ubuntu 16.04.3 LTS is using Qt 5.5.1.  We need to work with Qt versions that are available with our continuous integration provides, appveyor and travis.  We also need to work on Windows, Linux and macOS.  For this reason we refrain from using new Qt features that are not present in all Qt versions >= our minimum unless we can work with and without them.

Can you explain your concern about the Qt version?

It is very unlikely we will do the next release with Qt 5.10, which is now 12 days old.

fxcoudert, 2017-12-17:
I am from Homebrew, a software distributor. In Homebrew, we distribute Qt 5.10.0, so we'd like to distribute gpsbabel compiled against that Qt 5.10.0.

However, the latest release (1.5.4) does not build against Qt 5.10. We need to apply a patch (https://raw.githubusercontent.com/Homebrew/formula-patches/ca4c4730/gpsbabel/qt5.10.patch) to make it compile.

tsteven4, 2017-12-17:
Can you try to build with Qt 5.10 & without your patch against https://github.com/gpsbabel/gpsbabel.git
I think the issue you have has been fixed.

tsteven4, 2017-12-17:
FYI I was able to build the HEAD cfc6832a6f288703ee94a98923d1c04f880459e1 with Qt 5.10.0 without issue (both the command line and GUI) on linux.

adisbladis, 2018-04-24:
Currently patching `gpsbabel` to build with Qt >=5.9 results in test failures related to rounding (see this attempt at dropping older QT versions in NixOS https://github.com/NixOS/nixpkgs/pull/33836).

A new release (where the tests would hopefully be passing) would be of great value.

tsteven4, 2018-04-24:
> versions in NixOS NixOS/nixpkgs#33836

Actually the behavior of Qt changed with 5.7.  As it stands the test 
cases fail with Qt >= 5.7 and work with older Qt.  We are not ready to 
move our minimum to 5.7 at this time.

tsteven4, 2018-04-29:
Note that we just updated to require Qt >= 5.7.  Our travis CI is building and running regression with Qt 5.9.5 on a docker ubuntu/bionic based image.

tsteven4, 2019-05-02:
Release 1.6.0 is available at https://www.gpsbabel.org/download.html.

Qt 5.9 or newer is required.  We currently run CI against 5.9.8 and 5.12.2/5.12.3.  5.12.2 was used for the windows installer (appveyor is a little behind), and 5.12.3 was used for the macOS installer.

The patch referred to above should not be necessary.

closed 2019-05-02 by tsteven4