GPSBabel/gpsbabel #158: OZI: Unsupported datum 'GRS 80'

HarelM, 2018-01-08:
I'm getting the following error message:
`OZI: Unsupported datum 'GRS 80'`
Here is the relevant file:
[ozi.plt.txt](https://github.com/gpsbabel/gpsbabel/files/1612938/ozi.plt.txt)
partial command line:
`gpsbabel.exe -N -i ozi -f ...`
Any change to support this?

tsteven4, 2018-01-09:
GRS 80 is a known ellipse, but not a datum.  Thus we don't have enough 
information to translate your data.

ozi doesn't claim to use it as a datum:

http://www.oziexplorer3.com/namesearch/datum_list.html

Is it locally defined in your installation?

If your not worried about being exact changing line 2 of your input file 
from "GRS 80" to "WGS 84" will interpret your points as if the datum was 
WGS 84.

HarelM, 2018-01-09:
This is a file that was uploaded to my site, which uses GPSBabel in the background to transform GPS trace files, I have no clue what's the origin of this file...

HarelM, 2018-01-09:
Also it does appear in the following list as ellipsoid:
http://www.oziexplorer3.com/eng/help/userdatums.html

tsteven4, 2018-01-10:
Yes, but we don't know the dx , dy , dz part.

HarelM, 2018-01-13:
I see your point, since this information seems to be missing I can't expect GPSBabel to be able to convert this file. closing...

closed 2018-01-13 by HarelM