GPSBabel/gpsbabel #164: Support docker and rest API

HarelM, 2018-04-22:
Hi,
I've been using GPSBabel for quite a while now and I think it's great! (I even contributed a small bit).
**Short summary:**
Support REST API, support docker.
**Long explanation:**
I'm working a project and I would like to try and make it cross platform.
That's not an easy task. One of the things I would like to do is move everything to docker for easy deploy and to progress to micro-services architecture.
This means that I would like to convert all the executables running in my project to be HTTP services as it will make them cross platform by definition.
I know GPSBabel is a cross platform compiled library but in terms of deployment you'll need to manually deploy the right executable according to OS (correct me if I'm wrong).
A solution to this problem might be to create a cross-platform docker container to run GPSBabel as a web server (This way you can also remove the dependency on Qt if you'd like to create a web based UI, but it will introduce a dependency on an HTTP server).
I wonder what you think about it, I'd be happy to contribute my time to make it happen.

robertlipe, 2018-04-24:
Hi, Harel. As described in the FAQ, we're not a library and I running
GPSBabel as a library really is outside of the scope of our project.

As you observe, we already run on pretty much any interesting operating
system in a server environment. Many people have successfully wrapped the
GPSBabel executable in a web UI and provided that.
http://www.gpsvisualizer.com/gpsbabel/ is probably the best known of those,
though it looks a bit out of date. It's pretty easy to call GPSBabel's
command line. We follow the UNIX philosophy of the command line and pipes
being the API, but you could wrap it CGI or even build a socket service
that knocked out the command line environment but keeping it stable and
secure would keep you busy. A complex track filter or polygon query can
overflow a default PHP 60 second limit which is pretty unkind to a single
threaded service if you're seeking to process thousands of connections per
second. But I'm really not sure how interesting this class of problem is at
scale.

Probably the next portable target for GPSBabel would be Webassembly. That
should approach trivial once the Webassembly version of Qt hits stable; the
beta for it was just announced this morning
http://blog.qt.io/blog/2018/04/23/beta-qt-webassembly-technology-preview/
but I've been watching that project progress for well over a year. That
should let us work on most everything uphill of mobile OSes and it's not
like iOS or Android builds of GPSBabel are really very useful to the masses.

Let us know if you're interested in driving the Webassembly efforts. I'm
rather sidelined right now, but that's probably the most interesting area
left for us to conquer.

HarelM, 2018-04-24:
I think Webassembly is not mature enough to dive into it right now (for me at least).
I'll try to explain the use-case which I think is currently missed out right now:
I have a site, the site runs cross platform code such as dotnet core, java etc.
The site uses the services of GPSBabel executable.
Now if I need to deploy the site on different OSs I need to make sure that GPSBabel is compiled for the relevant OS and deploy it prior to deploying my site or pack it with my site for deployment which will create a non cross-platform package.
In order to avoid this complexity people had started to adopt Docker, which handles most of this stuff out of the box.
In our particular case I can create a docker image which compiles the GPSBabel to a specific platform and group multiple images to support the concept of cross platform image.
This however brings a new "problem": in order to run this cross platform image I need to know that it runs from docker and use docker specific command line arguments - which is not ideal in terms of decoupling.
So, in order not to know which hosting framework is hosting GPSBabel, a web API can be used to remove the docker specific arguments (HTTP rest is a standard and not specific).

The concept of micro services and exposing GPSBabel capabilities through web API (locally) does not break, IMO, the concept of GPSBabel, but only enhance it.
The above example of gpsvisualizer is an non reusable example as I need a browser to interact with GPSBabel and I can't do it using a simple HTTPClient, and the concept of a micros service does not mean it is exposed to the web, just exposed locally on the server that is running the site.

I have written yesterday a small NodeJs web-api that exposed some of the functionalities of GPBBabel to a HTTPClient (which exists in any language). NodeJS in theory is already cross platform and If I'll be able to make a docker image that is cross platform to run this web-api all you'll need to do in order to use GPSBabel is install docker and run this image.
In terms of deployment I think it's a great advantage as you don't need to manually download assemblies that are specific to an OS but rather just install Docker.

I'd be happy to share my code if you'd like.
Maybe it shouldn't be a part of GPSBabel's "core" but a side project/repository in this Github organization.

HarelM, 2018-04-29:
@tsteven4 I saw you are working on something with docker, I would be happy to compare notes.
I'm trying to create a cross platform docker image that will contain a HTTP server and gpsbabel executable.
I decided to update a NuGet package I'm using (IsraelHiking.GPSBabel) to contain both Linux (Debian) and Windows gpsbabel files necessary for runtime but I think I'll need a statically compiled gpsbabel version and I'm struggling to create one, any help would be appreciated (adding -static flag cause compilation error since, I think, Qt5 static libraris are not mapped).
Most of the code is already written here if anyone is interested in looking at it:
https://github.com/IsraelHikingMap/GPSBabelWebAPI

gpsbabel, 2020-05-19:
Closing. User has providined their own solution and offered it for others, so we'll leave this here.

closed 2020-05-19 by gpsbabel

HarelM, 2020-11-04:
For anyone finding this:
I have created a small docker image that wraps GPSBabel in a http server to facilitate micro-services architecture that is based on HTTP and can be distributed.
the image is here:
https://hub.docker.com/r/israelhikingmap/gpsbabelwebapi
It's not cross platform as I don't think someone is using docker windows images nowadays...