GPSBabel/gpsbabel #166: Ubuntu build error: This version of Qt is not supported

zstadler, 2018-04-27:
The _[Building from source](https://www.gpsbabel.org/htmldoc-development/Source.html)_ instructions say:
> Ubuntu users may need to 'apt-get install libqt4-core libqt4-dev'. Qt versions 4.6 and newer are supported. 

Following instructions, this error is produced:
```
g++ -std=gnu++11  -g -O2 -Wall -c   -I. -I/usr/include/qt4  -DHAVE_CONFIG_H -DNEW_STRINGS main.cc -o main.o
main.cc:232:2: error: #error This version of Qt is not supported.
 #error This version of Qt is not supported.
  ^
make: *** [main.o] Error 1
```

which is a result of 
https://github.com/gpsbabel/gpsbabel/blob/cbaa069f8d74b6aa85b5e3d279477807b5960c9f/main.cc#L231-L233

Please update the _[Building from source](https://www.gpsbabel.org/htmldoc-development/Source.html)_ instructions with respect to:
- The minimal Qt version supported
- The process for installing a compatible Qt version on Ubuntu

tsteven4, 2018-04-29:
FYI the minimum Qt version just moved to 5.7.  Ubuntu bionic has 5.9.5, or you can install Qt locally from https://www.qt.io/download.
If you using configure the Qt version is checked there.  There is still the runtime check that caught you.

We will try to update the web site instructions when we can.

tsteven4, 2018-04-29:
Another alternative on linux is to use docker and the ubuntu bionic image tsteven4/gpsbabel_build_environment.  This is what our travis build is doing now, it just runs the script https://github.com/gpsbabel/gpsbabel/blob/master/docker_hook to do the build & regression.  The script build_and_test probably does more than you want - it builds the command line and gui, make the documentation, and does extensive regression.

mjcherbert, 2019-10-16:
on ubuntu 18.04.3 (bionic) I needed to install qt5-qmake and qt5-default packages before configure would complete ...

tsteven4, 2019-10-16:
The manual is continuously out of date.  The best reference for the current Ubuntu packages may be tools/Dockerfile.  This is used in our CI process so it should always be up to date.  There are other Qt packages in the Dockerfile that are required to build the GUI.

Feel free to modify xmldoc/chapters/build.xml and submit a pull request.  There is some missing punctuation next to the relevant line for Ubuntu.

gpsbabel, 2020-05-19:
Instructions have been updated. Keeping the web, code, and internal doc in sequence is hard and we aren't always great at it.

closed 2020-05-20 by zstadler