GPSBabel/gpsbabel #173: [Feature Request] Allow to filter  Filetyp list by file extension

Mannshoch, 2018-05-03:
This huge list of File description is a shit to find your file extension. Allow to search by extension or Product.
Mock-up:
[image]

robertlipe, 2018-05-04:
Go into settings and disable formats you don't care about.

I've meant for some time to add a flag for the, say, 20 most popular types
(we have a loooong tail) and include them automatically but have a dialog
that guides you into setup to enable the ones not turned on by default or
such. It's a bit of a UX exercise to pick the ones that are right for most
people but not hide the more esoteric options for our users that need
those. I've been kicking this around off and on for years and haven't
actually done anything with it. I know that a drop-down of 200 choices is
intimidating and isn't great, either. Maybe the first time you start the
app, you're asked if you want all the formats enabled or just the most
common ones.

Relying on extension alone isn't awesome. Outside of Windows, it's not a
common convention and we have multiple formats that end in ".txt", "xml",
 ".wpt" and such. (Duly noted that even those "many" are better than the
wall of text we currently present people with...) I struggled with this a
bit when I added the drag and drop support.

If we have anyone with a strong sense of UX design willing to tackle it,
this would be a lovely chance to pair a UXer with someone with just a dash
of programming ability. Really the programming side of this task is the
easy part.

This is an area where we could do better. I could coach a couple of budding
babel-heads that are itching to tackle the design and implementation.

RJL

Mannshoch, 2018-05-04:
It would be easiest for the user to select only his file. In this situation, there must be a tool that analyzes the file and selects the file type itself. If the program is unable to detect the correct file type, it may ask for the remaining option. 

I'm sorry I can't help you with the programming.