GPSBabel/gpsbabel #178: mac gui build requires config.h

tsteven4, 2018-05-19:
The mac gui build requires config.h, but it is not generated using the qmake flow, i.e. "qmake GPSBabel.pro && make && cd gui && qmake app.pro && make"

Why do we define HAVE_CONFIG_H?

https://github.com/gpsbabel/gpsbabel/blob/82c969bf024673f33ac5d525a86314c65ff89406/gui/app.pro#L12

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -O2 -std=gnu++11  -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.10 -Wall -W -fPIC -DHAVE_CONFIG_H -DHAVE_WEBENGINE -DQT_NO_DEBUG -DQT_WEBENGINEWIDGETS_LIB -DQT_WEBENGINECORE_LIB -DQT_QUICK_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBCHANNEL_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_POSITIONING_LIB -DQT_CORE_LIB -I. -I../Qt/5.9.5/clang_64/lib/QtWebEngineWidgets.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtWebEngineCore.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtQuick.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtPrintSupport.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtWidgets.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtGui.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtWebChannel.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtQml.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtNetwork.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtXml.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtPositioning.framework/Headers -I../Qt/5.9.5/clang_64/lib/QtCore.framework/Headers -Iobjects -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/AGL.framework/Headers -Itmp -I../Qt/5.9.5/clang_64/mkspecs/macx-clang -F/Users/travis/build/tsteven4/gpsbabel/Qt/5.9.5/clang_64/lib -o objects/upgrade.o upgrade.cc
upgrade.cc:28:10: fatal error: '../config.h' file not found
#include "../config.h"
         ^
1 error generated.
make: *** [objects/upgrade.o] Error 1

robertlipe, 2018-05-21:
config.h is generated by configure. If you have a build system that isn't configure, you have to either provide a dummied up config.h or just pass -DHAVE_THIS and -DHAVE_THAT on the command line.

The way we shoved the top-level config.h into gui and (I think) jeeps is indeed awkward.

I thought I "fixed" this some time ago. Must be in one of those trees I committed but didn't push or something. :-(  Please don't set CONFIG_H in app.pro - but we have the unsolved issue of version numbers into the build files.

gpsbabel, 2018-05-21:
We've actually REALLY whittled the HAVE_FOO out of this project. I know I'd
bee chipping away at anything I'd touched, but we really have only N cases
left.
1) magproto uses globbing to handle wildcards if you treat a mounted file
system unit as a serial unit. There are so few affected GPSes in question
that I'd rather just delete all the affected code.
2) gbsleep has a bunch of OS abstraction. I think in modern times, that
file reduces to QThread::msleep(). That may not work well since we're not
_really_ threaded and a qApplication, but that's a clear future.
3) some of our external C stuff like strptime and lib relies on at least
some bits of autoconf plumbing.
4) We use it to find libusb. You have a scheme for doing those in .pro and
we ended up providing our own on MacOS.
5) There's some messiness about webkit vs. web engine for mapping in the
GUI. Does Linux with Qt >=5.9  usually ship with a sensible web engine?
6) osLoadDeviceNameCombos has some autoconf magic. QStringList ports ; auto
foo = QSerialPortInfo::availablePorts(); ports.append(foo[i].portname();
return ports; would let us use Qt for device enumeration in the menus and
keep the (expensive to replace/retest) hand-grown serial abstraction.

So, yes, I think that squishing our use of autoconf's #defines is quite
within reach. Even if we don't get it quite to zero but have a few for
strptime or zlib, still seems worthwhile.

PR #179 (2018-05-21): Add mac build to travis CI
1. This adds a build on macOS to travis CI.  The travis osx queues seem to be short at this time.
2. This adds a check target to GPSBabel.pro that just runs testo.
3. This attempts to decouple the GUI from config.h by making changes to app.pro and upgrade.cc.  These changes should be validated, and if valid this PR could resolve #178.
4. The new script, tools/travis_script_osx, attempts to build the CLI and test it, build the GUI, and then generate a dmg.  This was created based on some targets in Makefile.in.  There is an attempt at the end of the script to examine  [... cut]
closed 2018-05-21 by tsteven4