GPSBabel/gpsbabel #193: How to run on Ubuntu

answerquest, 2018-06-16:
Hi, could there be some mentioning on the main readme file about how this program can be used on Ubuntu? Meaning, what commands do I need to execute to start the program? There's 100s of files, I tried running some executables and am not getting anywhere.

tsteven4, 2018-06-16:
Do you just want to use gpsbabel, or do you want to build it from source?

Ubuntu supplies packages if you just want to use gpsbabel, but they may old:
trusty has gpsbabel (1.4.3-1ubuntu1)
xenial has gpsbabel (1.5.2-1ubuntu1)
bionic has gpsbabel (1.5.4-2)
To do this just install, .e.g 'sudo apt-get install gpsbabel', and run the command gpsbabel or 'man gpsbabel'.  There is full user documentation at https://www.gpsbabel.org/readme.html.

If you want or need to build from source you will need to install the dependencies and run the build.  The dependencies will depend on the version of ubuntu and the version of gpsbabel.  What version of ubuntu are you using?  Are you trying to build from the tarball at https://www.gpsbabel.org/download.html or from a github clone of the HEAD?

answerquest, 2018-06-16:
just wanted to run it. Thanks!

closed 2018-06-16 by answerquest