GPSBabel/gpsbabel #225: Build without qt

xmedeko, 2018-07-09:
It would be nice to have a possibility to build a command-line gpsbabel only without qt dependency. Thanks.

tsteven4, 2018-07-09:
This is not possible and is not the direction we are moving.  The 
command line client requires Qt5Core.  The latest development code 
requires version 5.7 or newer.  If this isn't available from your 
distribution then you can use an installer provided by qt.  There are 
online installers available here:

http://download.qt.io/archive/online_installers/3.0/

and offline installers here

http://download.qt.io/archive/qt/5.9/5.9.6/

I cited version 5.9.6 because our continuous integration tends to use 
5.9.x, but 5.7 or newer should work.

You don't need a qt login, you can hit skip at that step.  The only 
component you need to select is something like "Desktop gcc 64-bit" or a 
variation of that depending on your platform and compiler.

On the other hand, if you are installing for windows or mac from 
https://www.gpsbabel.org/download.html then everything you need is 
bundled with the gpsbabel installer already.

xmedeko, 2018-07-09:
@tsteven4 thanks for a quick reply. I know that command line client requires Qt5Core, but I guess just a few Qt classes are really needed by the command line client and the question is, if it would be beneficial to remove these Qt dependencies from the gpsbabel core and replace it with standard C++ structures or own implementation.

robertlipe, 2018-07-09:
We use Qt for our OS abstraction, sane conditioner classes and strings,
time, networking, and more. Even with c++17, the features we need aren't
there and a team our size isn't able to reimplement then.

We use Qt. If that's a deal breaker for, hopefully you can find another
project to suit your need.

closed 2018-07-09 by gpsbabel