GPSBabel/gpsbabel #241: unicsv fields for output

xmedeko, 2018-07-26:
[unicsv format may have fields specified for the input](https://www.gpsbabel.org/htmldoc-development/fmt_unicsv.html#fmt_unicsv_o_fields) I would like to have same parameter working for the unicsv output, too.

robertlipe, 2018-10-27:
If we have any data for any field, we'll write it in unicsv.
If you're creating a custom output file where you need control of the order
and length of output files, you'll probably be better served by defining
your own style definition.

See https://www.gpsbabel.org/htmldoc-development/Styles.html You can also
see that our own openoffice and tabsep formats are done in the same way.
See https://github.com/gpsbabel/gpsbabel/blob/master/style/tabsep.style for
example.

RJL

xmedeko, 2018-10-27:
Thanks very much, will try that.

closed 2018-10-27 by xmedeko