GPSBabel/gpsbabel #243: arc filter with gpx input

ftilmann, 2018-08-10:
I am trying to use the arc filter like this:
```gpsbabel -i gpx -f trip_all.gpx  -x arc,file=tracciato-blue.arc,distance=0.015K -o gpx -F trip_loop.gpx```
to obtain all gpx trackpoints near tracciato-blue.arc. However, the output file here is identical to the input file - no selection seems to take place.
If I use arc files for input and output it works as expected:
```gpsbabel -i arc -f trip_all.arc  -x arc,file=tracciato-blue.arc,distance=0.015K -o arc -F trip_loop.arc```
(obviously after having converted the input file to arc format). However, this loses the time stamp and elevation from the track points, which I absolutely need for my use case. 
Am I doing something wrong?  Are there any workarounds, e.g. another format preserving time stamps and working with the arc filter.

I only paste small excerpts of the files to show the format. Probably enough to reproduce partially, as the output should be empty. Let me know if you need full files.

tracciato-blue.arc:
```
37.69110	15.10301
37.69099	15.10296
37.69084	15.10358
37.69088	15.10366
37.69174	15.10417
...
```

trip_all.gpx:
```
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.1" creator="GPSBabel - http://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/1">
  <metadata>
    <time>2018-08-07T13:37:45.185Z</time>
    <bounds minlat="37.679170800" minlon="15.102472700" maxlat="37.839435500" maxlon="15.147717400"/>
  </metadata>
  <trk>
    <trkseg>
      <trkpt lat="37.838941900" lon="15.147717400">
        <ele>541.620000</ele>
        <time>2018-08-03T07:09:19Z</time>
        <hdop>8.000000</hdop>
        <extensions>
          <speed>2.236</speed>
        </extensions>
      </trkpt>
      <trkpt lat="37.838871200" lon="15.147557600">
        <ele>539.057000</ele>
        <time>2018-08-03T07:09:24Z</time>
        <hdop>4.000000</hdop>
        <extensions>
          <speed>2.123</speed>
        </extensions>
      </trkpt>
...
    </trkseg>
  </trk>
</gpx>
```
```
(GPSBabel Version 1.5.2. ; sorry if this was fixed in the meantime)

ftilmann, 2018-08-10:
Ok, I realise the arc filter works on waypoints not tracks. So the following solved my problem:
```gpsbabel -i gpx -f trip_all.gpx -x transform,wpt=trk,del -x arc,file=tracciato-blue.arc,distance=0.015K -x transform,trk=wpt,del -o gpx -F trip_loop.gpx```
I suggest to amend the examples in the documentation for this use case, as this is not obvious to the casual user of gpsbabel. I also noted the "trk" option of the "arc" filter but its usage is not clearly described and I could not figure out how to use it.  The issue can essentially be closed, except for a request for documentation improvement.

daganzdaanda, 2019-07-28:
Hi, I just came across the "arc filter works on waypoints not tracks" problem, too. +1 to please make this more clear in the documentation.