GPSBabel/gpsbabel #247: Reverse engineering tool Kaitai Struct

Kalli1964, 2018-09-01:
Hello,

maybe it helps for exploring binary file formats. There is a project named Kaitai Struct (kaitai.io) which provides tools where you describe the file format in a YAML file and then it generates code for you. There is a nice WEB IDE (ide.kaitai.io) where you can interactive work on the YAML file.

As a proof of concept I generated a gpi.ksy (have to named it gpi.txt to upload it here)  using the information from garmin_gpi.cc.

I guess for you guys it's more helpful for exploring fileformats than generating code. To check it out, just load the garmin_gpi.gpi and the attached file (renamed to .ksy) to the WEB IDE and have fun.

Regards, Kalle
[gpi.txt](https://github.com/gpsbabel/gpsbabel/files/2342375/gpi.txt)

robertlipe, 2018-09-03:
Interesting, but probably worth more a mention to our gpsbabel-code mailing
list than promoting the project here.  It's a visually beautiful project,
with some awkwardness in the IDE.

I think that much more of our time is spent in the actual reverse
engineering phase where we stare at the binaries and try to figure out
what's in them (generating the .ksy in your parlance) than the fairly
mundane task (not that we always get that right...) than of describing that
to the computer. It *is *nice to have a pointy clicky way to look at the
bytestream and automatically decode it as a double, a float, a long, etc
and to look for repeating patterns like I describe at
https://www.gpsbabel.org/FAQ.html

Certainly if someone came to us with a format described as well as you have
in gpi.txt and contracted us to produce code to support it, the cost would
have been a fraction as the big cost is discovering the tags, the
relationship to repeating blocks, etc. OTOH, even a description this
thorough leaves out things like the recursive bounding box nature of the
tags, so there's still some information that's not recovered in this way.
For highly repetitive data that's in simple-ish blocks, like most of our
track formats, I'd expect that a non-programmer could probably produce a
working ktf that'd work well enough for a programmer to generate code and
integrate.

This project actually reminds me of something I worked on in the 90's
called the Kernel Memory Editor where we could describe what was basically
C structs that represented device hardware registers or kernel data
structures, buffers, network packets, or whatever and display/edit them
like a spreadsheet. It never really lived too far once we released it
(admittedly, poorly) in the early days of Sourceforge as we original
developers had all moved on to other projects and companies.

Still, it's a neat tool for the toolbox, but not really appropriate for a
bugtracker for us, so I'll be closing this.

Good luck
RJL

closed 2018-09-03 by gpsbabel

KOLANICH, 2021-01-02:
>I think that much more of our time is spent in the actual reverse engineering phase where we stare at the binaries and try to figure out what's in them

Exactly one of use cases of KS WebIDE

KOLANICH, 2021-01-02:
@Kalli1964 , how about upstreaming your spec into KSF?