GPSBabel/gpsbabel #250: Holux M-241 Plus Support

bearbin, 2018-09-25:
Holux has released a new version of their M-241 GPS logger, the M-241 Plus, and extraction of logs from the new device does not appear to work with this software. Is it possible that this hardware will be supported?

cjolif, 2018-09-29:
I'm facing the same issue. I'm not sure what is going on but I can't extract logs with M-241 plus. The read operation goes on forever without stopping, it seems from the debug trace it reads the current reporting location despite the log being off on the device (so empty location). In case it helps here are the debug traces:

```
➜  ~ gpsbabel -D8 -t -i m241 -f /dev/cu.SLAB_USBtoUART -o gpx -F test.gpx
GPSBabel Version: 1.5.4
options: module/option=value: m241/erase="0" (=default)
options: module/option=value: m241/erase_only="0" (=default)
options: module/option=value: m241/log_enable="0" (=default)
options: module/option=value: m241/block_size_kb="1" (=default)
Opening port /dev/cu.SLAB_USBtoUART...
gbser_init("/dev/cu.SLAB_USBtoUART")
Verifying MTK based device...
Send $PMTK605*31
 Read 52 bytes: '$PMTK705,AXN_3.8_3333_16080400,0000,01106-00A,1.0*63'
NMEA command success !
Download /dev/cu.SLAB_USBtoUART -> /var/folders/lr/bdsp50f96bn3gw_sz88vr5b80000gn/T/data.bin
Send $PMTK182,2,7*3C
 Read 17 bytes: '$PMTK001,182,1*39'
NMEA command '$PMTK182,2,7*3C
' failed - Unsupported packet type
Send $PMTK182,5*20
 Read 17 bytes: '$PMTK001,182,1*39'
NMEA command '$PMTK182,5*20
' failed - Unsupported packet type
 ---- LOG DISABLE ---- Fail
Send $PMTK182,2,8*33
 Read 37 bytes: '$GNGGA,214350.199,,,,,0,0,,,M,,M,,*56'
Read 28 bytes: '$GPGSA,A,1,,,,,,,,,,,,,,,*1E'
Read 28 bytes: '$GLGSA,A,1,,,,,,,,,,,,,,,*02'
Read 16 bytes: '$GPGSV,1,1,00*79'
Read 16 bytes: '$GLGSV,1,1,00*65'
Read 47 bytes: '$GNRMC,214350.199,V,,,,,0.00,0.00,290918,,,N*58'
Read 35 bytes: '$GNVTG,0.00,T,,M,0.00,N,0.00,K,N*2C'
Read 17 bytes: '$PMTK001,182,1*39'
NMEA command '$PMTK182,2,8*33
' failed - Unsupported packet type
Download 2048kB from device
Download block size is 1024 bytes
Send $PMTK182,7,00000000,00000400*26
 Read 17 bytes: '$PMTK001,182,1*39'
Read 37 bytes: '$GNGGA,214351.199,,,,,0,0,,,M,,M,,*57'
Read 28 bytes: '$GPGSA,A,1,,,,,,,,,,,,,,,*1E'
Read 28 bytes: '$GLGSA,A,1,,,,,,,,,,,,,,,*02'
Read 16 bytes: '$GPGSV,1,1,00*79'
Read 16 bytes: '$GLGSV,1,1,00*65'
Read 47 bytes: '$GNRMC,214351.199,V,,,,,0.00,0.00,290918,,,N*59'
Read 35 bytes: '$GNVTG,0.00,T,,M,0.00,N,0.00,K,N*2C'
Read 37 bytes: '$GNGGA,214352.199,,,,,0,0,,,M,,M,,*54'
Read 28 bytes: '$GPGSA,A,1,,,,,,,,,,,,,,,*1E'
Read 28 bytes: '$GLGSA,A,1,,,,,,,,,,,,,,,*02'
Read 16 bytes: '$GPGSV,1,1,00*79'
Read 16 bytes: '$GLGSV,1,1,00*65'
Read 47 bytes: '$GNRMC,214352.199,V,,,,,0.00,0.00,290918,,,N*5A'
Read 35 bytes: '$GNVTG,0.00,T,,M,0.00,N,0.00,K,N*2C'
Read 37 bytes: '$GNGGA,214353.199,,,,,0,0,,,M,,M,,*55'
Read 28 bytes: '$GPGSA,A,1,,,,,,,,,,,,,,,*1E'
Read 28 bytes: '$GLGSA,A,1,,,,,,,,,,,,,,,*02'
Read 16 bytes: '$GPGSV,1,1,00*79'
Read 16 bytes: '$GLGSV,1,1,00*65'
Read 47 bytes: '$GNRMC,214353.199,V,,,,,0.00,0.00,290918,,,N*5B'
Read 35 bytes: '$GNVTG,0.00,T,,M,0.00,N,0.00,K,N*2C'
...
```

cjolif, 2018-09-30:
so it looks like $PMTK182 instruction is not supported anymore? If I sniff what Holux ezTour is doing they are using a totally different instructions set starting with $PHLX... I also found http://4river.a.la9.jp/gps/LogFormat.htm which seems to imply that even the data log format is totally different than for m241.

KosmicznyAlfons, 2019-07-24:
HI! I'm also having the same problem. Any chance to add support for the new Holux?

gpsbabel, 2019-07-25:
It looks like there's now a 241-c (green) and a 241-plus (red+white). Holux
is pretty notorious for changing their products and protocols in
incompatible ways, so I know it's frustrating for everyone when this
doesn't work. Roaming around the internet, I don't see a lot of discussion
about this device (well, beyond frustration) and I don't see that anyone
has really fully cracked the protocol and published an open-source
implementation so we can see any new $PMTK super sauce. It's likely that
our cost to implement this in reverse engineering will dwarf the cost of
the device itself.

Christophe, can you see enough of what ezTour is doing to describe the
protocol? If it was buying and importing the device and tweaking it (yes,
I'd pass the hat for costs...) I'd get one on order with enough demand, but
I'm reluctant to bite off a big reverse engineering effort.

I've tried to contact someone in Holux that can provide access to
engineering or at least published specifications on their devices, but I've
never succeeded. Perhaps you, their customers, can motivate them to provide
such help.

There's at least three of you interested. Can you help with the
engineering, finding the specification, or finding someone else that's done
it and published open-source code?

cjolif, 2019-07-26:
@gpsbabel a couple of hours looking at it at the time I wrote this was definitely not enough to guess the protocol. I would need more time that I unfortunately don't have at the moment. I might come back to this later though. I will also contact them in case it helps but even if I'm a customer I'm not sure they will be more responsive with me ;)

irvin, 2019-07-26:
Unfortunately the company just go bankrupt. The local news came out last week, and they will lay-off all the engineer after August.

robertlipe, 2019-07-27:
That's an interesting twist, Irvin. That doesn't help the priority of this,
but that's unlikely comforting to the that already own the product. That
means the pool of people needing this won't grow much more over time.

Even with that news in hand, I can't find any mention of it in the
(English-speaking) web. Are they shutting down specific product groups or
the whole company? Can you provide any more information?

Holux has always been an interesting case for us. The M-241 is very popular
in the photography/geotagging world and we've seen some amount of interest
in that device for a long time. I've long been aware that they make entire
product lines of GPS for motorbikes, marine, outdoors, fitness, and dash
cams (SRT format to merge location with frames) but we just don't seem to
hear much about them.

Is it time to add Holux to the pile that includes Wintec and AMOD of
companies that tried to make a go in this market and failed?  Is QStarz the
last one standing in the logger space? Do they have any competition left
between the generic AliExpress products and an outdoor-class GPS?

irvin, 2019-07-30:
@robertlipe It's the whole company. Because Holux are listed so they publish the press release and multiple local tech site report that. They already lay-off all staff on 15 July* but have not shut down the company yet.

Holux had tried get into fitness watch market but failed in past few years, and the market of their main product, the consumer car GPS is dead long. 

I hope that the CEO can successfully restart it as a startup, dedicating to build some GPS trackers like m-241 that target those niche market but can last and selling for more than 10 years. Or at least some other company purchase them with their patent, to keep M-241 available.  

besides QStarz, the i-gotU devices from Mobile Action (GT-600 at around 50 USD) is also available on the market, http://global.mobileaction.com/support/support_igotU_Faq.jsp 

* all story in Traditional Chinese 
https://finance.technews.tw/2019/07/17/holux-layoff-employee/
https://www.mirrormedia.mg/story/20190717edi007/
https://www.techbang.com/posts/71607-domestic-gps-brand-changtian-technology-holux-announced-that-it-will-send-all-employees-due-to-unpaid-salary-due-to-losses

robertlipe, 2019-07-30:
Very interesting. Thank you for additional details.

It's interesting that their main line was automotive GPS. I don't recall
seeing them or their fitness products ever mentioned on our mailing lists,
so that shows how different and localized the markets are. I'm also
familiar with GCJ-02  and because of the legal issues there, don't
particularly seek out Chinese support.

We had a person or two ask about the i-gotu's a long time ago, but I can't
tell that we've ever supported them as the interest just wasn't there.

I hate to lose GPS companies these days. In the logger market particularly,
the race to the bottom for price really had no winners.

I'll leave this open for now, but unless someone has the hardware, can
reverse engineer the protocol, and sends us a testable patch that we can
integrate, it's not likely to go far.

RJL

closed 2019-12-06 by robertlipe

irvin, 2019-12-30:
Just discover that this utility can read and download logs from M241 plus, http://4river.a.la9.jp/gps/#006

zdila, 2019-12-30:
@irvin sadly no linux support