GPSBabel/gpsbabel #251: Undefined Refernce during Make

PrimordialKangaroo, 2018-10-04:
Using Ubuntu 16.04 I installed Qt 5.7 from the qt archive as 5.7 is no longer an official release. I also tried 5.96 later and was still unsuccessful with the same errors. During the make I receive a number of undefined references to the following:

qt_version_tag
QString::resize(int, QChar)
QByteArray::insert(int, int, char)
QDateTime::QDateTime(QDateTime&&)
QJsonArray::detach2(unsigned int)
QJsonValue::toString() const

Attached is the error from the make.

[gpsbabel_error.txt](https://github.com/gpsbabel/gpsbabel/files/2448322/gpsbabel_error.txt)

tsteven4, 2018-10-08:
Sorry, I see your point now.  There is an issue with library ordering in linking and another with rpath whe using configure.  Rather than fool around with variables and configure it is easier to use qmake to create the makefile.  Another alternative is  to upgrade to ubuntu 18.04 and use the ubuntu supplied Qt.

tsteven4@ubuntu1604:\~/work/gpsbabel_clones$ git clone git@github.com:gpsbabel/gpsbabel.git gpsbabel_git
Cloning into 'gpsbabel_git'...
remote: Enumerating objects: 13, done.
remote: Counting objects: 100% (13/13), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 37482 (delta 5), reused 1 (delta 0), pack-reused 37469
Receiving objects: 100% (37482/37482), 48.48 MiB | 2.38 MiB/s, done.
Resolving deltas: 100% (27548/27548), done.
Checking connectivity... done.
tsteven4@ubuntu1604\:~/work/gpsbabel_clones$ cd gpsbabel_git
tsteven4@ubuntu1604:\~/work/gpsbabel_clones/gpsbabel_git$ /opt/Qt/5.7/gcc_64/bin/qmake 
Info: creating stash file /home/tsteven4/work/gpsbabel_clones/gpsbabel_git/.qmake.stash
Info: creating cache file /home/tsteven4/work/gpsbabel_clones/gpsbabel_git/.qmake.cache
Checking for unistd... yes
Checking for stdarg... yes
tsteven4@ubuntu1604:\~/work/gpsbabel_clones/gpsbabel_git$ make
g++ -c -pipe -O2 -std=gnu++11 -Wall -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_NANOSLEEP -DHAVE_LIBUSB -DHAVE_GLOB -DHAVE_LINUX_HID -DNEW_STRINGS -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/opt/Qt/5.7/gcc_64/include -I/opt/Qt/5.7/gcc_64/include/QtCore -I. -I/opt/Qt/5.7/gcc_64/mkspecs/linux-g++ -o gbser_posix.o gbser_posix.cc
g++ -c -pipe -O2 -std=gnu++11 -Wall -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_NANOSLEEP -DHAVE_LIBUSB -DHAVE_GLOB -DHAVE_LINUX_HID -DNEW_STRINGS -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/opt/Qt/5.7/gcc_64/include -I/opt/Qt/5.7/gcc_64/include/QtCore -I. -I/opt/Qt/5.7/gcc_64/mkspecs/linux-g++ -o magproto.o magproto.cc
g++ -c -pipe -O2 -std=gnu++11 -Wall -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_NANOSLEEP -DHAVE_LIBUSB -DHAVE_GLOB -DHAVE_LINUX_HID -DNEW_STRINGS -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/opt/Qt/5.7/gcc_64/include -I/opt/Qt/5.7/gcc_64/include/QtCore -I. -I/opt/Qt/5.7/gcc_64/mkspecs/linux-g++ -o explorist_ini.o explorist_ini.cc
g++ -c -pipe -O2 -std=gnu++11 -Wall -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_NANOSLEEP -DHAVE_LIBUSB -DHAVE_GLOB -DHAVE_LINUX_HID -DNEW_STRINGS -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/opt/Qt/5.7/gcc_64/include -I/opt/Qt/5.7/gcc_64/include/QtCore -I. -I/opt/Qt/5.7/gcc_64/mkspecs/linux-g++ -o gpx.o gpx.cc
g++ -c -pipe -O2 -std=gnu++11 -Wall -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_NANOSLEEP -DHAVE_LIBUSB -DHAVE_GLOB -DHAVE_LINUX_HID -DNEW_STRINGS -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/opt/Qt/5.7/gcc_64/include -I/opt/Qt/5.7/gcc_64/include/QtCore -I. -I/opt/Qt/5.7/gcc_64/mkspecs/linux-g++ -o geo.o geo.cc
g++ -c -pipe -O2 -std=gnu++11 -Wall -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_NANOSLEEP -DHAVE_LIBUSB -DHAVE_GLOB -DHAVE_LINUX_HID -DNEW_STRINGS -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/opt/Qt/5.7/gcc_64/include -I/opt/Qt/5.7/gcc_64/include/QtCore -I. -I/opt/Qt/5.7/gcc_64/mkspecs/linux-g++ -o mapsend.o mapsend.cc
g++ -c -pipe -O2 -std=gnu++11 -Wall -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_NANOSLEEP -DHAVE_LIBUSB -DHAVE_GLOB -DHAVE_LINUX_HID -DNEW_STRINGS -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/opt/Qt/5.7/gcc_64/include -I/opt/Qt/5.7/gcc_64/include/QtCore -I. -I/opt/Qt/5.7/gcc_64/mkspecs/linux-g++ -o mapsource.o mapsource.cc
g++ -c -pipe -O2 -std=gnu++11 -Wall -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_NANOSLEEP -DHAVE_LIBUSB -DHAVE_GLOB -DHAVE_LINUX_HID -DNEW_STRINGS -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/opt/Qt/5.7/gcc_64/include -I/opt/Qt/5.7/gcc_64/include/QtCore -I. -I/opt/Qt/5.7/gcc_64/mkspecs/linux-g++ -o garmin.o garmin.cc
g++ -c -pipe -O2 -std=gnu++11 -Wall -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_NANOSLEEP -DHAVE_LIBUSB -DHAVE_GLOB -DHAVE_LINUX_HID -DNEW_STRINGS -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/opt/Qt/5.7/gcc_64/include -I/opt/Qt/5.7/gcc_64/include/QtCore -I. -I/opt/Qt/5.7/gcc_64/mkspecs/linux-g++ -o garmin_device_xml.o garmin_device_xml.cc
g++ -c -pipe -O2 -std=gnu++11 -Wall -D_REENTRANT -fPIC -DNDEBUG -DHAVE_UNISTD_H -DHAVE_STDARG_H -DHAVE_NANOSLEEP -DHAVE_LIBUSB -DHAVE_GLOB -DHAVE_LINUX_HID -DNEW_STRINGS -DMAXIMAL_ENABLED -DFILTERS_ENABLED -DSHAPELIB_ENABLED -DCSVFMTS_ENABLED -DQT_NO_DEBUG -DQT_CORE_LIB -I. -Izlib -Ijeeps -I/opt/Qt/5.7/gcc_64/include -I/opt/Qt/5.7/gcc_64/include/QtCore -I. -I/opt/Qt/5.7/gcc_64/mkspecs/linux-g++ -o garmin_tables.o garmin_tables.cc
[... cut ...] gbfile.o parse.o session.o main.o globals.o usasciicodec.o xmlstreamwriter.o shpopen.o dbfopen.o safileio.o adler32.o compress.o crc32.o deflate.o inffast.o inflate.o infback.o inftrees.o trees.o uncompr.o gzlib.o gzclose.o gzread.o gzwrite.o zutil.o gpsapp.o gpscom.o gpsmath.o gpsmem.o gpsprot.o gpsread.o gpsdevice.o gpsdevice_ser.o gpsdevice_usb.o gpsrqst.o gpssend.o gpsserial.o jgpsutil.o gpsusbread.o gpsusbsend.o gpsusbcommon.o gpslibusb.o   -lusb -L/opt/Qt/5.7/gcc_64/lib -lQt5Core -lpthread

tsteven4, 2018-10-08:
A kludge to get configure working with your setup is:
QMAKE=/opt/Qt/5.7/gcc_64/bin/qmake ./configure
EXTRA_LDFLAGS="-Wl,-rpath,/opt/Qt/5.7/gcc_64/lib -L/opt/Qt/5.7/gcc_64/lib -lQt5Core" make

PR #256 (2018-10-16): have configure conditionally set rpath.
this may be necessary when using a Qt installation
that was not provided by the system.

This should resolve #251 provided that the correct qmake is found by configure.  This can be done by setting the path so the correct qmake is found or setting the QMAKE variable to point to the correct qmake.
closed 2018-10-16 by tsteven4