GPSBabel/gpsbabel #257: The sort filter does not take milliseconds into account

connesc, 2018-10-11:
Using the `sort,time` filter seems to ignore sub-second precision.

To reproduce the problem, you can consider this sample input (`test.gpx`):

```xml
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" xmlns="http://www.topografix.com/GPX/1/0">
  <wpt lat="48.8583701" lon="2.2922926">
    <time>2018-07-06T05:04:03.750Z</time>
  </wpt>
  <wpt lat="48.8583701" lon="2.2922926">
    <time>2018-07-06T05:04:03.250Z</time>
  </wpt>
</gpx>
```

Then the following command does not swap waypoints as expected:

```shell
gpsbabel -i gpx -f test.gpx -x sort,time -o gpx -F -
```

closed 2018-10-11 by tsteven4

connesc, 2018-10-12:
Awesome, thanks @tsteven4! :+1: 
Is there any plan to release a new version soon?

tsteven4, 2018-10-12:
> Awesome, thanks @tsteven4 <https://github.com/tsteven4>! 👍
> Is there any plan to release a new version soon?

It is hard to predict when the next version will be released, but there 
is a Pre-release version that has the changes here:

https://github.com/gpsbabel/gpsbabel/releases

For windows see the link currently titled
GPSBabel-20181011T1840Z-11bef4d-Setup.exe

There is a mac version there also.

The Pre-release versions have passed regression before creating the 
installer, but you will probably be the first person to actually run the 
installer and try it.

PR #258 (2018-10-11): correct sort filter time option to use msecs.
This should resolve #257.