GPSBabel/gpsbabel #314: OZI: character encoding not respected

jdpiguet, 2019-03-04:
When using gpsbabel to filter OziExplorer track file, the non ASCII characters in track name are lost.
For example a track description "àéè" will be transformed in "???".

Tested on Debian 9 and Mint 19.1

jdp@amroth:/tmp$ gpsbabel -t -i ozi -f 20070813.plt -x track,pack -x simplify,length,error=0.2k -o ozi,pack -F 20070813_out.plt
jdp@amroth:/tmp$ head -5 20070813.plt
OziExplorer Track Point File Version 2.1
WGS 84
Altitude is in Feet
Reserved 3
0,2,255,V�zelay / Cuncy-l�s-Varzy          ,1,0,2,8421376,0
jdp@amroth:/tmp$ head -5 20070813_out.plt
OziExplorer Track Point File Version 2.1
WGS 84
Altitude is in Feet
Reserved 3
0,2,255,V?zelay / Cuncy-l?s-Varzy,0,0,2,8421376
jdp@amroth:/tmp$ gpsbabel -V

GPSBabel Version 1.5.4

jdp@amroth:/tmp$

tsteven4, 2019-03-04:
I believe this is fixed in the HEAD 
https://github.com/gpsbabel/gpsbabel.  On unix you will have to buid 
from source, for windows and macos there are pre-release builds at 
https://github.com/gpsbabel/gpsbabel/releases

Can you try it?  If not, can you supply a test case and I will verify if 
it is fixed.

jdpiguet, 2019-03-04:
Files used for:
`gpsbabel -t -i ozi -f 20070813_short.plt -x track,pack -x simplify,length,error=0.2k -o ozi,pack -F 20070813_out.plt
`
Check the line number 5...
Thanks and regards, Jacques-D.

[tracks.zip](https://github.com/gpsbabel/gpsbabel/files/2927372/tracks.zip)

jdpiguet, 2019-03-04:
Very strange: if I first convert the file to UTF-8 with:
`iconv --from-code=ISO_8859-1 --to-code=UTF-8 20070813_short.plt > 20070813_short_utf8.plt`
The result produced by gpsbabel is back to ISO-8859-1 ???

tsteven4, 2019-03-04:
This may be a current bug.  I'll look further.

tsteven4, 2019-03-04:
I once asked

> I am a maintainer of gpsbabel and we are trying to clean up character 
> encoding issues.  I see the page 
> http://www.oziexplorer3.com/eng/help/fileformats.html and it's 
> explanation about commas, but I don't see any mention of what encoding 
> is used for text strings.  Is this encoding fixed?  If so, what is it 
> (utf-8, iso-8859-1, ...) If not, how do we know what encoding was used 
> for character strings in the .plt, .rte or .wpt files from ozi explorer? 

And the developer of ozi explore responded:

> There is no specified character set, it defaults to whatever 8 bit character set "Windows" defaults to - normally CP-1252 but can vary depending on Windows regional settings.
So we don't really have a way to tell how the strings are encoded in the 
.plt file.

Not knowing what to do we output Latin1.

jdpiguet, 2019-03-05:
I can understand that you don't know what to output.
My question is more to know why are you changing the content of the field (from UTF-8 to iso-8859-1) and not copying it "as is"?

tsteven4, 2019-03-05:
Internally most of our stringdata is stored as QStrings (UTF16).  For formats that have character strings we need to know how to convert those strings from/to UTF16.  

There is at least one other format where the encoding of character strings is variable, garmin_gpi.  In that format the encoding is stored in the file and we try to use that on read.  On write, we have an option to set the encoding.  The number of codecs available is limited.

For ozi it seems we need an option that could be used on read and write to specify the encoding.  We have codec for common encodings used for this format, e.g. CP1252, latin1, utf8.  On input a more knowledgeable use like yourself can use iconv, but on the output this doesn't work as the default latin1 encoding gpsbabel uses for this format is lossy.  If we adding an option to this format we could use it for input and output.

This comes up a few times a year.  

The Chief Babel-Head might take some convincing as we have been trying to get out of the encoding/decoding business.

tsteven4, 2019-03-05:
@jdpiguet , I have a lot of this working but I will need some new test cases.  I will send you some specific requests.  We need to beef up the test cases to make sure I didn't clobber the output when I converted it to a stream.

tsteven4, 2019-03-05:
@jdpiguet, I can get pretty good coverage using your file 20070813_short.plt.  Will you give us permission to use that file as a reference.  If we use it as a reference it will be publicly available.

The 5th line of that file seems to have an extra field.  According to https://www.oziexplorer4.com/eng/help/fileformats.html that line should have 8 fields, not 9.

jdpiguet, 2019-03-06:
@tsteven4 No problem for using the file.
For your information I found some older files with different headers, so most probably, the format has evolved over time and OziExplorer versions.
If you need them, I can dig in my archives and try to map files to Ozi versions...

robertlipe, 2019-03-06:
Can we just always write utf8?

tsteven4, 2019-03-06:
I don' t think we can always use utf-8 on write.  If the output is being 
used by OZI explorer, and the character set OZI is using is not utf-8, 
then I think we would cause problems.  In the example that led to this 
issue the encoding appears to be windows-1252.

The character encoding of the file is indeterminate according to the OZI 
explorer developer:

> There is no specified character set, it defaults to whatever 8 bit character set "Windows" defaults to - normally CP-1252 but can vary depending on Windows regional settings.
PR soon, QTextStream, option to set codec for read & write, 
encoding/decoding is trivial, default codec windows-1252.

There is also a bit we have never supported.  In some text fields:

> comma's not allowed in text fields, character 209 can be used instead 
> and a comma will be substituted.
This would seem to preclude utf-8.  I don't intent to start supporting 
character 209 remapping.

tsteven4, 2019-03-06:
this should be fixed in #315.

jdpiguet, 2019-03-09:
I have compiled and tested it: in my case, the problem is solved.
The line 5 is now copied "as is" from input to output...
Thanks, Jacques-D.

tsteven4, 2019-03-10:
resolved by #315.
The default codec for ozi is now windows-1252.  There is an option to set the codec for input and codec for output if your files use another encoding.

closed 2019-03-10 by tsteven4