GPSBabel/gpsbabel #330: GPSBabel doesn't want to launch under Windows 10

darekk2, 2019-03-25:
gpsbabelfe.exe. First time only the hourglass appears for a moment, next times no reaction at all. I reinstalled it few times, in different compatibility modes, but tast did not help. Maybe it was working few years ago, but something happened.
gpsbabel.exe can be launched.

robertlipe, 2019-03-25:
With thousands of new users a day, it's certain someone would notice. Far
more likely, something unique to your system is confusing things. On
Windows, suspect #1 is always antivirus and malware. Unfortunately, each
pretends to be the other.

Good luck debugging your system

darekk2, 2019-03-25:
But maybe it is possible to recognize that, for example using software detecting events or something.

robertlipe, 2019-03-25:
Our logging of such things is indeed showing that we have thousands of
successful Windows users each day, including this one. If something
systemic had changed, our lists would be full of screaming people.

We can't detect all the possible ways that antivirus software in particular
can damage the operation of our programs.  A/V is a common problem with the
geocaching GSAK, for example, which uses the same kind of a model of a main
GUI program that, amongst many other things, calls to GPSBabel for some of
the heavy lifting, much like our own GUI does. Stupid AV programs seem to
not like that and it's a leading source of support problems for them.
https://gsak.net/board/index.php?showtopic=32142
https://gsak.net/board/index.php?showtopic=34062

Nothing interesting seems to have changed on our end and it doesn't seem
like Microsoft has published an update to Windows 10 that's sent us down in
flames. That's why I'm suggesting you dig deeper into your own system.

Perhaps someone more in tune to the heartbeat of Windows 10 can offer
advice on the particular breed of goat to sacrifice. :-)

RJL

darekk2, 2019-03-25:
Maybe next version of the GPSBabel will be recognized by my system in different way. But it is not appearing from a long time.
Thank you for your answer.

tsteven4, 2019-03-25:
You can test something close to the next version of GPSBabel by trying a 
"Continuous build" from https://github.com/gpsbabel/gpsbabel/releases.

I can't remember anything we have changed that might influence the 
behavior on your system, but please let us know if the Continuous build 
solves your problem.

darekk2, 2019-04-21:
I didn't try that but GPS Babel installed with
https://www.gpsbabel.org/download.html
GPSBabel-1.6.0-beta20190414-Setup.exe
launches normally in Windows 10.

gpsbabel, 2019-04-21:
User reports new version working. Thanks.

closed 2019-04-21 by gpsbabel