GPSBabel/gpsbabel #335: Notarization Required for Mac Apps Created With New Developer IDs Starting in macOS 10.14.5

gpsbabel, 2019-04-08:
Todo: figure out this year's version of letting Apple charge for developer keys. 

https://www.macrumors.com/2019/04/08/mac-apps-notarization-macos-10-14-5/

gpsbabel, 2020-05-19:
Note to myself: this was punted until "Jan 2020", but doesn't seem to have actually impacted us.

Maybe NOT having a developer key was actually the winning move for us.

federicocappelli, 2022-05-19:
This is definitely a problem
[image]
GPSBabel-1.8.0 doesn't work at all on macOS 12.x (and probably 11.x), the DMG need to be notarised by apple.
I would be happy to help if needed (my [LinkedIn](https://www.linkedin.com/in/cappellifederico/))

GPSBabelDeveloper, 2022-05-19:
This has been the warning from gatekeeper for five years or more. You just
have to go into security and click accept. We've had probably a 100k Mac
installs get through this. This isn't the hard stop they announced all that
time ago.

But if you're able too submit a CL to sign our binaries and help us through
the process with apple tools, that would be super.

tsteven4, 2022-05-19:
I click on the DMG, drag GPSBabelFE to the Applications Folder, right 
click and select open.  The first time the options are "Move to Trash" 
and  "Cancel".  The second time the options include "Open".  After that 
you can just click on the GPSBabelFE icon. This is with macOS Monterey, 
12.4.

federicocappelli, 2022-05-19:
Turns out I have a system profile that stops me to do that -_- my bad, anyway the notarization process should be quite straight forward (with a macOS and apple developer program)

```
xcrun notarytool submit OvernightTextEditor_11.6.8.zip
                   --keychain-profile "AC_PASSWORD"
                   --wait
```

from https://developer.apple.com/documentation/security/notarizing_macos_software_before_distribution/customizing_the_notarization_workflow