GPSBabel/gpsbabel #337: serial port identification

tsteven4, 2019-04-16:
Can gui/serial_mac.cc, gui/serial_unix.cc, gui/serial_win.cc all be replaced with
```
#include <QtCore/QList>                    // for QList
#include <QtSerialPort/QSerialPortInfo>    // for QSerialPortInfo
#include <QtWidgets/QComboBox>             // for QComboBox

#include "mainwindow.h"

void MainWindow::osLoadDeviceNameCombos(QComboBox* box)
{
  const auto ports = QSerialPortInfo::availablePorts();
  for (const auto& info : ports) {
    box->addItem(info.systemLocation());
  }
}
```
This has the potential to resolve [https://bugzilla.redhat.com/show_bug.cgi?id=1337062](url)

robertlipe, 2019-04-16:
Yes. See
https://github.com/gpsbabel/gpsbabel/pull/42#diff-for-comment-58104108

I meant to do this years ago and, in fact, thought I had.

tsteven4, 2019-04-16:
On Linux (ubuntu bionic) it looks like we need systemLocation() to get things like /dev/ttyS0 instead of ttyS0, but on windows systemLocation gives things like \\.\COM1, and we probably want portName() instead.

robertlipe, 2019-04-17:
That's kind of gross. The qt doc is uncharacteristically unhelpful on the
difference.

ISTR that once the "X" in "COMx:" grew outside the bounds of [1-9]
 (somewhere around Windows 3.0) that they added the \.\COMx convention
<https://support.microsoft.com/en-us/help/115831/howto-specify-serial-ports-larger-than-com9>
so maybe this is something that Windows people just accept as normal these
days. That's when they added the whole notion of "friendly names" so you
could pick "3Com Modem" instead of "COM2" and certainly the common use of
BT and USB "serial" devices really doesn't fit well into the "COMx:"
convention. So how about we display one but use the other?

addItem(info.portName(), QVariant(info.systemLocation));

Then when we pull the name out of the array we read
box->currentIndex()->toString() in the onChange reader of the slot or
whatever.

That way we display the pretty names for humans but use the /dev and
"\.\COM" style names for the actual command line and resulting calls to
open ... though I couldn't fake shock if open() and CreateFile don't do the
same thing on Windows.

Man, so much for our awesome two line refactoring...

tsteven4, 2019-04-17:
I have something to test, but no means to really test it.

On windows the GUI looks like this, this is in a VM with nothing connected to COM1 or COM2.

[image]

tsteven4, 2019-04-17:
this doesn't work on windows for at least a few reasons.
1) in gbser_win.cc DEV_PREFIX is wrong.  To fix

```
-#define DEV_PREFIX "\\\\.\\\\"
+#define DEV_PREFIX "\\\\.\\"
```

2) after the above fix we don't handle names that start with DEV_PREFIX correctly, we prepend another DEV_PREFIX

```
"C:\Program Files (x86)\GPSBabel\gpsbabel.exe" -D 3 -i mtk -f \\.\com10
GPSBabel Version: 1.6.0-beta20190413
main: Compiled with Qt 5.12.2 for architecture i386-little_endian-ilp32
main: Running with Qt 5.12.2 on Windows 10 (10.0), x86_64
main: QLocale::system() is en_US
main: QLocale() is en_US
main: QTextCodec::codecForLocale() is System, mib 0
options: module/option=value: mtk/erase="0" (=default)
options: module/option=value: mtk/erase_only="0" (=default)
options: module/option=value: mtk/log_enable="0" (=default)
options: module/option=value: mtk/block_size_kb="1" (=default)
Opening port \\.\com10...
Translated port name: "\\.\\\.\com10"
mtk_logger: Can't initialise port "\\.\com10" (No error)
```

tsteven4, 2021-11-05:
fixed in #755

closed 2021-11-05 by tsteven4