GPSBabel/gpsbabel #338: fopen64 used incorrectly on NetBSD

gdt, 2019-04-17:
In the 1.6.0 beta, building under NetBSD leads to an error about fopen64 being undefined.  It is, of course, because off_t is big enough and there is no need for separate calls, and this is the situation on a number of systems -- but I don't have a list handy.

The problem is in zlib/contrib/minizip/ioapi.c, where the conditional
{{{
#if defined(__APPLE__) || defined(IOAPI_NO_64) || defined(__CYGWIN__)
}}}
doesn't include __NetBSD__, and probably __FreeBSD__ and others.

zlib ought to be using configure or similar to test for this, rather than hard-coding a list of operating systems, but that seems a big change.

I wonder if the overall configure can define IOAPI_NO_64 if fopen64 is not found?
Another approach might be to replace the conditional with a list of systems known to have fopen64.  Or some LARGEFILE define that is expected to be present if these functions exist.

No patch provided because the hard part is choosing an approach.  Also, there seems to be a patch in the gpsbabel distfile.

tsteven4, 2019-04-17:
I submitted #339 to avoid including minizip, we never completed the plans to utilize it so it is currently unnecessary.  It was only compiled in the configure flow, and was never included in the qmake flow.

zlib/contrib/minizip appears to be unmaintained.  If we need minizip in the future we may want to consider using https://github.com/nmoinvaz/minizip.

tsteven4, 2019-05-02:
Release 1.6.0 is available at https://www.gpsbabel.org/download.html.

The minizip source code is still in the tarball, but it is no longer compiled.  Thus your minizip patch will run, but it isn't necessary.

The tarball on www.gpsbabel.org will unpack into a directory gpsbabel-1.6.0.

Thanks for your testing and feedback on the beta.

closed 2019-05-02 by tsteven4