GPSBabel/gpsbabel #350: Garmin resettime called before gps_date_time_transfer variable is set so always fails

rlsolomon, 2019-05-13:
In garmin.cc main() the snippet
 
 if (resettime) {
    GPS_Command_Send_Time(qPrintable(fname), current_time().toTime_t());
    return;
  }

is called long before GPS_Init is called (3 if clauses lower)

Unfortunately, in gpscom.cc, the GPS_Command_Send_Time() function first checks that global variable gps_date_time_transfer has value pA600.  This cannot be true, as gps_date_time_transfer is first set in function GPS_A000 in gpsapp.cc - where it's set to -1, then later (potentially) overwritten by function GPS_A001 in gpsapp.cc called from within GPS_A000.  GPS_A000 is called from GPS_Init.

So resettime never works, even for units supporting pA600 because gps_date_time_transfer hasn't been set when the value is checked.  It seems like moving the resettime check *AFTER* the GPS_Init() call would solve the problem?

robertlipe, 2019-05-13:
It's entirely possible that doesn't work. It's a super-esoteric "feature"
added over ten years ago and exists solely as a workaround for a rarely
encountered GPS bug. It's entirely possible that hunk got moved to the
wrong place over the years. That flag was kind of a stab in the dark
anyway; it's not like it's a suggested thing in their protocol and GPSes
that need this are kind of inherently in a semi-broken state anyway.

If you move that down and confirm that it works for you, please send us a
pull request and we'll merge it. I agree with your analysis that Send_Time
before Gps_Init is probably bad but there may be something in _Init (which
causes an avalanche of traffic on the connection) that falls off the track
if the GPS date is super hosed.

rlsolomon, 2019-05-13:
Yeah, the bug seems to be back - or at least something which manifests itself similarly is back, presumably due to the 2nd week rollover, but other folks online have had the same problem for 8-9 years, so possibly back to the 1st week rollover?  In any case, I'm a Windows guy these days and don't have an environment to rebuild - even just initializing gps_date_time_transfer in main() or somewhere might be enough.  I'll try and throw together a Linux box and build there as that seems simpler than getting a working Windoze environment going.  Garmin claimed their original settime.exe utility "no longer works" but no details on why, so it's certainly possible this is a new firmware bug and not actually the clock going psycho as the original seemed to be...

rlsolomon, 2019-05-14:
I hacked up a local copy to just blindly send the timeset command but I don't think the GPS is accepting it - I also kludged the time send to change a few days in hopes of seeing the GPS show ANY different date/time...no joy there either.  WIthout a GPS known to accept and respond to the time set command, I'm not sure I can make any more progress.

TX [20]:14 00 00 00 00 00 00 00 08 00 00 00 05 0e 07 e3 00 0a 13 20 ....................(CMDDAT  Unknown)
RX (bulk) [0]:(RET2INTR)
RX (intr) [-110]:(CMDDAT  Abort)

robertlipe, 2019-05-14:
That's a trace from a USB device. We've only heard the problem - and this,
perhaps superstitious "fix" - on the serial era devices.

rlsolomon, 2019-05-14:
I'll have to go back and search more to see if I can find anyone claiming the garmin timeset util worked on a USB device.  I thought so, but given the zillions of google pages I've been through lately, I could easily have missed that the units where folks reported success were all serial vs USB.  I definitely found more users seeking the tool than reporting having used it, so superstitious may be pretty close to a dead-on description!

It's entirely possible Garmin's statement that the utility "no longer works" means current firmware on all units no longer accepts the timeset command...  I can confirm mine reports having the A600/D600 capability, not that that means much of course!

Unit:   eTrex Vista HCx Software Version 3.30
ID:     694
Version:        3.30RX (bulk) [65]:14 00 00 00 f8 00 00 00 35 00 00 00 56 45 52 47 50 52 4f 4d 20 4e 6f 6e 65 00 47 50 53 20 4d 54 33 33 31 38 20 30 2e 39 36 56 20 53 6f 66 74 77 61 72 65 20 56 65 72 73 69 6f 6e 20 32 2e 39 30 00 ........5...VERGPROM.None.GPS.MT3318.0.96V.Software.Version.2.90.(UNKNOWN )
RX (bulk) [150]:14 00 00 00 fd 00 00 00 8a 00 00 00 50 00 00 4c 01 00 41 0a 00 54 01 00 41 64 00 44 6e 00 41 c9 00 44 ca 00 44 6e 00 44 d2 00 41 2d 01 44 38 01 44 2e 01 41 90 01 44 6e 00 41 f4 01 44 f5 01 41 58 02 44 58 02 41 59 02 44 59 02 41 bc 02 44 bc 02 41 20 03 44 20 03 41 21 03 44 21 03 41 84 03 41 86 03 41 87 03 41 88 03 41 89 03 44 84 03 41 8b 03 44 8b 03 44 8c 03 44 8d 03 44 8e 03 41 8c 03 44 8f 03 41 92 03 41 94 03 41 95 03 44 95 03 41 96 03 44 96 03 ............P..L..A..T..Ad.Dn.A..D..Dn.D..A..D8.D..A..Dn.A..D..AX.DX.AY.DY.A..D..A..D..A..D..A..A..A..A..A..D..A..D..D..D..D..A..D..A..A..A..D..A..D..(UNKNOWN )

Capability A10:
Capability A100: D110
Capability A201: D202 D110 D210
Capability A301: D312 D302
Capability A400: D110
Capability A500: D501
Capability A600: D600
Capability A601: D601
Capability A700: D700
Capability A800: D800
Capability A801: D801
Capability A900:
Capability A902:
Capability A903:
Capability A904:
Capability A905: D900
Capability A907: D907 D908 D909 D910
Capability A908: D911
Capability A914:
Capability A916:
Capability A917: D917
Capability A918: D918

tsteven4, 2019-08-01:
https://github.com/gpsbabel/gpsbabel/commit/14108999a53d1e870dadc63743c0c28a3f796735 helped get my GPS12 into the correct epoch.  Previously I had gone through autolocate to 3D and let it sit for hours outside, but the date was stuck in 1999.  I tried this code and the date came up to 2019, but it seems I had to do an autolocate again.   I am wondering if sending time and sending position without first sending an almanac invalidates the existing almanac.

There is a potential problem I have observed with this setup:  GPS12 <-> usb serial adapter <-> windows 10 host <-> virtual box unbutu bionic guest.  If I run gpsbabel in the guest and have virtualbox pass through the com port the device usually communicates OK.  But I have seen the "CMDDAT  Xfer Posn" request get NAK'd.  If this happens then gps_save_lat and gps_save_lon are zero, and a bad position will get sent when we reset the time.  Since I had to autolocate anyway maybe it isn't a big deal.  On the other hand we may want to detect we got NAK'd and retry or abort, or at least check that we have a position before we send it.

With the setup GPS12 <-> usb serial adapter <-> windows 10 I haven't observed any NAKs.

Here is the trace.  I edited out the positions (*****).  The error exit is due to a write permission issue in the writer and can be ignored.

gpsbabel -D9 -w -i garmin,get_posn=1,resettime=1 -f COM3 -o gpx -F xx.gpx
GPSBabel Version: 1.6.0 
main: Compiled with Qt 5.12.4 for architecture i386-little_endian-ilp32
main: Running with Qt 5.12.4 on Windows 10 (10.0), x86_64
main: QLocale::system() is en_US
main: QLocale() is en_US
main: QTextCodec::codecForLocale() is System, mib 0
options: module/option=value: garmin/get_posn="1"
options: module/option=value: garmin/resettime="1"
Opening COM3
Tx Data:10 fe 00 02 10 03 : ...(PRDREQ  )
Rx Data:10 06 02 fe 00 fa 10 03  .. (ACK     )
Rx Data:10 ff 1b 57 00 cc 01 47 50 53 20 31 32 20 53 4f 46 54 57 41 52 45 20 20 34 2e 36 30 20 00 a2 10 03  W...GPS.12.SOFTWARE..4.60.. (PRDDAT  )
Tx Data:10 06 02 ff 00 f9 10 03 : .....(ACK     )
Unit:	GPS 12 SOFTWARE  4.60 
ID:	87
Version:	4.60Rx Data:10 fd 3c 50 00 00 4c 01 00 41 0a 00 41 64 00 44 67 00 41 c8 00 44 c9 00 44 67 00 41 2c 01 44 2c 01 41 90 01 44 93 01 41 f4 01 44 f5 01 41 58 02 44 58 02 41 bc 02 44 bc 02 41 20 03 44 20 03 d0 10 03  P..L..A..Ad.Dg.A..D..Dg.A..D..A..D..A..D..AX.DX.A..D..A..D.. (UNKNOWN )
Tx Data:10 06 02 fd 00 fb 10 03 : .....(SESACK  )

Capability A10:
Capability A100: D103
Capability A200: D201 D103
Capability A300: D300
Capability A400: D403
Capability A500: D501
Capability A600: D600
Capability A700: D700
Capability A800: D800
Link_type 1  Device_command 0
Waypoint: Transfer 100 Type 103
Route:    Transfer 200 Header 201 Type 103
Track:    Transfer 300 Type 300
Tx Data:10 0a 02 32 00 c2 10 03 : 2....(CMDDAT  Xfer PVT Stop)
Rx Data:10 06 02 0a 00 ee 10 03  .. (ACK     )
Opening COM3
Tx Data:10 0a 02 05 00 ef 10 03 : .....(CMDDAT  Xfer Time)
Rx Data:10 06 02 0a 00 ee 10 03  .. (ACK     )
Rx Data:10 0e 0c 0c 10 10 cf 07 14 00 24 18 1e 99 4b 00 a2 10 03  ..........K. (DATTIM  )
Tx Data:10 06 02 0e 00 ea 10 03 : .....(ACK     )
Opening COM3
Tx Data:10 0a 02 02 00 f2 10 03 : .....(CMDDAT  Xfer Posn)
Rx Data:10 06 02 0a 00 ee 10 03  .. (ACK     )
Rx Data:10 11 10 10 ***** 10 03  .....T.......d.. (POS     )
Tx Data:10 06 02 11 00 e7 10 03 : .....(ACK     )
Issuing Time Reset...
Opening COM3
Tx Data:10 0e 08 08 01 e3 07 0e 00 24 15 b0 10 03 : ...........(DATTIM  )
Rx Data:10 06 02 0e 00 ea 10 03  .. (ACK     )
Rx Data:10 0a 02 02 00 f2 10 03  .. (CMDDAT  Xfer Posn)
Tx Data:10 06 02 0a 00 ee 10 03 : .....(ACK     )
INFO: GPS position request. Sending....Tx Data:10 11 10 10 ***** 10 03 : ......T.......d.....(POS     )
Rx Data:10 06 02 11 00 e7 10 03  .. (ACK     )
done.
Waypoint type: 103
Chosen waypoint length 6
Cannot open 'xx.gpx' for write.  Error was 'Access is denied.'.
cet_util: Converting from "US-ASCII" to "UTF-8", done.
options: module/option=value: gpx/snlen="32" (=default)
options: module/option=value: gpx/elevprec="3" (=default)

Error running gpsbabel: Process exited unsuccessfully with code 1

gpsbabel, 2020-05-19:
Fixed in https://github.com/gpsbabel/gpsbabel/commit/14108999a53d1e870dadc63743c0c28a3f796735

closed 2020-05-19 by gpsbabel