GPSBabel/gpsbabel #355: Warnings again...

rickrich, 2019-05-17:
$ make
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS main.cc -o main.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS globals.cc -o globals.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS route.cc -o route.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS waypt.cc -o waypt.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS filter_vecs.cc -o filter_vecs.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS util.cc -o util.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS vecs.cc -o vecs.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS mkshort.cc -o mkshort.o
mkshort.cc: In function ‘char* mkshort_add_to_list(mkshort_handle_imp*, char*)’:
mkshort.cc:148:28: warning: ‘%d’ directive writing between 1 and 11 bytes into a region of size 9 [-Wformat-overflow=]
     int dl = sprintf(tbuf, ".%d", s->conflictctr);
                            ^~~~~
mkshort.cc:148:28: note: directive argument in the range [-2147483647, 2147483647]
mkshort.cc:148:21: note: ‘sprintf’ output between 3 and 13 bytes into a destination of size 10
     int dl = sprintf(tbuf, ".%d", s->conflictctr);
              ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS csv_util.cc -o csv_util.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS grtcirc.cc -o grtcirc.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS xmlgeneric.cc -o xmlgeneric.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS formspec.cc -o formspec.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS xmltag.cc -o xmltag.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS cet.cc -o cet.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS cet_util.cc -o cet_util.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS fatal.cc -o fatal.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS rgbcolors.cc -o rgbcolors.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS inifile.cc -o inifile.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS garmin_fs.cc -o garmin_fs.o
In file included from garmin_fs.cc:25:
garmin_fs.cc: In function ‘void garmin_fs_garmin_before_write(const Waypoint*, GPS_PWay, int)’:
garmin_fs.h:61:63: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 2 equals destination size [-Wstringop-truncation]
 #define GMSD_GETNSTR(a,b,c) if (gmsd && gmsd->flags.a) strncpy((b),gmsd->a,(c))
                                                        ~~~~~~~^~~~~~~~~~~~~~~~~
garmin_fs.cc:481:3: note: in expansion of macro ‘GMSD_GETNSTR’
   GMSD_GETNSTR(cc, way->cc, sizeof(way->cc));
   ^~~~~~~~~~~~
garmin_fs.h:61:63: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 24 equals destination size [-Wstringop-truncation]
 #define GMSD_GETNSTR(a,b,c) if (gmsd && gmsd->flags.a) strncpy((b),gmsd->a,(c))
                                                        ~~~~~~~^~~~~~~~~~~~~~~~~
garmin_fs.cc:482:3: note: in expansion of macro ‘GMSD_GETNSTR’
   GMSD_GETNSTR(city, way->city, sizeof(way->city));
   ^~~~~~~~~~~~
garmin_fs.h:61:63: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 2 equals destination size [-Wstringop-truncation]
 #define GMSD_GETNSTR(a,b,c) if (gmsd && gmsd->flags.a) strncpy((b),gmsd->a,(c))
                                                        ~~~~~~~^~~~~~~~~~~~~~~~~
garmin_fs.cc:483:3: note: in expansion of macro ‘GMSD_GETNSTR’
   GMSD_GETNSTR(state, way->state, sizeof(way->state));
   ^~~~~~~~~~~~
garmin_fs.h:61:63: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 32 equals destination size [-Wstringop-truncation]
 #define GMSD_GETNSTR(a,b,c) if (gmsd && gmsd->flags.a) strncpy((b),gmsd->a,(c))
                                                        ~~~~~~~^~~~~~~~~~~~~~~~~
garmin_fs.cc:484:3: note: in expansion of macro ‘GMSD_GETNSTR’
   GMSD_GETNSTR(facility, way->facility, sizeof(way->facility));
   ^~~~~~~~~~~~
garmin_fs.h:61:63: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 52 equals destination size [-Wstringop-truncation]
 #define GMSD_GETNSTR(a,b,c) if (gmsd && gmsd->flags.a) strncpy((b),gmsd->a,(c))
                                                        ~~~~~~~^~~~~~~~~~~~~~~~~
garmin_fs.cc:485:3: note: in expansion of macro ‘GMSD_GETNSTR’
   GMSD_GETNSTR(cross_road, way->cross_road, sizeof(way->cross_road));
   ^~~~~~~~~~~~
garmin_fs.h:61:63: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 52 equals destination size [-Wstringop-truncation]
 #define GMSD_GETNSTR(a,b,c) if (gmsd && gmsd->flags.a) strncpy((b),gmsd->a,(c))
                                                        ~~~~~~~^~~~~~~~~~~~~~~~~
garmin_fs.cc:486:3: note: in expansion of macro ‘GMSD_GETNSTR’
   GMSD_GETNSTR(addr, way->addr, sizeof(way->addr));
   ^~~~~~~~~~~~
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS units.cc -o units.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS gbser_posix.cc -o gbser_posix.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS gbser.cc -o gbser.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS gbfile.cc -o gbfile.o
[... 321 lines cut ...]
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS mapfactor.cc -o mapfactor.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS energympro.cc -o energympro.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS mynav.cc -o mynav.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS ggv_bin.cc -o ggv_bin.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS globalsat_sport.cc -o globalsat_sport.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS geojson.cc -o geojson.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS bend.cc -o bend.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS position.cc -o position.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS radius.cc -o radius.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS duplicate.cc -o duplicate.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS arcdist.cc -o arcdist.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS polygon.cc -o polygon.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS smplrout.cc -o smplrout.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS reverse_route.cc -o reverse_route.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS sort.cc -o sort.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS stackfilter.cc -o stackfilter.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS trackfilter.cc -o trackfilter.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS discard.cc -o discard.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS nukedata.cc -o nukedata.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H -DNEW_STRINGS interpolate.cc -o interpolate.o
[... cut ...] o pocketfms_wp.o naviguide.o enigma.o vpl.o teletype.o jogmap.o bushnell.o bushnell_trl.o wintec_tes.o subrip.o garmin_xt.o garmin_fit.o mtk_locus.o googledir.o mapbar_track.o f90g_track.o mapfactor.o energympro.o mynav.o ggv_bin.o globalsat_sport.o geojson.o bend.o position.o radius.o duplicate.o arcdist.o polygon.o smplrout.o reverse_route.o sort.o stackfilter.o trackfilter.o discard.o nukedata.o interpolate.o transform.o height.o swapdata.o validate.o  -lm  -L/usr/lib64 -lQt5Core  -o gpsbabel

gpsbabel, 2019-05-17:
This appears to be new in Fedora/Redhat's version of GCC.

I'm about ready to move -Wall out of the default flags and into debug flags. Most of the "discoveries" are busy work, (I don't care if a debug variable is set and not used) or invite low-quality fixes (you COULD just subtract one from the last argument of strncpy, but is it really going into a zero-terminated field of this protocol? Should the whole system be reworked to use a real string class?) tend to be OS-specific, and the number of reports we get about the warnings exceeds the number of people willing to work on them.

I'll prioritize this as low because it invites revisiting our stance on warnings. "Enable them all and try to fix them" isn't proving to be sustainable across environments.

Thanx for the report, Rick. Sorry that you're getting some hand-wringing...

rickrich, 2019-05-17:
I would try and fix ALL warnings if it was my project.

rickrich, 2019-05-17:
First one: char tbuf[10] should be, say, char tbuf[16].

rickrich, 2019-05-17:
Second one:
  GMSD_GETNSTR(cc, way->cc, sizeof(way->cc)-1);
  GMSD_GETNSTR(city, way->city, sizeof(way->city)-1);
  GMSD_GETNSTR(state, way->state, sizeof(way->state)-1);
  GMSD_GETNSTR(facility, way->facility, sizeof(way->facility)-1);
  GMSD_GETNSTR(cross_road, way->cross_road, sizeof(way->cross_road)-1);
  GMSD_GETNSTR(addr, way->addr, sizeof(way->addr)-1);

gpsbabel, 2020-05-19:
We've fixed hundreds (really!) of warnings in various environments and with lots of lint-like tools.

Right now, we build without a peep on my MBP.
% make -j10 > /dev/null
%

I think our automated builds are all silent, too. Rick, before we wrap 1.7, are we clean on whatever you're using?

tsteven4, 2020-05-19:
Its an ever changing goal, and we are not perfect.

We are certainly not clean on with g++ 9.3.0 on Ubuntu focal fossa.
We are almost but not quite clean with clang++ 10.0.0 on Ubuntu focal fossa.
I could add the above two cases to regression so they would be visible but not cause failures.

We are also not clean on windows with MSVC 2017 32 bit, which we use for release, let alone any MSVC 64 bit build (which we don't release).

robertlipe, 2020-05-20:
I suspect our downfall is using -Wall. A new warning is added in some
combination of compiler version and OS environment that we're not using and
someone like Rick is just the first one to ever see that combination. I
wonder if our default build should instead use some fixed set of flags that
we agree upon and cast in stone - and for it, we might even consider
-Werror -  and only our debug build use -Wall for all the chatter. That
might provide a better first impression while still providing actual signal
over noise. A huge percentage of signed comparison warnings, for example,
are just annoying noise for us.

I agree with Rick's perception that a default build that's full of warnings
looks sloppy. While we pull at this string All The Time, we're a finite
group. Should we perhaps lower the expectations?

rickrich, 2020-05-20:
$ cat xxx
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H main.cc -o main.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H globals.cc -o globals.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H route.cc -o route.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H waypt.cc -o waypt.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H filter_vecs.cc -o filter_vecs.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H util.cc -o util.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H vecs.cc -o vecs.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H mkshort.cc -o mkshort.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H csv_util.cc -o csv_util.o
gcc  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H strptime.c -o strptime.o
strptime.c: In function ‘strptime_internal’:
strptime.c:289:21: warning: variable ‘era’ set but not used [-Wunused-but-set-variable]
  289 |   struct era_entry *era;
      |                     ^~~
strptime.c:277:15: warning: variable ‘rp_backup’ set but not used [-Wunused-but-set-variable]
  277 |   const char *rp_backup;
      |               ^~~~~~~~~
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H grtcirc.cc -o grtcirc.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H xmlgeneric.cc -o xmlgeneric.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H formspec.cc -o formspec.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H xmltag.cc -o xmltag.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H cet.cc -o cet.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H cet_util.cc -o cet_util.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H fatal.cc -o fatal.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H rgbcolors.cc -o rgbcolors.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H inifile.cc -o inifile.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H garmin_fs.cc -o garmin_fs.o
garmin_fs.cc: In function ‘void garmin_fs_garmin_before_write(const Waypoint*, GPS_PWay, int)’:
garmin_fs.cc:381:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 2 equals destination size [-Wstringop-truncation]
  381 |   strncpy(way->cc, STRFROMUNICODE(garmin_fs_t::get_cc(gmsd, nullptr)), sizeof(way->cc));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
garmin_fs.cc:382:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 24 equals destination size [-Wstringop-truncation]
  382 |   strncpy(way->city, STRFROMUNICODE(garmin_fs_t::get_city(gmsd, nullptr)), sizeof(way->city));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
garmin_fs.cc:383:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 2 equals destination size [-Wstringop-truncation]
  383 |   strncpy(way->state, STRFROMUNICODE(garmin_fs_t::get_state(gmsd, nullptr)), sizeof(way->state));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
garmin_fs.cc:384:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 32 equals destination size [-Wstringop-truncation]
  384 |   strncpy(way->facility, STRFROMUNICODE(garmin_fs_t::get_facility(gmsd, nullptr)), sizeof(way->facility));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
garmin_fs.cc:385:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 52 equals destination size [-Wstringop-truncation]
  385 |   strncpy(way->cross_road, STRFROMUNICODE(garmin_fs_t::get_cross_road(gmsd, nullptr)), sizeof(way->cross_road));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
garmin_fs.cc:386:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 52 equals destination size [-Wstringop-truncation]
  386 |   strncpy(way->addr, STRFROMUNICODE(garmin_fs_t::get_addr(gmsd, nullptr)), sizeof(way->addr));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H units.cc -o units.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H gbser_posix.cc -o gbser_posix.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H gbser.cc -o gbser.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H gbfile.cc -o gbfile.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H parse.cc -o parse.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H session.cc -o session.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H src/core/textstream.cc -o src/core/textstream.o
g++  -g -O2 -Wall -fPIC -c   -I. -I./zlib -I/usr/include/qt5  -DHAVE_CONFIG_H src/core/usasciicodec.cc -o src/core/usasciicodec.o
[... cut ...] adius.o duplicate.o arcdist.o polygon.o smplrout.o reverse_route.o sort.o stackfilter.o trackfilter.o discard.o nukedata.o interpolate.o transform.o height.o swapdata.o validate.o  -lm  -L/usr/lib64 -lQt5Core  -o gpsbabel
/usr/bin/ld: jeeps/gpsdevice_usb.o: in function `gdu_on(char const*, gpsdevh**)':
/home/rick/proj/gpsbabel/jeeps/gpsdevice_usb.cc:37: undefined reference to `gusb_init(char const*, gpsdevh**)'
collect2: error: ld returned 1 exit status
make: *** [Makefile:156: gpsbabel] Error 1

tsteven4, 2020-05-21:
with gcc 9.3.0 on ubuntu focal:
      1 [-Wformat-truncation=]
      3 [-Wsizeof-pointer-memaccess]
     23 [-Wstringop-truncation]
      2 [-Wunused-but-set-variable]
      1 [-Wunused-result]

The format-truncation error is in shapelib, and I loathe to make changes in third part code.  The unused-but-set-variable errors are in strptime, same comment.

Tempting to try -Wall -Wno-stringop-truncation.  This will lead to new warnings like [-Wunknown-warning-option].

rickrich, 2020-05-21:
For example, in bushnell.cc you have:

  strncpy(tbuf, CSTRc(wpt->shortname), sizeof(tbuf));
  tbuf[sizeof(tbuf)-1] = 0;

but it should be:
  strncpy(tbuf, CSTRc(wpt->shortname), sizeof(tbuf)-1);
  tbuf[sizeof(tbuf)-1] = 0;

-Wall is right. The programmer is wrong.

tsteven4, 2020-05-21:
which assumes that the binary format expects a fixed length null-terminated string.  It may, the comment a few lines above seems to indicate that.  But if it is a fixed length string it doesn't really need a null terminator, so who knows.  And then we follow that field with two bytes of zero padding.

In many cases we just don't know what the binary format specification really is.  We may or may not have any real reference files.  Some users may or may not have used some of these formats in the last decade.

tsteven4, 2020-05-21:
but i will grant you that in the case you cited it is silly to write the last byte of tbuf twice so your proposal isn't any worse than what we have now.

tsteven4, 2020-05-21:
Three of the stringop-truncation warnings are silenced in #565.  In these cases it was apparently intended that the target be a null terminated string.  However, none of these were an actual  problem as they were explicitly terminated after the strncpy operation.

for gcc version >= 8, but not version 7, we could use a non-standard gcc specific attribute nonstring to let gcc know some character arrays are not intended to be null terminated.  In general we would rather avoid adding non-standard tool specific code.  Alternatively, memcpy could be used instead of strncpy.  Both of these methods are mentioned in the gcc manual.

robertlipe, 2020-05-21:
That's a very real problem. For A lot of these formats, we really don't
know for certain what the formats expect in the same way we often don't
know encoding.

Bushnell is a good example where I'd rather spend ten minutes deleting the
format than ten minutes reasoning out if a field is fixed length zero
terminated or padded. Bushnell hasn't made GPSes in years.

Formats like these or an1 are just a tarpit of maintenance costs for us. We
need better metrics for what's actually used and to be more ready to cut
dead weight.

We should also probably say no to some new formats until there's a proven
demand of user base...

Welcome to the curses of an 19 year old program.

tsteven4, 2020-05-21:
Agreed.  I suspect some of these formats nobody has used for the last 20 commits.

tsteven4, 2020-05-25:
The warnings from shapelib are fixed upstream but not released.
shpopen.c -Wformat-truncation=
dbfopen.c -Wstringop-truncation

Neither appears important.

rickrich, 2020-05-25:
Getting better!

$ make >xxx
strptime.c: In function ‘strptime_internal’:
strptime.c:289:21: warning: variable ‘era’ set but not used [-Wunused-but-set-variable]
  289 |   struct era_entry *era;
      |                     ^~~
strptime.c:277:15: warning: variable ‘rp_backup’ set but not used [-Wunused-but-set-variable]
  277 |   const char *rp_backup;
      |               ^~~~~~~~~
garmin_fs.cc: In function ‘void garmin_fs_garmin_before_write(const Waypoint*, GPS_PWay, int)’:
garmin_fs.cc:381:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 2 equals destination size [-Wstringop-truncation]
  381 |   strncpy(way->cc, STRFROMUNICODE(garmin_fs_t::get_cc(gmsd, nullptr)), sizeof(way->cc));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
garmin_fs.cc:382:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 24 equals destination size [-Wstringop-truncation]
  382 |   strncpy(way->city, STRFROMUNICODE(garmin_fs_t::get_city(gmsd, nullptr)), sizeof(way->city));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
garmin_fs.cc:383:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 2 equals destination size [-Wstringop-truncation]
  383 |   strncpy(way->state, STRFROMUNICODE(garmin_fs_t::get_state(gmsd, nullptr)), sizeof(way->state));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
garmin_fs.cc:384:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 32 equals destination size [-Wstringop-truncation]
  384 |   strncpy(way->facility, STRFROMUNICODE(garmin_fs_t::get_facility(gmsd, nullptr)), sizeof(way->facility));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
garmin_fs.cc:385:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 52 equals destination size [-Wstringop-truncation]
  385 |   strncpy(way->cross_road, STRFROMUNICODE(garmin_fs_t::get_cross_road(gmsd, nullptr)), sizeof(way->cross_road));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
garmin_fs.cc:386:10: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 52 equals destination size [-Wstringop-truncation]
  386 |   strncpy(way->addr, STRFROMUNICODE(garmin_fs_t::get_addr(gmsd, nullptr)), sizeof(way->addr));
      |   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function ‘void GPS_D120_Get(int, char*)’,
    inlined from ‘int32 GPS_A101_Get(const char*)’ at jeeps/gpsapp.cc:1182:19:
jeeps/gpsapp.cc:2038:12: warning: ‘char* strncpy(char*, const char*, size_t)’ output may be truncated copying 17 bytes from a string of length 1023 [-Wstringop-truncation]
 2038 |     strncpy(gps_categories[cat_num], s, sizeof(gps_categories[0]));
      |     ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
jeeps/gpscom.cc: In function ‘int32 GPS_Command_Send_Track_As_Course(const char*, GPS_STrack**, int32, GPS_SWay**, int32, int)’:
jeeps/gpscom.cc:1200:12: warning: ‘char* strncpy(char*, const char*, size_t)’ output may be truncated copying 15 bytes from a string of length 255 [-Wstringop-truncation]
 1200 |     strncpy(crs[n_crs]->course_name, trk[i]->trk_ident,
      |     ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1201 |             sizeof(crs[n_crs]->course_name)-1);
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
jeeps/gpscom.cc:1273:12: warning: ‘char* strncpy(char*, const char*, size_t)’ output may be truncated copying 10 bytes from a string of length 255 [-Wstringop-truncation]
 1273 |     strncpy(cpt[i+n_cpt]->name, wpt[i]->cmnt,
      |     ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1274 |             sizeof(cpt[i+n_cpt]->name) - 1);
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
shapelib/shpopen.c: In function ‘SHPReadObject’:
shapelib/shpopen.c:2433:21: warning: ‘. Probably broken SHP file’ directive output may be truncated writing 26 bytes into a region of size between 25 and 48 [-Wformat-truncation=]
 2433 |                     "Not enough memory to allocate requested memory (nPoints=%u, nParts=%u) for shape %d. "
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
shapelib/shpopen.c:2433:105: note: format string is defined here
 2433 |                     "Not enough memory to allocate requested memory (nPoints=%u, nParts=%u) for shape %d. "
      |                                                                                                         ^~~
 2434 |                     "Probably broken SHP file", nPoints, nParts, hEntity );
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~                                                            
shapelib/shpopen.c:2432:13: note: ‘snprintf’ output between 107 and 130 bytes into a destination of size 128
 2432 |             snprintf(szErrorMsg, sizeof(szErrorMsg),
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 2433 |                     "Not enough memory to allocate requested memory (nPoints=%u, nParts=%u) for shape %d. "
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 2434 |                     "Probably broken SHP file", nPoints, nParts, hEntity );
[... cut ...]
  632 |     printf(MYNAME "-offs 0x%08X: Next IFD=0x%08X\n", next_ifd_offs,  ifd->next_ifd);
      |     ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
humminbird.cc: In function ‘void humminbird_rte_tail(const route_head*)’:
humminbird.cc:878:12: warning: ‘char* strncpy(char*, const char*, size_t)’ specified bound 20 equals destination size [-Wstringop-truncation]
  878 |     strncpy(humrte->name, CSTR(name), sizeof(humrte->name));
      |     ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~