GPSBabel/gpsbabel #375: lowrance usr ver 4 reports as version 0 from plotter

erkswede, 2019-07-13:


erkswede, 2019-07-13:
[to-plotter.gpx.zip](https://github.com/gpsbabel/gpsbabel/files/3389082/to-plotter.gpx.zip)
Attaching the file I'm trying to convert.

tsteven4, 2019-07-13:
Please provide the command line you are using, the version of gpsbabel, 
and the platform.  If you are using the GUI the command line to gpsbabel 
is visible in the big box towards the bottom.

Thanks

erkswede, 2019-07-14:
> Please provide the command line you are using, the version of gpsbabel, and the platform.

I think it's all in my first comment but here it is again:

`gpsbabel -D 2 -i gpx -f plotter/to-plotter.gpx -o lowranceusr,wversion=4 -F to-plotter.usr`

`gpsbabel version 1.6.0`

`Ubuntu 16.04.6 LTS, x86_64`

Just let me know if there is anything else needed.

BTW, using `wversion=3` (or 2) works fine.

Thanks

tsteven4, 2019-07-14:
Thanks.  The automatic email forwarding of your original report omitted the quoted details.  This is a new bug introduced in 1.6.0.

PR #376 (2019-08-29): fix lowranceusr4 writer issues
This should resolve #375.

In addition, further writer issues with lowranceusr version 4 should be resolved:

- BOMs were erroneously included in internal UTF-16LE strings.
- Timestamps were erroneously offset by 12 hours.
- The written number of mysterious attribute bytes didn't match the written count.

Also, rounding is now used when converting real latitude and longitude numbers to the integer format used by all lowranceusr files.  This makes round trip testing much more likely to succeed.
closed 2019-08-29 by tsteven4