GPSBabel/gpsbabel #381: Garmin eTrex Vista HCX not detected on USB on Macos

turepalsson, 2019-07-31:
### Summary

Recently, probably after upgrading MacOS to 10.14.6 (from .5), gpsbabel (version 1.6; also tried from GIT gpsbabel_1_5_4-604-gafc73a03) no longer detects my Garmin eTrex Vista HCX on USB. Error message is "Found no Garmin USB devices".

Sample command line: `/Applications/GPSBabelFE.app/Contents/MacOS/gpsbabel -w -t -i garmin -f usb: -o gpx -F out.gpx`

### Output from a run with enviroment variable USB_DEBUG=99

    usb_set_debug: Setting debugging level to 99 (on)
    usb_os_find_busses: Found 001
    usb_os_find_devices: Found USB device at location 0x14000000
    usb_os_find_devices: Found 000-05ac-8007-09-ff on 001 at location 0x14000000
    usb_os_find_devices: Found USB device at location 0x14a00000
    usb_os_find_devices: Found 002-0bc2-2322-00-00 on 001 at location 0x14a00000
    usb_os_find_devices: Found USB device at location 0x14500000
    usb_os_find_devices: Found 003-05ac-8511-ef-02 on 001 at location 0x14500000
    usb_os_find_devices: Found USB device at location 0x14600000
    usb_os_find_devices: Found 004-0a5c-4500-09-00 on 001 at location 0x14600000
    usb_os_open: 0a5c:4500
    usb_control_msg: 128 6 512 0 0x7ffeee51ee20 8 1000
    usb_control_msg: 128 6 512 0 0x7fd947c1aab0 25 1000
    usb_os_close: 0a5c:4500
    usb_os_open: 05ac:8511
    usb_os_open(USBDeviceOpenSeize): another process has device opened for exclusive access
    usb_control_msg: 128 6 512 0 0x7ffeee51ee20 8 1000
    usb_control_msg: 128 6 512 0 0x7fd947d008c0 782 1000
    skipping descriptor 0xB
    skipped 1 class/vendor specific endpoint descriptors
    skipped 4 class/vendor specific interface descriptors
    skipping descriptor 0x25
    skipped 1 class/vendor specific endpoint descriptors
    skipped 18 class/vendor specific interface descriptors
    usb_os_close: 05ac:8511
    usb_os_open: 0bc2:2322
    usb_control_msg: 128 6 512 0 0x7ffeee51ee20 8 1000
    usb_control_msg: 128 6 512 0 0x7fd947d018a0 121 1000
    skipping descriptor 0x30
    skipped 1 class/vendor specific endpoint descriptors
    skipping descriptor 0x30
    skipped 1 class/vendor specific endpoint descriptors
    skipping descriptor 0x30
    skipping descriptor 0x24
    skipped 2 class/vendor specific endpoint descriptors
    skipping descriptor 0x30
    skipping descriptor 0x24
    skipped 2 class/vendor specific endpoint descriptors
    skipping descriptor 0x30
    skipping descriptor 0x24
    skipped 2 class/vendor specific endpoint descriptors
    skipping descriptor 0x30
    skipping descriptor 0x24
    skipped 2 class/vendor specific endpoint descriptors
    usb_os_close: 0bc2:2322
    usb_os_open: 05ac:8007
    usb_os_open(USBDeviceOpenSeize): another process has device opened for exclusive access
    usb_control_msg: 128 6 512 0 0x7ffeee51ee20 8 1000
    usb_control_msg: 128 6 512 0 0x7fd947c19c70 9 1000
    usb_os_close: 05ac:8007
    Found no Garmin USB devices.

### Output from system_profiler SPUSBDataType

_(Notice Garmin device at the end! Also notice error message at the beginning; I wonder if that's related and Apple bungled something...)_

    2019-07-31 17:04:12.209 system_profiler[798:30808] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
    USB:

    USB 3.0 Bus:

      Host Controller Driver: AppleUSBXHCILPTH
      PCI Device ID: 0x8c31 
      PCI Revision ID: 0x0005 
      PCI Vendor ID: 0x8086 

        Expansion:

          Product ID: 0x2322
          Vendor ID: 0x0bc2  (Seagate LLC)
          Version: 0.00
          Serial Number: NA8152T0
          Speed: Up to 5 Gb/sec
          Manufacturer: Seagate
          Location ID: 0x14a00000 / 2
          Current Available (mA): 900
          Current Required (mA): 144
          Extra Operating Current (mA): 0
          Media:
            Expansion:
              Capacity: 2 TB (2 000 398 933 504 bytes)
              Removable Media: No
              BSD Name: disk3
              Logical Unit: 0
              Partition Map Type: MBR (Master Boot Record)
              USB Interface: 0
              Volumes:
                Seagate Expansion:
                  Capacity: 2 TB (2 000 396 321 280 bytes)
                  Available: 481,82 GB (481 821 736 960 bytes)
                  Writable: Yes
                  File System: Journaled HFS+
                  BSD Name: disk3s1
                  Mount Point: /Volumes/Seagate Expansion
                  Content: Apple_HFS
                  Volume UUID: 2EE71CA1-B2C9-300D-A278-0CBB1C32AD76

        BRCM20702 Hub:

          Product ID: 0x4500
          Vendor ID: 0x0a5c  (Broadcom Corp.)
          Version: 1.00
          Speed: Up to 12 Mb/sec
          Manufacturer: Apple Inc.
          Location ID: 0x14600000 / 4
          Current Available (mA): 500
          Current Required (mA): 94
          Extra Operating Current (mA): 0
          Built-In: Yes

            Bluetooth USB Host Controller:

              Product ID: 0x828d
              Vendor ID: 0x05ac (Apple Inc.)
              Version: 1.50
              Manufacturer: Apple Inc.
              Location ID: 0x14630000

        FaceTime HD Camera (Built-in):

          Product ID: 0x8511
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 72.30
          Serial Number: CCGE9D01RBFRQNL0
          Speed: Up to 480 Mb/sec
          Manufacturer: Apple Inc.
          Location ID: 0x14500000 / 3
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
          Built-In: Yes

        Vendor-Specific Device:

          Product ID: 0x0003
          Vendor ID: 0x091e  (Garmin International)
          Version: 0.01
          Speed: Up to 12 Mb/sec
          Location ID: 0x14400000 / 8
          Current Available (mA): 500
          Extra Operating Current (mA): 0

### Tinkering

I tried checking the return value of IOCreatePlugInInterfaceForService in mac/libusb/darwin.c, and it returns 0xffffffffe00002be. I found [this stack overflow answer](https://stackoverflow.com/questions/48832072/iocreateplugininterfaceforservice-returns-mysterious-error) which mentions app sandboxing... I have no idea whether that's a red herring or not! :-)

tsteven4, 2019-07-31:
Interesting.  I think app sandboxing is likely a red herring. All the 
dmgs we have distributed are not sandboxed (yet).  So if you installed 
one of our dmgs from gpsbabel.org or 
github.com/gpsbabel/gpsbabel/releases it seems like it is something else.

You can check by finding the PID of the running gpsbabel app or command 
line tool and using the asctl command.

The gui.

> 1952:~ administrator$ asctl sandbox check --pid 5035
> /Users/administrator/Desktop/GPSBabelFE.app:
>     not signed with App Sandbox entitlements
>     running without App Sandbox enabled
>     running unsandboxed
The command line tool.  You can start it and get it to wait with by 
running gpsbabel with no parameters, it will wait with a [Press Enter] 
message.
> 1952:~ administrator$  asctl sandbox check --pid 5089
> /Users/administrator/Desktop/GPSBabelFE.app/Contents/MacOS/gpsbabel:
>     not signed with App Sandbox entitlements
>     running without App Sandbox enabled
>     running unsandboxed

turepalsson, 2019-07-31:
I checked the command-line versions, both from the distributed DMG and the binary I built myself from GIT, and both are reported as "unsandboxed" (sample from home-made one, but the one from the DMG gives identical output, except the path)

    $ asctl sandbox check --pid 2453
    /Users/ture/Desktop/gpsbabel/gpsbabel:
	    not signed with App Sandbox entitlements
	    running without App Sandbox enabled
	    running unsandboxed

tsteven4, 2019-07-31:
Ture,

Can you either run the testlibusb I built or build it from 
github.com/libusb/libusb and run it?  It is one of the libusb 1.0 
examples.  I want to see if it can find your garmin.  gpsbabel uses a 
modified version of the very old libusb 0.1.12.  If it finds it we 
should see something like "Dev (bus x, device y): 091e - zzzz", where 
0x091e is the Garmin vendor ID.

You can fetch my version by visiting 
https://filetransfer.io/data-package/2dZ7Pv7q

After you download the zip file you can check it:

> 1952:~ administrator$ openssl dgst -sha256 testlibusb.zip
> SHA256(testlibusb.zip)= 
> bcc14a64e9cfc6eeaffb7ec8c14b010c080f7aea78ea45555ca628dafc926f45

To run it:

> 1952:xxx administrator$ unzip testlibusb.zip
> Archive:  testlibusb.zip
>    creating: testlibusb/
>   inflating: testlibusb/libusb-1.0.1.dylib
>   inflating: testlibusb/testlibusb
> 1952:xxx administrator$ ./testlibusb/testlibusb
> Dev (bus 29, device 4): Apple, Inc. - IR Receiver
> Dev (bus 29, device 3): Apple Inc. - BRCM20702 Hub
> Dev (bus 29, device 2): 0424 - 2512
> Dev (bus 26, device 1): 8087 - 0024
> Dev (bus 29, device 1): 8087 - 0024
Steve

gpsbabel, 2019-07-31:
I don't have an HCX, but I have a Venture HC here in the GPSBabel-lab.
Modulo some silliness in the debug printing (patch coming) is seems OK to
me on my iMac.

The ugly squares just need some && isascii added. Actually, maybe I byte
the bullet and unify the four different places we print these lines...

This is 1.5.4 just to show the ugli ones as I've "fixed" my work tree to
not do that. But 1.6.0 from the master DMG works in the same way for me.

I don't see anything in the upstream libusb project that's pertinent. That
project was dead for a long time. Maybe I'll try grafting in a current
release (if they've not destroyed the API again) just to see if that brings
any happiness.

I don't *think* it's a sandboxing issue. We don't pretend to be able to
live in a sandbox - when you read and write arbitrary files and devices
(which is normally a bad idea, but that's what we DO) that's just not a
good fit for us.

I'll look a little more. Today is Geriatric Garmin Day....

RJL

./gpsbabel -D9 -i garmin -f usb:
GPSBabel Version: 1.5.4
TX [12]:00 00 00 00 05 00 00 00 00 00 00 00 ............(SESREQ  )
TX [12]:00 00 00 00 05 00 00 00 00 00 00 00 ............(SESREQ  )
TX [12]:00 00 00 00 05 00 00 00 00 00 00 00 ............(SESREQ  )
RX (intr) [16]:00 00 00 00 06 00 00 00 04 00 00 00 f2 73 f1 ce
............�s��(ACK     )
TX [12]:14 00 00 00 fe 00 00 00 00 00 00 00 ....�.......(PRDREQ  )
RX (intr) [16]:00 00 00 00 06 00 00 00 04 00 00 00 f2 73 f1 ce
............�s��(ACK     )
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK  )
RX (bulk) [102]:14 00 00 00 ff 00 00 00 5a 00 00 00 b7 02 0e 01 65 54 72 65
78 20 56 65 6e 74 75 72 65 20 48 43 20 53 6f 66 74 77 61 72 65 20 56 65 72
73 69 6f 6e 20 32 2e 37 30 00 56 45 52 42 4d 41 50 20 41 6d 65 72 69 63 61
73 20 52 65 63 20 42 61 73 65 6d 61 70 20 34 2e 30 30 00 56 45 52 53 4d 41
50 20 4e 6f 6e 65 00
....�...Z.......eTrex.Venture.HC.Software.Version.2.70.VERBMAP.Americas.Rec.Basemap.4.00.VERSMAP.None.(PRDDAT
 )
RX (bulk) [0]:(RET2INTR)
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK  )
RX (bulk) [26]:14 00 00 00 f8 00 00 00 0e 00 00 00 56 45 52 47 50 52 4f 4d
20 4e 6f 6e 65 00 ....�.......VERGPROM.None.(PRDEDA  )
RX (bulk) [0]:(RET2INTR)
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK  )
RX (bulk) [150]:14 00 00 00 fd 00 00 00 8a 00 00 00 50 00 00 4c 01 00 41 0a
00 54 01 00 41 64 00 44 6e 00 41 c9 00 44 ca 00 44 6e 00 44 d2 00 41 2d 01
44 38 01 44 2e 01 41 90 01 44 6e 00 41 f4 01 44 f5 01 41 58 02 44 58 02 41
59 02 44 59 02 41 bc 02 44 bc 02 41 20 03 44 20 03 41 21 03 44 21 03 41 84
03 41 86 03 41 87 03 41 88 03 41 89 03 44 84 03 41 8b 03 44 8b 03 44 8c 03
44 8d 03 44 8e 03 41 8c 03 44 8f 03 41 92 03 41 94 03 41 95 03 44 95 03 41
96 03 44 96 03
....�.......P..L..A..T..Ad.Dn.A�.D�.Dn.D�.A..D8.D..A..Dn.A�.D�.AX.DX.AY.DY.A..D..A..D..A..D..A..A..A..A..A..D..A..D..D..D..D..A..D..A..A..A..D..A..D..(PRTARR
 )
TX [12]:14 00 00 00 fe 00 00 00 00 00 00 00 ....�.......(PRDREQ  )
RX (bulk) [0]:(RET2INTR)
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK  )
RX (bulk) [102]:14 00 00 00 ff 00 00 00 5a 00 00 00 b7 02 0e 01 65 54 72 65
78 20 56 65 6e 74 75 72 65 20 48 43 20 53 6f 66 74 77 61 72 65 20 56 65 72
73 69 6f 6e 20 32 2e 37 30 00 56 45 52 42 4d 41 50 20 41 6d 65 72 69 63 61
73 20 52 65 63 20 42 61 73 65 6d 61 70 20 34 2e 30 30 00 56 45 52 53 4d 41
50 20 4e 6f 6e 65 00
....�...Z.......eTrex.Venture.HC.Software.Version.2.70.VERBMAP.Americas.Rec.Basemap.4.00.VERSMAP.None.(PRDDAT
 )
Unit: eTrex Venture HC Software Version 2.70
ID: 695
Version: 2.70RX (bulk) [0]:(RET2INTR)
[... 192 lines cut ...]
52 20 4d 49 43 52 4f 20 31 00 00 00 00 00
........a...............�������������.�...v.QY.iQY.iQY.i....����QY.i......HLI�CHERMICR1.HLI�CHER.MICRO.1.....(WPTDAT
 )
RX (bulk) [0]:(RET2INTR)
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK  )
RX (bulk) [124]:14 00 00 00 23 00 00 00 70 00 00 00 01 00 1f 80 3f 20 00 00
00 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff 72 f2 c1 25 ae bd ed 05 51
59 04 69 51 59 04 69 51 59 04 69 20 20 20 20 ff ff ff ff 51 59 04 69 00 00
00 00 00 00 9c 62 65 72 44 69 65 73 65 42 72 fc 63 6b 00 9c 62 65 72 20 64
69 65 73 65 20 42 72 fc 63 6b 65 20 6d 75 df 74 20 44 75 20 67 65 68 6e 00
00 00 00 00
........p...............������������r��...�.QY.iQY.iQY.i....����QY.i.......berDieseBr�ck..
ber.diese.Br�cke.mu�t.Du.gehn.....(WPTDAT  )
RX (bulk) [0]:(RET2INTR)
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK  )
RX (bulk) [14]:14 00 00 00 0c 00 00 00 02 00 00 00 07 00
..............(XFRCMP  )
RX (bulk) [0]:(RET2INTR)
cet_util: Converting from "US-ASCII" to "UTF-8", done.
cet_util: Converting from "UTF-8" to "US-ASCII", done.
53.096633N 8.337200E $berDieseBruck/$ber diese Brucke mu$t Du gehn
51.932433N 9.086233E $HLIOCHERMICR1/$HLIOCHER MICRO 1
51.932250N 9.086033E AA149BH/FINAL
51.932433N 9.086233E HLIÖCHERMICR1/HLIÖCHER MICRO 1
53.096633N 8.337200E œberDieseBrück/œber diese Brücke mußt Du gehn

tsteven4, 2019-07-31:
What version of MacOS were you running?  There was some suspicion this 
was related to 10.14.6.

Not that upgrading to libusb 1.0 is feasible, but that project is 
active, 10 commits this month.  I am interested if that sees Ture's device.

It's hard to get the usb serial port adapter and device to work when you 
have to plug it into a MacStadium mini!

tsteven4, 2019-07-31:
What version of MacOS were you running?  There was some suspicion this was
related to 10.14.6.

Not that upgrading to libusb 1.0 is feasible, but that project is active,
10 commits this month.  I am interested if that sees Ture's device.

It's hard to get the usb serial port adapter and device to work when you
have to plug it into a MacStadium mini!

On 7/31/2019 2:54 PM, GPSBabel wrote:

I don't have an HCX, but I have a Venture HC here in the GPSBabel-lab.
Modulo some silliness in the debug printing (patch coming) is seems OK to
me on my iMac.

The ugly squares just need some && isascii added. Actually, maybe I byte
the bullet and unify the four different places we print these lines...

This is 1.5.4 just to show the ugli ones as I've "fixed" my work tree to
not do that. But 1.6.0 from the master DMG works in the same way for me.

I don't see anything in the upstream libusb project that's pertinent. That
project was dead for a long time. Maybe I'll try grafting in a current
release (if they've not destroyed the API again) just to see if that brings
any happiness.

I don't *think* it's a sandboxing issue. We don't pretend to be able to
live in a sandbox - when you read and write arbitrary files and devices
(which is normally a bad idea, but that's what we DO) that's just not a
good fit for us.

I'll look a little more. Today is Geriatric Garmin Day....

RJL

./gpsbabel -D9 -i garmin -f usb:
GPSBabel Version: 1.5.4
TX [12]:00 00 00 00 05 00 00 00 00 00 00 00 ............(SESREQ )
TX [12]:00 00 00 00 05 00 00 00 00 00 00 00 ............(SESREQ )
TX [12]:00 00 00 00 05 00 00 00 00 00 00 00 ............(SESREQ )
RX (intr) [16]:00 00 00 00 06 00 00 00 04 00 00 00 f2 73 f1 ce
............�s��(ACK )
TX [12]:14 00 00 00 fe 00 00 00 00 00 00 00 ....�.......(PRDREQ )
RX (intr) [16]:00 00 00 00 06 00 00 00 04 00 00 00 f2 73 f1 ce
............�s��(ACK )
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK )
RX (bulk) [102]:14 00 00 00 ff 00 00 00 5a 00 00 00 b7 02 0e 01 65 54 72 65
78 20 56 65 6e 74 75 72 65 20 48 43 20 53 6f 66 74 77 61 72 65 20 56 65 72
73 69 6f 6e 20 32 2e 37 30 00 56 45 52 42 4d 41 50 20 41 6d 65 72 69 63 61
73 20 52 65 63 20 42 61 73 65 6d 61 70 20 34 2e 30 30 00 56 45 52 53 4d 41
50 20 4e 6f 6e 65 00
....�...Z.......eTrex.Venture.HC.Software.Version.2.70.VERBMAP.Americas.Rec.Basemap.4.00.VERSMAP.None.(PRDDAT
)
RX (bulk) [0]:(RET2INTR)
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK )
RX (bulk) [26]:14 00 00 00 f8 00 00 00 0e 00 00 00 56 45 52 47 50 52 4f 4d
20 4e 6f 6e 65 00 ....�.......VERGPROM.None.(PRDEDA )
RX (bulk) [0]:(RET2INTR)
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK )
RX (bulk) [150]:14 00 00 00 fd 00 00 00 8a 00 00 00 50 00 00 4c 01 00 41 0a
00 54 01 00 41 64 00 44 6e 00 41 c9 00 44 ca 00 44 6e 00 44 d2 00 41 2d 01
44 38 01 44 2e 01 41 90 01 44 6e 00 41 f4 01 44 f5 01 41 58 02 44 58 02 41
59 02 44 59 02 41 bc 02 44 bc 02 41 20 03 44 20 03 41 21 03 44 21 03 41 84
03 41 86 03 41 87 03 41 88 03 41 89 03 44 84 03 41 8b 03 44 8b 03 44 8c 03
44 8d 03 44 8e 03 41 8c 03 44 8f 03 41 92 03 41 94 03 41 95 03 44 95 03 41
96 03 44 96 03
....�.......P..L..A..T..Ad.Dn.A�.D�.Dn.D�.A..D8.D..A..Dn.A�.D�.AX.DX.AY.DY.A..D..A..D..A..D..A..A..A..A..A..D..A..D..D..D..D..A..D..A..A..A..D..A..D..(PRTARR
)
TX [12]:14 00 00 00 fe 00 00 00 00 00 00 00 ....�.......(PRDREQ )
RX (bulk) [0]:(RET2INTR)
[... 255 lines cut ...]
> > Dev (bus 29, device 4): Apple, Inc. - IR Receiver
> > Dev (bus 29, device 3): Apple Inc. - BRCM20702 Hub
> > Dev (bus 29, device 2): 0424 - 2512
> > Dev (bus 26, device 1): 8087 - 0024
> > Dev (bus 29, device 1): 8087 - 0024
> Steve
>
>
> >
> > I checked the command-line versions, both from the distributed DMG and
> > the binary I built myself from GIT, and both are reported as
> > "unsandboxed" (sample from home-made one, but the one from the DMG
> > gives identical output, except the path)
> >
> > |$ asctl sandbox check --pid 2453
> > /Users/ture/Desktop/gpsbabel/gpsbabel: not signed with App Sandbox
> > entitlements running without App Sandbox enabled running unsandboxed |
> >
> > <
https://github.com/gpsbabel/gpsbabel/issues/381?email_source=notifications&email_token=ADHXMMLBCHE6QEN4HHZQQITQCHC35A5CNFSM4IIH7JXKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD3H6UFQ#issuecomment-516942358
>,
>
> > <
https://github.com/
,

robertlipe, 2019-08-01:
I had to change systems, but that same Venture HC was working fine with
GPSBabel. I forced an update to 10.14.6 and it broke as described:

 ./gpsbabel -D2 -i garmin -f usb:
make: Nothing to be done for `all'.
GPSBabel Version: 1.6.0
main: Compiled with Qt 5.12.0 for architecture x86_64-little_endian-lp64
main: Running with Qt 5.12.0 on macOS 10.14, x86_64
main: QLocale::system() is en_US
main: QLocale() is en_US
main: QTextCodec::codecForLocale() is UTF-8, mib 106
Found no Garmin USB devices.

So it's definitely something in 10.14.6.  We simply never see the Garmin
vendor ID in the enumeration cycle. The device shows in System
Informatino->USB, so it's not totally hosed.

Honestly, these Garmin USB units were a huge disaster to support when I did
them back in 2005 and 2006. We found the Garmins doing crazy things on the
bus like returning the number of bytes that Windows requested instead of
the bytes that the OS requested just because they didn't care about
anything but Windows. Garmin fixed some of their USB sins and we actually
helped get some patches into OS/X 10.3 or 10.4 or something.

I no longer have any engineering contacts with either group, so please push
back onto Apple about this. Breaking apps (hey, that's us!) in a
maintenance point release is something they generally don't do, so
hopefully we can get some help from them.

And, yes, Steven, I know well what it's like to be on the wrong end of a
wire from the hardware. That was my career for about 20 years. :-) Writing
a webusb layer that looked a lot like our existing libusb/windows layer
(the abstraction below garmin packaets and above USB frames) is something
that's probably possible, but in the land of crazy talk.

robertlipe, 2019-08-01:
Actually, I now see Steven was one step ahead of me.  Again.  The libusb
app that he asked you to run correctly identifies the 91e (Garmin):

./testlibusb
Dev (bus 20, device 2): 091E - 0003
Dev (bus 29, device 4): Apple Inc. - Apple Internal Keyboard / Trackpad
Dev (bus 29, device 3): Apple Inc. - BRCM20702 Hub
Dev (bus 29, device 2): 0424 - 2512
Dev (bus 29, device 1): 8087 - 0024
Dev (bus 26, device 1): 8087 - 0024

So we either go through the libusb stack and cherry pick a fix from
somewhere in the last 15 years or we bring in a new libusb with an
incompatible interface or we bring in the libusb0.1compat. My guess is that
the middle of those is the best, but it's also the biggest pain.  That was
requested by someone that was reselling GPSBabel a while back, but that
wasn't interested in doing the work, IIRC.

Just drat. Garmins are just causing grief this week after years of silence.

RJL

tsteven4, 2019-08-01:
Good to know, at least we are narrowing the field.

tsteven4, 2019-08-01:
So now we should build testlibusb from 0.1.12 and see if it is something 
in gpsbabel or something in the old libusb.

robertlipe, 2019-08-01:
Maybe it won't be too bad. IIRC, our enumeration loop was right out of one
of their example and we have a pretty pedestrian use that's synchronous
with no overlapping I/Os. It was painful to develop because I was trying to
use the same upper-level everywhere with the (alto then-reverse engineered)
Garmin Windows driver and libUSB (gusb_llops_t) and do it on all the OSes
at once. Beyond all the screaming you can hear in gpslibusb.cc, I got to
find out things like the GPS would ignore the packet following a packet
that was exactly the size of the packet for some crazy reason.

The Garmin USB work is easily the most painful work I've ever put into
GPSBabel, so I wince at the thought of being radical with it. We'd need to
touch-test it on Linux and maybe some of the BSD's, too.

As for being something in GPSBabel, I'm 90% sure it's something the OS that
a newer libusb just knows how to deal with. While these Garmins may not be
popular now I *know* this was a heavily trafficed path for us for some time
and really hasn't changed much.

It's also at least 8 years newer than the serial devices that are next in
line after a break.

tsteven4, 2019-08-01:
I have testlibusb built with 0.1.12.  I had to turn of Werror, but it 
seems to run on macstadium.  I'll post a link in a minute.

tsteven4, 2019-08-01:
Here is testlibusb from 0.1.12.

https://filetransfer.io/data-package/Dk0BF6DL

If this sees the garmin device on 10.14.6 then the problem is in 
gpsbabel.  If not, its in the libusb 0.1.12.

tsteven4, 2019-08-01:
BTW, if you want to verify the 0.1.12 test:

> 1952:Desktop administrator$ openssl dgst -sha256 testlibusb_0.1.12.zip
> SHA256(testlibusb_0.1.12.zip)= 
> 908d7ada6e28f6a4c75e83a1e2f5e7a714875bd99918d6ec40c0a6342f52a02c

turepalsson, 2019-08-01:
Sorry I'm so slow to respond... been night here in .se. :-)

Not sure whether you still need this, but just for good measure, here's the output from testusb, built from git (v1.0.23-rc1-12-gdb99ef3) on my hardware:

    Dev (bus 20, device 10): 091E - 0003
    Dev (bus 20, device 4): Apple Inc. - BRCM20702 Hub
    Dev (bus 20, device 3): Apple Inc. - FaceTime HD Camera (Built-in)
    Dev (bus 20, device 2): Seagate - Expansion

tsteven4, 2019-08-01:
Ture,

Thanks for this report.  It matches Roberts results, recent versions of 
libusb 1.0 can see the garmin device on 10.14.6.

gpsbabel uses a slightly modified version of libusb 0.1.12. 
Unfortunately the API is incompatible between libusb 0.x and 1.y. We 
would like to know if the problem exists in libusb 0.1.12 independent of 
gpsbabel.

Therefore we would like to know if libusb 0.1 can see the device on 
10.14.6 using testlibusb (in the tests directory of libusb-0.1.12).

The source code is available here: 
https://sourceforge.net/projects/libusb/files/libusb-0.1%20%28LEGACY%29/0.1.12/

In order to build it I had to remove "-Werror" from the Makefile.

I built it myself and the executable and library is at 
https://filetransfer.io/data-package/Dk0BF6DL but feel free to build it 
yourself from the source if you prefer.

> 1952:Desktop administrator$ openssl dgst -sha256 testlibusb_0.1.12.zip
> SHA256(testlibusb_0.1.12.zip)= 
> 908d7ada6e28f6a4c75e83a1e2f5e7a714875bd99918d6ec40c0a6342f52a02c
If you can let us know the results of this test it would be helpful.

Thanks,

Steve

turepalsson, 2019-08-01:
tests/testlibusb build from libusb-0.1.12.tar.gz:

    Dev #0: 05AC - 8007
    Dev #0: Seagate - Expansion
    Dev #0: Apple Inc. - FaceTime HD Camera (Built-in)
    Dev #0: Apple Inc. - BRCM20702 Hub

tsteven4, 2019-08-01:
So it didn't see it which suggests the underlying problem is with libusb 
0.1.12.

Thanks

tsteven4, 2019-08-01:
using libusb 1.0 without the compat layer may not be too difficult, despite the apparent amount of blood that was spilled in getting usb to originally work with the various garmin devices.  I think all the libusb calls are from gpslibusb.cc.   When I look how some of these (bulk_read, bulk_write, interrupt_read) are mapped in the compat layer it seems pretty straight forward to do it directly.  Testing will be an issue.

~/work/magproto_glob/jeeps$ readelf -s *.o | grep UND | grep " usb_"
    44: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_release_interface
    45: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_close
    47: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_bulk_read
    48: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_interrupt_read
    49: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_bulk_write
    52: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_strerror
    63: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_open
    65: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_claim_interface
    69: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_init
    72: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_find_busses
    73: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_find_devices
    74: 0000000000000000     0 NOTYPE  GLOBAL DEFAULT  UND usb_get_busses

gpsbabel, 2019-08-12:
WIP: https://github.com/gpsbabel/gpsbabel/tree/libusbv1 When I left, It built and "just" needed the real work done to move to the new API. I'll come back to it. If we shortcut with libusb_open_device_with_vid_pid() to simplify enumeration, the conversion to libusb_{bulk,interrupt}_transfer looks pretty mechanical.

tsteven4, 2019-08-20:
@turepalsson If you are brave you could try branch libusb1_conversion.  Since I don't have a Garmin USB device I can only get as far as "Found no Garmin USB devices."

I would be interested in the output of:
LIBUSB_DEBUG=4 /Applications/GPSBabelFE.app/Contents/MacOS/gpsbabel -w -t -i garmin -f usb: -o gpx -F out.gpx

https://github.com/gpsbabel/gpsbabel/tree/libusb1_conversion

turepalsson, 2019-08-20:
_It's possible that I bungled something when compiling, but here's the output (let's see if I can attach a file in github; never done that before...)_

[typescript.txt](https://github.com/gpsbabel/gpsbabel/files/3522078/typescript.txt)

turepalsson, 2019-08-20:
_Here's an attempt at getting a traceback_

* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2b8)
  * frame #0: 0x000000010015add7 gpsbabel`libusb_exit + 119
    frame #1: 0x00007fff6d1703d6 libsystem_c.dylib`__cxa_finalize_ranges + 326
    frame #2: 0x00007fff6d1706b3 libsystem_c.dylib`exit + 55
    frame #3: 0x0000000100000beb gpsbabel`main(argc=11, argv=<unavailable>) at main.cc:746:3 [opt]
    frame #4: 0x00007fff6d0ca3d5 libdyld.dylib`start + 1
    frame #5: 0x00007fff6d0ca3d5 libdyld.dylib`start + 1

tsteven4, 2019-08-20:
>       o frame #0: 0x000000010015add7 gpsbabel|libusb_exit + 119 frame
>         0x0000000100000beb gpsbabel|main(argc=11, argv=) at

It seems like it got pretty far, perhaps dying as it tried to close libusb.

The traceback would be more useful if you did a debug build.

If you are using qmake

qmake CONFIG+=debug

make clean

make

If you have used configure in the directory you may need to delete 
config.h first, best start with a clean clone.

turepalsson, 2019-08-20:
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2b8)
      * frame #0: 0x00000001000047fc gpsbabel`list_del(entry=0x00000000000002b8) at libusbi.h:151:29
        frame #1: 0x0000000100006ef4 gpsbabel`libusb_exit(ctx=0x0000000000000000) at core.c:2252:2
        frame #2: 0x00000001001f56a8 gpsbabel`gusb_teardown(dh=0x0000000000000000) at gpslibusb.cc:187:3
        frame #3: 0x00000001001f5371 gpsbabel`gusb_atexit_teardown() at gpslibusb.cc:194:3
        frame #4: 0x00007fff6d1703d6 libsystem_c.dylib`__cxa_finalize_ranges + 326
        frame #5: 0x00007fff6d1706b3 libsystem_c.dylib`exit + 55
        frame #6: 0x00000001001a538d gpsbabel`main(argc=11, argv=0x00007ffeefbff908) at main.cc:746:3
        frame #7: 0x00007fff6d0ca3d5 libdyld.dylib`start + 1
        frame #8: 0x00007fff6d0ca3d5 libdyld.dylib`start + 1

tsteven4, 2019-08-20:
wow, we made it to the very last line in main before we died.  I may 
know why.  Did you get your gpx file?  It should have been written by 
the time we got this far in main.

turepalsson, 2019-08-20:
Heh, I actually did not think of looking for the GPX. :-) Yes, it’s there, and it looks reasonable!

tsteven4, 2019-08-20:
please do a 'git pull', rebuild and retest.  I think I fixed it with 
commit a1d6866b58a05e78af8251fcc88ac05e331715d7

turepalsson, 2019-08-20:
I'm off to bed, and then to work... I'll give it a go in 22 hours or so. :)

tsteven4, 2019-08-21:
While I believe a1d6866 would have alleviated your symptoms I committed a better fix, 5b41236.  git pull in the libusb1_conversion branch should now get 5b41236, please try it instead.

turepalsson, 2019-08-21:
gpsbabel_1_5_4-627-g5b41236d runs without segfaults, and seems to produce a correct GPX file. I also tried uploading the file back to the device and that, too, looks like it's working.

tsteven4, 2019-08-21:
Thanks for your report and testing.

I recognize https://github.com/gpsbabel/gpsbabel/commit/5b41236dcc4736f9ebbe20f8964b1d69b62f2600 as a commit on libusb1_conversion, but I am not sure about the rest of your name "gpsbabel_1_5_4-627-g...".  If we are still naming something 1_5_4 please let me know how you got that, nothing current should generate 1_5_4 strings.

turepalsson, 2019-08-21:
It's the output of 'git describe'.

tsteven4, 2019-08-21:
Thanks, I created an annotated tag for the 1.6.0 release, gpsbabel_1_6_0.1, so the git describe output makes a bit more sense now (gpsbabel_1_6_0.1-47-g5b41236d).

gpsbabel, 2019-08-30:
Notepad testing as I go in case I lose my place:
roberts-mbp:gpsbabel robertlipe$ ./gpsbabel -r -w -t -D9 -i garmin -f usb:-1
0 XXX 382 StreetPilot c320 Software Version 5.70
/Applications/Xcode.app/Contents/Developer/usr/bin/lldb -- ./gpsbabel -r -w -t -D9 -i garmin -f usb:1
 120 	  int ret = libusb_bulk_transfer(udev, gusb_bulk_out_ep, buf, sz,
   121 	                                 &transferred, TMOUT_B);
(lldb) p gusb_bulk_out_ep
(unsigned char) $4 = '\0'

Meh. Don't specify an out of bounds device number. Yeah, we should do better.

0 XXX 292 GPSMap60CSX Software Version 3.60
 /gpsbabel -r -w -t -D9 -i garmin -f usb:0
Confirmed RTE, TRK, WPT flew by. Good.

60CS FAILS
Edge 305 FAILS
./gpsbabel -r -w -t -D9 -i garmin -f usb:-1
GPSBabel Version: 1.6.0
main: Compiled with Qt 5.12.0 for architecture x86_64-little_endian-lp64
main: Running with Qt 5.12.0 on macOS 10.14, x86_64
main: QLocale::system() is en_US
main: QLocale() is en_US
main: QTextCodec::codecForLocale() is UTF-8, mib 106
libusb_bulk_transfer failed. 'Input/Output Error'

Model Venture HC
Track, Route, Waypoint reads are fine.
Self identification fails:
 ./gpsbabel -D9 -i garmin -f usb:-1
main: QTextCodec::codecForLocale() is UTF-8, mib 106
TX [12]:00 00 00 00 05 00 00 00 00 00 00 00 ............(SESREQ  )
TX [12]:00 00 00 00 05 00 00 00 00 00 00 00 ............(SESREQ  )
TX [12]:00 00 00 00 05 00 00 00 00 00 00 00 ............(SESREQ  )
RX (intr) [16]:00 00 00 00 06 00 00 00 04 00 00 00 f2 73 f1 ce .............s..(ACK     )
TX [12]:14 00 00 00 fe 00 00 00 00 00 00 00 ............(PRDREQ  )
RX (intr) [16]:00 00 00 00 06 00 00 00 04 00 00 00 f2 73 f1 ce .............s..(ACK     )
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK  )
RX (bulk) [102]:14 00 00 00 ff 00 00 00 5a 00 00 00 b7 02 0e 01 65 54 72 65 78 20 56 65 6e 74 75 72 65 20 48 43 20 53 6f 66 74 77 61 72 65 20 56 65 72 73 69 6f 6e 20 32 2e 37 30 00 56 45 52 42 4d 41 50 20 41 6d 65 72 69 63 61 73 20 52 65 63 20 42 61 73 65 6d 61 70 20 34 2e 30 30 00 56 45 52 53 4d 41 50 20 4e 6f 6e 65 00 ........Z.......eTrex.Venture.HC.Software.Version.2.70.VERBMAP.Americas.Rec.Basemap.4.00.VERSMAP.None.(PRDDAT  )
RX (bulk) [0]:(RET2INTR)
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK  )
RX (bulk) [26]:14 00 00 00 f8 00 00 00 0e 00 00 00 56 45 52 47 50 52 4f 4d 20 4e 6f 6e 65 00 ............VERGPROM.None.(PRDEDA  )
RX (bulk) [0]:(RET2INTR)
RX (intr) [12]:00 00 00 00 02 00 00 00 00 00 00 00 ............(REQBLK  )
RX (bulk) [150]:14 00 00 00 fd 00 00 00 8a 00 00 00 50 00 00 4c 01 00 41 0a 00 54 01 00 41 64 00 44 6e 00 41 c9 00 44 ca 00 44 6e 00 44 d2 00 41 2d 01 44 38 01 44 2e 01 41 90 01 44 6e 00 41 f4 01 44 f5 01 41 58 02 44 58 02 41 59 02 44 59 02 41 bc 02 44 bc 02 41 20 03 44 20 03 41 21 03 44 21 03 41 84 03 41 86 03 41 87 03 41 88 03 41 89 03 44 84 03 41 8b 03 44 8b 03 44 8c 03 44 8d 03 44 8e 03 41 8c 03 44 8f 03 41 92 03 41 94 03 41 95 03 44 95 03 41 96 03 44 96 03 ............P..L..A..T..Ad.Dn.A..D..Dn.D..A..D8.D..A..Dn.A..D..AX.DX.AY.DY.A..D..A..D..A..D..A..A..A..A..A..D..A..D..D..D..D..A..D..A..A..A..D..A..D..(PRTARR  )
RX (bulk) [0]:(RET2INTR)

It just stops.  It's actually failing to parse both the Garmin-level device stuff AND the usb-level stuff, but the device seems to mostly work. Weird.

Plugging both the 60CSx and the C320 at the same time, which work individually, we crash:
   313 	    return;
   314 	  }
-> 315 	  ret  = libusb_open(dev, &udev);
   316 	  if (ret != LIBUSB_SUCCESS) {
   317 	    fatal("libusb_open failed: %s\n",
(lldb) p dev
(libusb_device *) $4 = 0x0000000100a017b0
(lldb) p udev
(libusb_device_handle *) $5 = 0x0000000000000000

Summary:
Multi-device support is broken. It's likely that nobody cares.
Edge 305, 60CS have problem. ISTR they bounce between control and interrrupt pipes like the others don't - or vice versa. I def recall the older ones shared a trait that was blindsided by the newer ones.
It's definitely better than current on the latest MacOS, but getting back to all devices working should be possible. 

More later. (maybe)

tsteven4, 2019-08-31:
I think I may understand the last failure you reported: 

> Plugging both the 60CSx and the C320 at the same time, which work individually, we crash:

You didn't say if you were doing a list (usb:-1 or usb:list).  If so, then in garmin_usb_scan after we call garmin_usb_start on the first garmin device we will call gusb_close. 

https://github.com/gpsbabel/gpsbabel/blob/3b656719d096660d16c099cca24b954711f82dd8/jeeps/gpslibusb.cc#L473-L482

The problem is that gusb_close will result in gusb_teardown, which now calls libusb_exit.  There wasn't an analogous call with libusb 0.1.  Then, returning to the loop above, when we try to open the second device, we have deinitialized libusb (libusb_exit) so when garmin_usb_start tries to open the second device it fails.

https://github.com/gpsbabel/gpsbabel/blob/3b656719d096660d16c099cca24b954711f82dd8/jeeps/gpslibusb.cc#L188-L191

The code needs to be restructured so we can separate 
`udev != nullptr`
part of gusb_teardown from the 
`libusb_successfully_initialized`
part.  At line 481 above we only want to do the udev stuff, everywhere else we can do both.

tsteven4, 2019-10-02:
> Multi-device support is broken.
> Plugging both the 60CSx and the C320 at the same time, which work individually, we crash:

should be fixed with https://github.com/gpsbabel/gpsbabel/pull/396/commits/80015bcefa265d21151f404b852333aea227c545

> Edge 305, 60CS have problem
I don't understand this failure.

@gpsbabel, can you please retest multidevice support.

tsteven4, 2019-10-06:
I have access to a etrex vista cx.  I am running gpsbabel in a virtual box ubuntu guest.  The host is windows 10.
Reading waypoints, routes and tracks usually works.  I have seen some hangs, but I am suspicious of the virtualization layer.
Self identification works (usb:-1).

tsteven4, 2019-10-07:
> Meh. Don't specify an out of bounds device number. Yeah, we should do better.

Fixed with https://github.com/gpsbabel/gpsbabel/pull/396/commits/b8857a6eaac058e64ad619d976a7efa20bd96b6b

PR #396 (2019-11-14): Replace libusb 0.1 usage with libusb 1.0
Replace our modified libusb 0.1.12 with libusb 1.0.22 on macos.  On macos we statically link to our inclcuded libusb.
Use libusb 1.0 instead of libusb 0.1 on linux.  On linux we dynamically link to a distribution provided libusb, or optionally build without libusb support.

This fixes #381 
This resolves #25 as requested (albeit 3.5 years later)
closed 2019-11-14 by tsteven4