GPSBabel/gpsbabel #389: gpsbabelfe "Translation options" traffic lights not shown

cjmayo, 2019-08-07:
If I run `gpsbabelfe` (with my default LANG environment variable, set to en_GB.utf8) I see the traffic light symbols against the Waypoints, Routes and Tracks check boxes.

But if I run `LANG=de_DE gpsbabelfe` I just see "hyphens" in line with the bottom of the check boxes. Changing the Input Format using the combo box causes the traffic lights to be displayed.

Starting values

Input Format: GPX XML
Output Format: Alan Map500 Waypoints and Routes (.wpr)

This is on GPSBabel 1.6.0 plus the patches for translations.
I don't see it on GPSBabel 1.5.4 - the Gentoo installation with translations in /usr/share/gpsbabel and without the code modified to load them.

tsteven4, 2019-08-07:
Thanks.  I can reproduce this.  Please open an issue on github.

tsteven4, 2019-08-08:
That was a great find, thanks @cjmayo for noticing and reporting it.

tsteven4, 2019-08-10:
This was fixed by #390

closed 2019-08-10 by tsteven4