GPSBabel/gpsbabel #397: Power information lost when converting .fit to .gpx

Qwertie-, 2019-08-23:
I have a giant neostrack gps device which records a bunch of data like cadence, power, temperature, and heart rate. When I use gpsbabel to convert the .fit file to .gpx the power information gets lost.

I am using the option `garminextensions=1` which converts most of the data over but not power. When I open the .fit file using GPXSee the power information is in there but not after converting. Is there another setting to enable to get this or is it missing from gpsbabel.
I have attached an example .fit file which contains power data

[fit file.zip](https://github.com/gpsbabel/gpsbabel/files/3534320/fit.file.zip)

tsteven4, 2019-08-23:
We read the power data from your fit file.  You can see this with

./gpsbabel -t -i garmin_fit -f fit\ file/190801165626.fit  -o unicsv -F -

But the gpx schema and all the extensions we support do not have a 
element for the power data, so we don't write it when using gpx as an 
output format.

Qwertie-, 2019-08-25:
Well that is an interesting problem. I guess if it was just put in the track point extension section there is the probability that no software will use/understand the data.

The CSV export bit is interesting. The reason I am trying to convert these files is because strava errors out at about half of my files but every other program seems to read them fine. I imagine there is probably some bad value in the file somewhere that every other program handles fine. Does gpsbabel provide any tools for inspecting the content of fit files or any error logging to show when bad values were ignored?

tsteven4, 2019-08-25:
In general we cannot simple add an element to an existing schema, it 
would invalidate the document.  In the cases where the schema allows an 
extension there is little chance anyone will recognized it since it is 
non-standard.  So without a recognized gpx schema extension that 
includes power that we don't know about I think gpx is out.

I think Strava should upload fit, tcx or gpx.  I assume you have tried 
to upload your fit file, how about

gpsbabel -i garmin_fit -f 190801165626.fit -o gtrnctr -F 190801165626.tcx

I have verified that

i) the tcx file contains your power data.

ii) the gpsbabel created tcx file meets the tcx schema and garmin 
extension schema, so if Strava doesn't handle it it's on them.

If you add -D9 or -D some other numeral you can see some diagnostics 
from gpsbabel reading your fit file.  Some parts of it we don't 
recognized and ignore.

gpx and tcx are xml files with schema definitions, and validators exist 
to check conformance.  fit is a garmin binary format, I don't know of 
any way to validate it.

>
>       *Upload a file from your computer *
>
> You can upload files directly to Strava via our file uploader found here:
>
> http://www.strava.com/upload/select.
>
> You can upload the following file types as long as the file is 25MB or 
> smaller and contains workout data:
>
>   * GPX
>   * TCX
>   * FIT
>

Qwertie-, 2019-08-25:
Thanks for the help. Looks like there is nothing the project can do to fix this so I guess it makes sense to close this issue now. I'll take a look at those other options you suggested.

closed 2019-12-06 by robertlipe