GPSBabel/gpsbabel #402: not working with my FIT file

evazzoler, 2019-09-25:
Hi,
when I try to convert this ([download_site_1](https://gofile.io/?c=3P0PqF) - [download_site_2](http://nullupload.com/download-8ac7ed84d3)) FIT file (from command line passing "-i garmin_fit"), i get this error:
`fit: record truncated: expecting char[4], but only got 1`
The file is produced by a Garmin Oregon 700 latest firmware version.

tsteven4, 2019-10-02:
When reporting issues like this please supply the information requested at
https://www.gpsbabel.org/lists.html, i.e.

> NOTE: When asking a question, please include enough information to let us help you. Be sure to include version numbers, the name of the operating system, and details of any programs or hardware involved. If you're having a problem, describe it in enough detail that someone else can reproduce it. Include any needed data files. If they're more than a few kilobytes, please try simplifying the data first. (We really don't need 50,000 trackpoints to reproduce a crash or see if we support a file when 10 will do...) Of course, if you're reporting something that is large and does require a huge data set to reproduce it, please put it on a personal web site or use something like http://drop.io to share the file.

In this case your issue was fixed in 1.6.0, which has been available since May 2019 at https://www.gpsbabel.org/download.html

tsteven4, 2019-10-19:
Fixed in release 1.6.0.

closed 2019-10-19 by tsteven4