GPSBabel/gpsbabel #436: Invalid NMEA checksum even if i verify them manually that are correct

Akanichi, 2019-12-13:
Hi,

I just download GPSBABEL and try to use as an input an NMEA file and for Output GARMIN Serial Protocol.

when i start conversion, i get this error message for all the lines  :

The NMEA FILE : 
$GPGGA,145546.00,4841.7270743,N,00211.6664674,E,5,04,0.0,79.758,M,44.971,M,0.0,0*40
$GPGGA,145547.00,4841.7232178,N,00211.6619604,E,5,04,0.0,79.870,M,44.971,M,0.0,0*41
$GPGGA,145548.00,4841.7233369,N,00211.6620859,E,5,04,0.0,78.832,M,44.971,M,0.0,0*46
$GPGGA,145549.00,4841.7248957,N,00211.6524695,E,5,04,0.0,55.568,M,44.971,M,0.0,0*48
$GPGGA,145550.00,4841.7268614,N,00211.6564853,E,5,04,0.0,81.284,M,44.971,M,0.0,0*46
$GPGGA,145551.00,4841.7295935,N,00211.6540671,E,5,04,0.0,95.723,M,44.971,M,0.0,0*4C

The Error Message :

Invalid NMEA checksum. Computed  36  but found  32 . Ignoring sentence 
Invalid NMEA checksum. Computed  36  but found  91 . Ignoring sentence 
Invalid NMEA checksum. Computed  46  but found  81 . Ignoring sentence 
Invalid NMEA checksum. Computed  33  but found  94 . Ignoring sentence 
Invalid NMEA checksum. Computed  39  but found  35 . Ignoring sentence 
Invalid NMEA checksum. Computed  34  but found  38 . Ignoring sentence 
Invalid NMEA checksum. Computed  36  but found  91 . Ignoring sentence 

I verified the lines on https://nmeachecksum.eqth.net/
and i get exactly the same checksum.

tsteven4, 2019-12-13:
Works for me.

> :~/work/gpsbabel-1.6.0$ ./gpsbabel -i nmea,date=20190101 -f 
> ../csqaured/x.nmea  -o gpx -F -
> <?xml version="1.0" encoding="UTF-8"?>
> <gpx version="1.0" creator="GPSBabel - http://www.gpsbabel.org" 
> xmlns="http://www.topografix.com/GPX/1/0">
>   <time>2019-12-13T14:50:55.818Z</time>
>   <bounds minlat="48.695386963" minlon="2.194207825" 
> maxlat="48.695493225" maxlon="2.194441123"/>
>   <trk>
>     <trkseg>
>       <trkpt lat="48.695451238" lon="2.194441123">
>         <ele>79.758</ele>
>         <time>2019-01-01T14:55:46Z</time>
>         <geoidheight>45.0</geoidheight>
>         <sat>4</sat>
>       </trkpt>
>       <trkpt lat="48.695386963" lon="2.194366007">
>         <ele>79.870</ele>
>         <time>2019-01-01T14:55:47Z</time>

Akanichi, 2019-12-13:
I just found the solution on the NMEA file the line is written this way 
$GPGGA,145546.00,4841.7270743,N,00211.6664674,E,5,04,0.0,79.758,M,44.971,M,0.0,040
and it have to be written without the $ at the beginning like this example:
GPGGA,145546.00,4841.7270743,N,00211.6664674,E,5,04,0.0,79.758,M,44.971,M,0.0,040

Akanichi, 2019-12-13:
> I just found the solution on the NMEA file the line is written this way
> $GPGGA,145546.00,4841.7270743,N,00211.6664674,E,5,04,0.0,79.758,M,44.971,M,0.0,040
> and it have to be written without the $ at [...]

closed 2019-12-13 by Akanichi

tsteven4, 2019-12-13:
The start of sentence delimiter ('$') should be included, not excluded.

> ./gpsbabel-1.6.0/gpsbabel -t -i nmea,date=20190101 -f x.nmea -o unicsv 
> -F -
> No,Latitude,Longitude,Altitude,Satellites,Date,Time
> 1,48.695451,2.194441,79.8,4,2019/01/01,07:55:46
> 2,48.695387,2.194366,79.9,4,2019/01/01,07:55:47
> 3,48.695389,2.194368,78.8,4,2019/01/01,07:55:48
> 4,48.695415,2.194208,55.6,4,2019/01/01,07:55:49
> 5,48.695448,2.194275,81.3,4,2019/01/01,07:55:50
> 6,48.695493,2.194234,95.7,4,2019/01/01,07:55:51
> cat x.nmea
> $GPGGA,145546.00,4841.7270743,N,00211.6664674,E,5,04,0.0,79.758,M,44.971,M,0.0,040
> $GPGGA,145547.00,4841.7232178,N,00211.6619604,E,5,04,0.0,79.870,M,44.971,M,0.0,041
> $GPGGA,145548.00,4841.7233369,N,00211.6620859,E,5,04,0.0,78.832,M,44.971,M,0.0,046
> $GPGGA,145549.00,4841.7248957,N,00211.6524695,E,5,04,0.0,55.568,M,44.971,M,0.0,048
> $GPGGA,145550.00,4841.7268614,N,00211.6564853,E,5,04,0.0,81.284,M,44.971,M,0.0,046
> $GPGGA,145551.00,4841.7295935,N,00211.6540671,E,5,04,0.0,95.723,M,44.971,M,0.0,04C