GPSBabel/gpsbabel #445: Error parsing valid IGC  Headers

rolfoz, 2019-12-22:
The IGC standard was updated to include new headers. If these are present in an IGC file gpsbabel throws an error and does not process the file.
I confirmed this on the latest version built on linux a few minutes ago.

 gpsbabel -i igc -f Documents/Protected/Personal_Docs/igc/2019/2019-12-18-XCT-RSC-01.igc -o gpx -F testigc.gpx
IGC: header (H) record parse error
HSCCLCOMPETITION CLASS:FAI-3

If I remove the offending header the file is processed OK. I have seen the same error with a couple of other headers from different files created by other devices.

This is happening with all IGC files produced by the program XCTrack, but also by files generated by other newer programs that utilise some of the new header types.

I looked for the latest specification for IGC files, hoping to find a list of all the valid headers, but I was unable to locate it anywhere.

I have attached the IGC file.

[2019-12-18-XCT-RSC-01.zip](https://github.com/gpsbabel/gpsbabel/files/3992518/2019-12-18-XCT-RSC-01.zip)

rolfoz, 2019-12-22:
This is what generates the error:

 case rec_header:
      // Get the header sub type
      if (sscanf(ibuf, "H%*1[FOP]%3s", tmp_str) != 1) {
        fatal(MYNAME ": header (H) record parse error\n%s\n%s\n", ibuf, tmp_str);
      }

I tried commenting out the fatal line and rebuilding and now it appears to work correctly.

tsteven4, 2019-12-22:
http://www.ukiws.demon.co.uk/GFAC/documents/tech_spec_gnss.pdf

rolfoz, 2019-12-23:
I have read through the new IGC specification and identified the issue and I can propose a fix.

The original IGC proposal required H records to start with either an O or and F , the new IGC specification adds the letter S.

A3.2.6 Additional H records. These are optional and use the appropriate Three Letter Codes given in para A7.
Additional data may be appended after the mandatory records. Two additional records (Competition ID and class) are
shown below, with extra spaces for clarity.
H S CID COMPETITION ID : TEXT STRING CR LF
H S CCL COMPETITION CLASS : TEXT STRING CR LF

The fix is the addition of the letter S to the code shown below. I have made this change and rebuilt the Linux binary and I can confirm that it works.

Line 308 needs to have the letter S appended to the matched letters as shown below.

   if (sscanf(ibuf, "H%*1[FOPS]%3s", tmp_str) != 1) {

closed 2019-12-24 by tsteven4